sebanc/brunch #65: Macbook 4,1 Not Working

santhoshr, 2020-04-24:
Hi,

I have very old laptop Macbook 4,1, cloudready works fine, installing "rammus" based chrome-os fails while booting it shows some failed error, sometimes it goes through it after long time but loops on some USB/Apple IR device found message continuously. The processor is Core 2 duo, and it supports UEFI booting.

sebanc, 2020-04-25:
Could you attach a short video of the boot process to this issue ?

santhoshr, 2020-04-26:
CloudReady and crouton works, I'm not missing Android, so if you interested in fixing this let me know, I will attach video, otherwise you can delete this issue. Thanks.

sebanc, 2020-05-03:
Hi, I have reviewed my kernel config a lot in lastest testing release and I think it could improve things, don't hesitate to test it if you want, else I will close the issue.

closed 2020-05-18 by sebanc