sebanc/brunch #66: Video Problem in Arch and Linux Error in  Chrome Os (Brunch)

mrLoliKing, 2020-04-25:
There's a problem in hardware acceleration on Arch the video playback is green.
And the Linux is going to error in my machine it won't install.
The the screen capture or recording in arch won't work and it going to freeze the screen and the arch will crash.

My Machine and Chrome Version:
Chrome Version: 

80.0.3987.162 (Official Build) (64-bit)

My Machine:
HP Probook 4520s
Intel Core i3-350M Processor (2.26 GHz, 3MB L3 cache)
Intel HD Graphics (Ironlake)

sebanc, 2020-04-25:
Hi,

I am working on improving crostini compatibility, however i am not sure if it will help for your config. If you can test, It would be helpful for me to know if crostini works on cloudready ?

mrLoliKing, 2020-04-25:
Ok i'll send you tomorrow or Monday the results because its 12:43 am in my time right now and i need to rest for tomorrow to some research and solve this problem too

sebanc, 2020-04-25:
no worries, when you have the time

mrLoliKing, 2020-04-26:
Um the crostini in cloudready is working on my machine 100%

mrLoliKing, 2020-04-26:
There's no problem in cloudready using a crostini linux at all thats the results

sebanc, 2020-04-26:
Thanks for the feedback, i will try to achieve the same result

mrLoliKing, 2020-04-27:
Good Luck

mrLoliKing, 2020-04-27:
I'll wait for the results

sebanc, 2020-05-05:
Could you try the latest testing release ?

mrLoliKing, 2020-05-05:
Ok wait a sec

mrLoliKing, 2020-05-05:
Can i test this tommorow its already night tho

mrLoliKing, 2020-05-05:
I will give you the results when im done testing

mrLoliKing, 2020-05-06:
Still not fix:
[image]
[image]

mrLoliKing, 2020-05-06:
[image]
S

sebanc, 2020-05-06:
Could you try with "samus" recovery image ?

Anyway, you will have to switch to "samus" as "rammus" will fail on your device from r83 (google has added cpu instructions to the build which are not supported by your device...

mrLoliKing, 2020-05-06:
Im going to test tommorow its already night

mrLoliKing, 2020-05-11:
Sorry i cant test this month im so busy ill going tell you the results if i have exact time to test 🤔😟

sebanc, 2020-06-02:
You have to remove "virtio_gpu" (crostini gpu acceleration) from /etc/ui_use_flags.txt for crostini to install.

mrLoliKing, 2020-06-02:
Um idk if i can do this in time but i will try it

sebanc, 2020-06-26:
I am closing this issue as the solution provided should work but feel free to reopen it if it is not.

closed 2020-06-26 by sebanc