sebanc/brunch #67: BCM Bluetooth error

rote66, 2020-04-25:
Hello.
My bluetooth hardware is BCM4334,and the bluetooth can be enabled,but can't scan the devices.
And the dmesg says : bluetooth hci0: Direct firmware load for brcm/BCM.hcd failed with error -2
I know it has BCM4324B3.hcd  in firmware,but it doesn't use it.
Thanks your work,its much appreciated.

sebanc, 2020-04-25:
Hi,
Could you try unzip the attached file, rename it to "BCM.hcd" and copy it to the firmware folder ?
[bcm4334.hcd.zip](https://github.com/sebanc/brunch/files/4532400/bcm4334.hcd.zip)

rote66, 2020-04-25:
> Hi,
> Could you unzip the attached file, rename it to "BCM.hcd" and copy it to the firmware folder ?
> [bcm4334.hcd.zip](https://github.com/sebanc/brunch/files/4532400/bcm4334.hcd.zip)

I tried it.
After i put it in brcm and reboot,then bluetooth can not be enabled.
Can't see load BCM.hcd error in log.
In /var/log/messages:
    
INFO kernel: [   46.309825] Bluetooth: af_bluetooth.c:bt_init() Core ver 2.22
INFO kernel: [   46.310828] Bluetooth: af_bluetooth.c:bt_init() HCI device and connection manager initialized
INFO bluetoothd[3338]: Bluetooth daemon 5.44
INFO bluetoothd[3338]: Starting SDP server
ERR bluetoothd[3338]: Failed to open RFKILL control device
INFO bluetoothd[3338]: Bluetooth management interface 1.14 initialized

sebanc, 2020-04-25:
What is the output of "rfkill list all" ?

rote66, 2020-04-25:
> What is the output of "rfkill list all" ?

When i put the hcd,it just have:

ERR bluetoothd[2696]: Failed to open RFKILL control device
ERR bluetoothd[3073]: Failed to open RFKILL control device
ERR bluetoothd[2507]: Failed to open RFKILL control device
ERR bluetoothd[2493]: Failed to open RFKILL control device
ERR bluetoothd[2550]: Failed to open RFKILL control device
ERR bluetoothd[3338]: Failed to open RFKILL control device

When i removed hcd and reboot :

INFO kernel: [   44.630169] Bluetooth: af_bluetooth.c:bt_init() Core ver 2.22
INFO kernel: [   44.630315] Bluetooth: af_bluetooth.c:bt_init() HCI device and connection manager initialized
WARNING kernel: [   45.017417] bluetooth hci0: Direct firmware load for brcm/BCM.hcd failed with error -2
INFO bluetoothd[2550]: Bluetooth daemon 5.44
INFO bluetoothd[2550]: Starting SDP server
ERR bluetoothd[2550]: Failed to open RFKILL control device
INFO bluetoothd[2550]: Bluetooth management interface 1.14 initialized
INFO bluetoothd[2550]: Endpoint registered: sender=:1.45 path=/org/chromium/Cras/Bluetooth/A2DPSource
INFO bluetoothd[2550]: Player registered: sender=:1.45 path=/org/chromium/Cras/Bluetooth/DefaultPlayer
INFO bluetoothd[2550]: adapter /org/bluez/hci0 has been enabled
INFO bluetoothd[2550]: adapter /org/bluez/hci0 set power to 0

lfom, 2020-04-25:
@rote66 What is your device, exactly? Are you sure it uses the B3 hcd? You may want to filter you log for "brcmfmac" as well to see if it is the firmware that is actually missing... I have a ThinkPad 8 that has the same SDIO, and normally the problem is the firmware, but a few time the wrong hcd ws sim-linked...

rote66, 2020-04-26:
> @rote66 What is your device, exactly? Are you sure it uses the B3 hcd? You may want to filter you log for "brcmfmac" as well to see if it is the firmware that is actually missing... I have a ThinkPad 8 that has the same SDIO, and normally the problem is the firmware, but a few times the wrong hcd ws sim-linked...

ThinkPad10,I grep the "brcmfmac" and it just says missing bcm.hcd.....

lfom, 2020-04-26:
@rote66 You may want to check this:
https://github.com/winterheart/broadcom-bt-firmware

rote66, 2020-04-26:
> @rote66 You may want to check these pages:
> https://askubuntu.com/questions/632336/bluetooth-broadcom-43142-isnt-working
> https://github.com/winterheart/broadcom-bt-firmware

I replaced the BCM.hcd from bcm4324b5 hcd files.
And bluetooth can be enabled.but can't scan devices

lfom, 2020-04-26:
@rote66 Any error messages still related to bluetooth/brcmfmac in the log? Did you reboot? What is the output of `rfkill list all`?

rote66, 2020-04-26:
> @rote66 Any error messages still related to bluetooth/brcmfmac in the log? Did you reboot? What is the output of `rfkill list all`?

Yes,rebooted.
In rfkill list:
0: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
1: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no

In bluetooth log: 
INFO bluetoothd[2473]: Bluetooth daemon 5.44
INFO bluetoothd[2473]: Starting SDP server
ERR bluetoothd[2473]: Failed to open RFKILL control device
 INFO bluetoothd[2473]: Bluetooth management interface 1.14 initialized
INFO btdispatch[2485]: Bluetooth daemon started with passthrough mode = 1
INFO kernel: [   60.434972] Bluetooth: tty.c:rfcomm_init_ttys() RFCOMM TTY layer initialized
INFO kernel: [   60.434996] Bluetooth: sock.c:rfcomm_init_sockets() RFCOMM socket layer initialized
INFO kernel: [   60.435058] Bluetooth: core.c:rfcomm_init() RFCOMM ver 1.11
INFO bluetoothd[2473]: Endpoint registered: sender=:1.43 path=/org/chromium/Cras/Bluetooth/A2DPSource
INFO bluetoothd[2473]: Player registered: sender=:1.43 path=/org/chromium/Cras/Bluetooth/DefaultPlayer
INFO bluetoothd[2473]: adapter /org/bluez/hci0 has been enabled
INFO bluetoothd[2473]: adapter /org/bluez/hci0 set power to 0
INFO bluetoothd[2473]: adapter /org/bluez/hci0 has been disabled
INFO bluetoothd[2473]: adapter /org/bluez/hci0 set power to 1
INFO bluetoothd[2473]: adapter /org/bluez/hci0 has been enabled
INFO bluetoothd[2473]: Set Advertising Intervals: 0x00a0, 0x00a0

And brcmfmac log same.

rote66, 2020-04-26:
> What is the output of "rfkill list all" ?

0: phy0: Wireless LAN
Soft blocked: no
Hard blocked: no
1: hci0: Bluetooth
Soft blocked: no
Hard blocked: no

RFKILL can see bluetooth now,but can't scan any devices

lfom, 2020-04-26:
Well, I am writing this from my brunch installation (HD img, thank you @sebanc very much!), and I am having a similar issue: when I try to do a BT scan, WiFi disconnects, and BT does not find anything. I have to disable BT so WiFi works again (I keep keep both ON without scanning). My SDIO is an Intel via i2c, the drivers seem OK, both are listed in rfkill... Maybe both issues are related?

Also, as a side note, I have some system crashes, the last time I was able to go to shell and dmesg showed a crash related to i2c (and if IIRIC, it listed BayTrail as well, aslthough I am running on CherryTrail). It is silent, the system simply hangs forever withou restarting or freezing the UI. I used rammus v80 with latest brunch.

rote66, 2020-04-27:
> Well, I am writing this from my brunch installation (HD img, thank you @sebanc very much!), and I am having a similar issue: when I try to do a BT scan, WiFi disconnects, and BT does not find anything. I have to disable BT so WiFi works again (I keep keep both ON without scanning). My SDIO is an Intel via i2c, the drivers seem OK, both are listed in rfkill... Maybe both issues are related?
> 
> Also, as a side note, I have some system crashes, the last time I was able to go to shell and dmesg showed a crash related to i2c (and if IIRIC, it listed BayTrail as well, aslthough I am running on CherryTrail). It is silent, the system simply hangs forever withou restarting or freezing the UI. I used rammus v80 with latest brunch.
> 
> Edit: maybe this helps somehow:
> 
> > 2020-04-26T15:02:59.845123-04:00 INFO bluetoothd[2169]: Bluetooth daemon 5.44
> > 2020-04-26T15:02:59.854358-04:00 INFO bluetoothd[2169]: Starting SDP server
> > 2020-04-26T15:02:59.867060-04:00 INFO kernel: [   50.100343] Bluetooth: btusb.c:btusb_setup_intel() hci0: Intel Bluetooth fw patch 0x32 completed and activated
> > 2020-04-26T15:02:59.870132-04:00 ERR bluetoothd[2169]: Failed to open RFKILL control device
> > 2020-04-26T15:02:59.870390-04:00 INFO bluetoothd[2169]: Bluetooth management interface 1.14 initialized
> > 2020-04-26T15:02:59.885892-04:00 ERR cras_server[1637]: RegisterProfile returned error: org.bluez.Error.AlreadyExists
> > 2020-04-26T15:02:59.887357-04:00 ERR cras_server[1637]: RegisterProfile returned error: org.bluez.Error.AlreadyExists
> > 2020-04-26T15:03:00.019794-04:00 INFO newblued[2212]: LE splitter not enabled, newblued running in idle mode.
> > 2020-04-26T15:03:00.020091-04:00 INFO newblued[2212]: To enable, run 'newblue enable' in crosh and reboot.
> 
> And the correct command is actually `newblued enable,` what also gives this error when run as root:
> 
> > [ERROR:bus.cc(584)] Failed to get the ownership of org.chromium.Newblue: Connection ":1.134" is not allowed to own the service "org.chromium.Newblue" due to security policies in the configuration file
> > [ERROR:newblue_daemon.cc(34)] Failed to acquire D-Bus name ownership: org.chromium.Newblue
> > [INFO:newblue_daemon.cc(39)] LE splitter not enabled, newblued running in idle mode.
> > [INFO:newblue_daemon.cc(40)] To enable, run 'newblue enable' in crosh and reboot.
> 
> Edit2: I am not able to boot anymore after using `newblued enable`, but I was also trying to enable sensors and I am not sure if it triggered the kernel panic at boot...

I am using newblued enable and get the error same with you.
But i can enable wifi and bluetooth at the sametime.Wifi not disconnected.
Maybe this issue about the txt file?My wifi works when i put the "brcmfmac43241b5-sdio.txt" to the brcm folder,And the txt need modify?

lfom, 2020-04-27:
The TXT file is actually the firmware AFAIK... And I meant that WiFi turns off when I open Bluetooth panel to scan for devices, otherwise I can keep both turned on.

Anyway, I was able to boot again using iwlwifi_backport option, WiFi works, but the same happens with Bluetooth. Maybe it's a system issue that needs to be patched.

rote66, 2020-04-27:
> The TXT file is actually the firmware AFAIK... And I meant that WiFi turns off when I open Bluetooth panel to scan for devices, otherwise I can keep both turned on.
>
> Anyway, I was able to boot again [...]

I know.I meant that WIFI is normally when i open BT Panel to scan devices.Just in my devices.
And maybe it's about kernel drivers.

lfom, 2020-04-27:
Well, it seems that BT actually works: I was able to connect to my phone using `bluetoothctl`:

> # bluetoothctl
> [NEW] Controller XX:XX:XX:XX:XX:XX Chromebook_FB63 [default]
> [bluetooth]# devices
> [bluetooth]# scan on
> Discovery started
> [CHG] Controller XX:XX:XX:XX:XX:XX Discovering: yes
> [bluetooth]# pair YY:YY:YY:YY:YY:YY
> Device YY:YY:YY:YY:YY:YY not available
> [NEW] Device YY:YY:YY:YY:YY:YY Name
> [bluetooth]# pair YY:YY:YY:YY:YY:YY
> Attempting to pair with YY:YY:YY:YY:YY:YY
> [CHG] Device YY:YY:YY:YY:YY:YY Connected: yes

rote66, 2020-04-28:
> Well, it seems that BT actually works: I was able to connect to my phone using `bluetoothctl`:
> 
> > sudo bluetoothctl
> > [NEW] Controller XX:XX:XX:XX:XX:XX Chromebook_FB63 [default]
> > [bluetooth]# devices
> > [bluetooth]# scan on
> > Discovery started
> > [CHG] Controller XX:XX:XX:XX:XX:XX Discovering: yes
> > [bluetooth]# pair YY:YY:YY:YY:YY:YY
> > Device YY:YY:YY:YY:YY:YY not available
> > [NEW] Device YY:YY:YY:YY:YY:YY Name
> > [bluetooth]# pair YY:YY:YY:YY:YY:YY
> > Attempting to pair with YY:YY:YY:YY:YY:YY
> > [CHG] Device YY:YY:YY:YY:YY:YY Connected: yes
> 
> `btmon` also shows that scan works with GUI, so I am out of ideas now...

same
And I connect to my phone,system will appear "have connected",but it disconnect just in one second and can't connect to phone with GUI.

sebanc, 2020-05-06:
If I am not mistaken, you cannot connect a phone by bluetooth through the settings menu on chromeos, you have to use the specific phone connection setup menu for that:
https://support.google.com/chromebook/answer/7457148?hl=en

The bluetooth menu should be used for mouses, sound devices...

Could you confirm what are exactly your issues ?

rote66, 2020-05-06:
> If I am not mistaken, you cannot connect a phone by bluetooth through the settings menu on chromeos, you have to use the specific phone connection setup menu for that: [...]

Ok fine,I test it and i can connect to my bluetooth earphone.Thanks.

closed 2020-05-06 by rote66