sebanc/brunch #69: Device Compatibility Report: Boots on Chrome splash screen, then goes blank (Lenovo IdeaPad 330-15ARR)

Vaintelog, 2020-04-26:
I'm aware that this won't work on AMD based systems but I tried it anyway. 
I could boot using a SanDisk Ultra 16GB flash drive. The process worked without a cinch and it would boot up the Chrome OS splash screen (the blinding white screen with Chrome logo). It would stay like that for about 10 minutes or so, and then it would adjust the brightness itself and then the screen would turn off. Keys seem to respond as Num lock and Caps lock key lights turn on and off. I've also tried adding `fbcon=map:1` as the developer suggested in GRUB but same results.

Laptop: Lenovo IdeaPad 330-15ARR
CPU: AMD Ryzen 5 2500U
GPU 1: AMD Radeon Vega 8
GPU 2: AMD Radeon 540
BIOS Version: 7VCN48WW (09/26/2019)

sebanc, 2020-04-26:
Hi, I have no idea if it will work or not in the end but I am willing to help you try the few things I have in mind to get it to work on AMD.

First, if it is not already the case, could you try "grunt" image which is made for AMD hardware ?

Vaintelog, 2020-04-27:
I'll report back the progress

sebanc, 2020-04-28:
> I'll report back the progress
> 
> > First, if it is not already the case, could you try "grunt" image which is made for AMD hardware ?
> 
> Quick question:
> What's the difference between unpacking archives using Linux command vs. unpacking it using an archiving program (7-Zip)?

I think there is no difference

Vaintelog, 2020-04-28:
I've compiled the image file for the Chrome OS. I was bothered with the log so I'm just making sure nothing out of ordinary has happened. Here's the last lines from the log:

```
> Writing partition 1
> chromeos-install.sh: line 278: pv: command not found
> Writing partition 2
> chromeos-install.sh: line 278: pv: command not found
> Writing partition 3
> chromeos-install.sh: line 278: pv: command not found
> Writing partition 4
> chromeos-install.sh: line 278: pv: command not found
> Writing partition 5
> chromeos-install.sh: line 278: pv: command not found
> Writing partition 6
> chromeos-install.sh: line 278: pv: command not found
> Writing partition 7
> chromeos-install.sh: line 299: pv: command not found
> Writing partition 8
> chromeos-install.sh: line 278: pv: command not found
> Writing partition 9
> chromeos-install.sh: line 278: pv: command not found
> Writing partition 10
> chromeos-install.sh: line 278: pv: command not found
> Writing partition 11
> chromeos-install.sh: line 278: pv: command not found
> Writing partition 12
> chromeos-install.sh: line 306: pv: command not found
> ChromeOS installed.
```

juicypop, 2020-04-28:
Did you sudo apt-get install pv?

Make sure it's installed and try again

Vaintelog, 2020-04-28:
> Did you sudo apt-get install pv?
> 
> Make sure it's installed and try again

Seems like it wasn't installed like the `cgpt`. The three packages are installed correctly now. Progress bars have appeared. Also, I don't remember it well but I think there were also errors like these on my first try on making an image file of Chrome OS? Could that be the reason why things didn't work out using `rammus` recovery?

sebanc, 2020-04-28:
@juicypop thanks for the help :)

@Vaintelog Indeed the first image was probably not correct but amd devices won't boot "rammus" for sure, try "grunt" directly

sebanc, 2020-04-28:
Unfortunately no, could you take a short video of the boot process and post it somewhere or attach it as zip file to this issue ?

Vaintelog, 2020-04-29:
Sorry if it took long. If you need a better version, ask me.

[lenovo_ideapad_330_15arr.zip](https://github.com/sebanc/brunch/files/4552145/lenovo_ideapad_330_15arr.zip)

sebanc, 2020-04-30:
Hi,

Thanks for the video, it is perfect.

Unfortunately, what it shows is that the grunt recovery image is compiled using specific cpu instructions which are not supported by your cpu.

As the AMD "grunt" image is not compatible with your cpu and other recoveries are not compatible with AMD gpu, I am sorry to say that brunch cannot work on your laptop.

I heard recently that FydeOS had added support for AMD GPU and they probably build chromeos without specific cpu instructions, so that would probably be the way to go in your case.

Vaintelog, 2020-04-30:
That's sad.
Thanks for the project anyways! I hope this could revive a lot of old Intel laptops everywhere.

closed 2020-04-30 by Vaintelog

xeu100, 2020-12-19:
#722