sebanc/brunch #70: Bcm43142 System Unable to Activate Device

castlec, 2020-04-26:
BCM43142 in HP Pavilion 300-240 Desktop. The OS shows wifi as disabled. It does not enable. Internet functions with USB tethered device. Realtek Ethernet untested.

lfom, 2020-04-27:
Any change you tried activating broadcom_wl as expained in the main ReadMe? If you did, what happened?

castlec, 2020-04-27:
I did not see that. I will find the directions and report back.

castlec, 2020-04-27:
With broadcom_wl enabled, wifi scanning now works. Upon connection attempt to wpa2 psk, error is given. I have seen two errors. Out of range, and unassociated. 

This is on latest stable with rammus 80.

Update - after turning off security on my mobile hotspot, device connected to my home network with security enabled. It seems there was an error that it did not recover from well. 

Please let me know what logs I may collect or if the beta may resolve the issue.

sebanc, 2020-04-28:
Do you have secure boot enabled ?

Could you provide a dmesg ? (run ""dmesg > ~/Downloads/dmesg.txt" in crosh shell and attach dmesg.txt to this issue)

castlec, 2020-05-17:
Secure boot is not enabled. 

broadcom_wl isn't loading at all for whatever reason now....

`
HBRT Signal
[ 576.181294] init: shil1 main process ended, respawning
[ 576.181264] init: shil11 main process (18866) killed by ABRT signal
[ 575.029005] init: shi11 main process ended, respawning
[ 575.0289621 init: shill main process (18830) killed by ABRT signal
573.8820411 init: shill main process ended, respawning
[ 573.8819951 init: shill main process (18791) killed by ABRT signal
573.4579011 init: broadcom_wl main process (18796) terminated with status
573. 4534201 wl: Unknown symbol wiphy_free (err -2)
573. 4534081 wl: Unknown symbol cfg80211_connect_done (err -2)
573.4533891 wl: Unknown symbol ieee80211 frequency_to_channel (err -2)
573. 4538761 wl: Unknown symbol cfg80211 inform_bss_data (err -2)
573.4533591 wl: Unknown symbol cfg80211_report_wowlan_wakeup (err -2)
[ 573.4533391 wl: Unknown symbol ieee80211_channel to_frequency (err -2)
[ 573.4533091 wl: Unknown symbol cfg80211_get_bss (err -2)
573. 453291] wl: Unknown symbol wiphy_unregister (err -2)
573.4532741 wl: Unknown symbol ieee80211_get_channel (err -2)
573. 4532571 wl: Unknown symbol cfg80211_michael_mic_failure (err -2)
573.4532371 wl: Unknown symbol cfg80211 ibss_joi ned (err -2)
573.4532201 wl: Unknown symbol cfg80211_gtk_rekey_notify (err -2)
573.4532051 wl: Unknown symbol cfg80211_roamed (err -2)
573.453193) wl: Unknown symbol cfg80211_put_bss (err -2)
573.4531771 wl: Unknown symbol wiphy_register (err -2)
573.4531601 wl: Unknown symbol wiphy_new_nm (err -2)
573.4531321 wl: Unknown symbol cfg80211_disconnected (err -2)
573.4530901 wl: Unknown symbol cfg80211_scan_done (err -2)
573. 4529441 wl: Unknown symbol cfg80211_inform_bss frame_data (err -2)
572.7293971 init: shill main process ended. respawning
572.7293351 init: shill main process (18755) killed by ABRT signal
--More--

`

I think I will need to rebuild the image. I'm guessing something has gotten corrupted as I've bounced the drive from two PCs and have had lockups requiring hard power off.

sebanc, 2020-05-17:
Indeed the partition ROOT-A seems corrupt, you can try to rebuild rootfs by booting once with "options=rebuild" kernel command line parameter.
If it does not work a full reinstall will be needed.

castlec, 2020-05-17:
System had the same error after rebuild. I'll recreate from scratch.

closed 2020-06-02 by sebanc