sebanc/brunch #73: iwork10 flagship's touch screen flips left and right

moexco, 2020-04-27:
I tried to install chromeOS on my iwork10 flagship, Almost everything works.

But the screen touch has flipped left and right.

I don't know how to fix it, I cannot modify the touch screen parameters using xinput in chromeOS.

This is a great project, Thank you for everything you’ve done.

sebanc, 2020-04-27:
Hi,

Could you attach a dmesg to this issue please ?

To get the dmesg, go to crosh shell, run "dmesg > ~/Downloads/dmesg.txt" and attach the dmesg.txt file from your Downloads folder.

moexco, 2020-04-27:
[dmesg.txt](https://github.com/sebanc/brunch/files/4539267/dmesg.txt)
Sorry I was negligent, it's here

sebanc, 2020-04-27:
No worries, actually forgot to precise but I would need a dmesg after a fresh reboot

moexco, 2020-04-27:
[dmesg.txt](https://github.com/sebanc/brunch/files/4539470/dmesg.txt)
after reboot

sebanc, 2020-05-05:
Could you update the latest testing release, normally it should be fixed ?

sebanc, 2020-06-02:
I believe this issue is fixed so I will close it, don't hesitate to re-open it if needed.

closed 2020-06-02 by sebanc