sebanc/brunch #74: Play store and android apps are not opening

Sarojpadhy, 2020-04-27:
The playstore and other android apps installed are not opening . 

using chrome os Rammus recovery 80 and Brunch r81 k4.19 testing 20200419

PC
Intel pentium G3240m 
Intel integrated GPU

[image]

NSbuilder, 2020-04-27:
Same for me, except I used eve image instead of rammus so it might be this

lfom, 2020-04-27:
Can you open `chrome://flags`? Your processor does not have support for AES nor Trusted Exection, maybe it cannot enable the security layer properly... What is the result of the command below?
`cat /var/log/messages | grep -i "android|arc"`

Sarojpadhy, 2020-04-27:
> Can you open `chrome://flags`? Your processor does not have support for AES nor Trusted Exection, maybe it cannot enable the security layer properly... What is the result of the command below?
> `cat /var/log/messages | grep -i "android|arc"`

Yes  i can open `chrome://flags`

I didn't got that [> `cat /var/log/messages | grep -i "android|arc"`] . 
Can you explain it in a simple way

lfom, 2020-04-27:
You must open the shell: CTRL + ALT +T to open crosh, then type `shell`, then copy and paste the command, and paste back the results here...

Sarojpadhy, 2020-04-27:
> You must open the shell: CTRL + ALT +T to open crosh, then type `shell`, then copy and paste the command, and paste back the results here...

bash: cat/var/log/messages: No such file or directory

lfom, 2020-04-27:
You didn't type it correctly (there is a missing space after cat), it is better to copy and paste, that works differently in crosh.

Sarojpadhy, 2020-04-27:
> You didn't type it correctly (there is a missing space after cat), it is better to copy and paste, that works differently in crosh.

nothing happens if i insert a space 

chronos@localhost / $ cat /var/log/messages | grep -i "android|arc"
chronos@localhost / $

lfom, 2020-04-27:
My bad, `grep` in chrosh shell is very limited, the command below should be enough to show any issues:
`cat /var/log/messages | grep Android`

Sarojpadhy, 2020-04-27:
> My bad, `grep` in chrosh shell is very limited, the command below should be enough to show any issues:
> `cat /var/log/messages | grep Android`

chronos@localhost / $ cat /var/log/messages | grep Android
2020-04-27T15:34:59.256519+05:30 INFO session_manager[1127]: [INFO:session_manager_impl.cc(1779)] Android container with PID 2095 stopped
2020-04-27T15:35:00.002168+05:30 ERR session_manager[1127]: [ERROR:session_manager_impl.cc(1318)] Error getting Android container pid.
2020-04-27T15:35:00.002322+05:30 ERR session_manager[1127]: [ERROR:dbus_util.cc(13)] CreateError(...): Domain=dbus, Code=org.chromium.SessionManagerInterface.ContainerShutdownFail, Message=Error getting Android container pid.
2020-04-27T18:03:29.662719+05:30 INFO session_manager[1126]: [INFO:session_manager_impl.cc(1695)] Started Android container with PID 1995
2020-04-27T18:03:29.972645+05:30 INFO session_manager[1126]: [INFO:session_manager_impl.cc(1260)] Android container is running with PID 1995
2020-04-27T18:03:58.453733+05:30 INFO session_manager[1126]: [INFO:session_manager_impl.cc(1779)] Android container with PID 1995 stopped
2020-04-27T18:18:35.099827+05:30 INFO session_manager[1125]: [INFO:session_manager_impl.cc(1695)] Started Android container with PID 1860
2020-04-27T18:18:42.718623+05:30 INFO session_manager[1125]: [INFO:session_manager_impl.cc(1260)] Android container is running with PID 1860
2020-04-27T18:19:12.093890+05:30 INFO session_manager[1125]: [INFO:session_manager_impl.cc(1779)] Android container with PID 1860 stopped
2020-04-27T20:25:58.871298+05:30 INFO session_manager[1287]: [INFO:session_manager_impl.cc(1695)] Started Android container with PID 2002
2020-04-27T20:26:07.413353+05:30 INFO session_manager[1287]: [INFO:session_manager_impl.cc(1260)] Android container is running with PID 2002

lfom, 2020-04-27:
Weird, it looks fine... Are you running brunch from a USB flash drive? Does it have enough free space? What is the result of the command below?
`lsblk | grep container`

Sarojpadhy, 2020-04-27:
> Weird, it looks fine... Are you running brunch from a USB flash drive? Does it have enough free space? What is the result of the command below?
> `lsblk | grep container`

- No i installed it from the USB with Linux, as shown in the following video [https://www.youtube.com/watch?v=4gZYV0RWJQ8](url)
- Result 
loop1           7:1    0 486.8M  1 loop /opt/google/containers/android/rootfs/root
loop2           7:2    0     4K  1 loop /opt/google/containers/arc-sdcard/mountpoints/container-root
loop3           7:3    0     4K  1 loop /opt/google/containers/arc-obb-mounter/mountpoints/container-root

Sarojpadhy, 2020-04-27:
Well a good news, i tried running the play store again now, and it successfully opened.

lfom, 2020-04-27:
Good, everything looked normal. Maybe you just needed to wait for it to finish downloading and installing. Please close the issue.

Sarojpadhy, 2020-04-27:
> Good, everything looked normal. Maybe you just needed to wait for it to finish downloading and installing. Please close the issue.

I shut down the pc at that time, and when i booted now it took some time to boot and did some stuff like loading rootfc etc.
Now its working fine.
Hope i won't get that issue again.
thank you for your time and effort.

Sarojpadhy, 2020-04-28:
@lfom facing the same problem again. 
chronos@localhost / $ cat /var/log/messages | grep Android
2020-04-27T15:34:59.256519+05:30 INFO session_manager[1127]: [INFO:session_manager_impl.cc(1779)] Android container with PID 2095 stopped
2020-04-27T15:35:00.002168+05:30 ERR session_manager[1127]: [ERROR:session_manager_impl.cc(1318)] Error getting Android container pid.
2020-04-27T15:35:00.002322+05:30 ERR session_manager[1127]: [ERROR:dbus_util.cc(13)] CreateError(...): Domain=dbus, Code=org.chromium.SessionManagerInterface.ContainerShutdownFail, Message=Error getting Android container pid.
2020-04-27T18:03:29.662719+05:30 INFO session_manager[1126]: [INFO:session_manager_impl.cc(1695)] Started Android container with PID 1995
2020-04-27T18:03:29.972645+05:30 INFO session_manager[1126]: [INFO:session_manager_impl.cc(1260)] Android container is running with PID 1995
2020-04-27T18:03:58.453733+05:30 INFO session_manager[1126]: [INFO:session_manager_impl.cc(1779)] Android container with PID 1995 stopped
2020-04-27T18:18:35.099827+05:30 INFO session_manager[1125]: [INFO:session_manager_impl.cc(1695)] Started Android container with PID 1860
2020-04-27T18:18:42.718623+05:30 INFO session_manager[1125]: [INFO:session_manager_impl.cc(1260)] Android container is running with PID 1860
2020-04-27T18:19:12.093890+05:30 INFO session_manager[1125]: [INFO:session_manager_impl.cc(1779)] Android container with PID 1860 stopped
2020-04-27T20:25:58.871298+05:30 INFO session_manager[1287]: [INFO:session_manager_impl.cc(1695)] Started Android container with PID 2002
2020-04-27T20:26:07.413353+05:30 INFO session_manager[1287]: [INFO:session_manager_impl.cc(1260)] Android container is running with PID 2002
2020-04-27T22:34:21.995810+05:30 INFO session_manager[1287]: [INFO:session_manager_impl.cc(1779)] Android container with PID 2002 stopped
2020-04-27T22:34:22.660073+05:30 ERR session_manager[1287]: [ERROR:session_manager_impl.cc(1318)] Error getting Android container pid.
2020-04-27T22:34:22.660267+05:30 ERR session_manager[1287]: [ERROR:dbus_util.cc(13)] CreateError(...): Domain=dbus, Code=org.chromium.SessionManagerInterface.ContainerShutdownFail, Message=Error getting Android container pid.
2020-04-28T12:18:44.569688+05:30 INFO session_manager[1160]: [INFO:session_manager_impl.cc(1695)] Started Android container with PID 1847
2020-04-28T12:18:52.549282+05:30 INFO session_manager[1160]: [INFO:session_manager_impl.cc(1260)] Android container is running with PID 1847
2020-04-28T12:19:20.683237+05:30 INFO session_manager[1160]: [INFO:session_manager_impl.cc(1779)] Android container with PID 1847 stopped

sebanc, 2020-04-28:
Hi, Could you try deleting google sync ?
https://support.google.com/chromebook/answer/2914794?hl=en

Sarojpadhy, 2020-04-28:
> Hi, Could you try deleting google sync ?
> https://support.google.com/chromebook/answer/2914794?hl=en

[image]

sebanc, 2020-04-28:
Have you rebooted after deleting sync ? It's still not working ?

Sarojpadhy, 2020-04-28:
> Have you rebooted after deleting sync ? It's still not working ?

Yes i rebooted twice, but it didn't worked, now i signed out and sign in again. and it's working.
Don't know what's wrong.

sebanc, 2020-04-28:
Are you booting chromeos on a usb flashdrive ?

Sarojpadhy, 2020-04-28:
> Are you booting chromeos on a usb flashdrive ?

No, i have installed it on my hard disk.

sebanc, 2020-04-28:
Ok, I am currently refactoring the kernel and hope it will work better with the next release.

Sarojpadhy, 2020-04-28:
> Ok, I am currently refactoring the kernel and hope it will work better with the next release.

Thank you
BTW it always shows some processes for a minute while booting, is it normal  ?

sebanc, 2020-04-28:
If you mean the text on screen before the bootanimation appears, it's normal. It's so that I can debug issues, you can add "console=" to the kernel command line (after cros_debug) if you want a text free boot.

Nootboot64, 2020-04-28:
Hi, sorry, I have a similar problem. After booting for the second time (after I set it up, and am done with my computer for the day) Android apps do not work. No matter what, I can't figure it out. I've tried Croissant and Brunch images, with eve for Croissant and rammus for Brunch. These are my
 command results.
```
chronos@localhost / $ cat /var/log/messages | grep Android
2020-04-28T08:52:03.339357+00:00 INFO session_manager[1063]: [INFO:session_manager_impl.cc(1695)] Started Android container with PID 2348
2020-04-28T12:53:10.257379+00:00 INFO session_manager[1063]: [INFO:session_manager_impl.cc(1260)] Android container is running with PID 2348
2020-04-28T17:29:37.618466+00:00 INFO session_manager[1063]: [INFO:session_manager_impl.cc(1779)] Android container with PID 2348 stopped
2020-04-28T17:29:38.226238+00:00 ERR session_manager[1063]: [ERROR:session_manager_impl.cc(1318)] Error getting Android container pid.
2020-04-28T17:29:38.226919+00:00 ERR session_manager[1063]: [ERROR:dbus_util.cc(13)] CreateError(...): Domain=dbus, Code=org.chromium.SessionManagerInterface.ContainerShutdownFail, Message=Error getting Android container pid.
2020-04-28T13:31:28.471383-04:00 INFO session_manager[1060]: [INFO:session_manager_impl.cc(1695)] Started Android container with PID 2504
2020-04-28T13:31:29.088068-04:00 INFO session_manager[1060]: [INFO:session_manager_impl.cc(1260)] Android container is running with PID 2504
2020-04-28T13:32:00.121828-04:00 INFO session_manager[1060]: [INFO:session_manager_impl.cc(1779)] Android container with PID 2504 stopped
```
If you have any suggestions or help you can give me it would be nice. Thank you for your work on this project.

sebanc, 2020-04-28:
@Nootboot64 If you have not already done it, you try deleting google sync ?
https://support.google.com/chromebook/answer/2914794?hl=en

Nootboot64, 2020-04-28:
I did but am I supposed to log back in afterwards?

sebanc, 2020-04-28:
Yes, try to reboot afterwards

Nootboot64, 2020-04-28:
Ok, thank you

Nootboot64, 2020-04-28:
It didn't work... I think I did it right

sebanc, 2020-04-28:
@Nootboot64 Are you on brunch currently or on croissant ? the issues with android are not the same on both

Nootboot64, 2020-04-28:
Brunch

sebanc, 2020-04-28:
@Nootboot64 Ok, what's your laptop model ?

Nootboot64, 2020-04-28:
I have a Lenovo Ideapad 110 with 6gb ram, an Intel core i3, and integrated intel graphics.

Sarojpadhy, 2020-04-29:
I have to reset sync every time i face this issue .

Nootboot64, 2020-04-29:
After seeing your latest commit, I looked up which cpu generation I have, and came up with Intel Core i3-6100U CPU, which is a 6th gen cpu. So I still don't know why I'm having play store problems, but at least I know that it most likely isn't being caused by the recovery image (I'm using rammus).

juicypop, 2020-04-30:
> If you mean the text on screen before the bootanimation appears, it's normal. It's so that I can debug issues, you can add "console=" to the kernel command line (after cros_debug) if you want a text free boot.

There's a boot animation? I just tried your "console=" but this just shows an empty console with cursor until the login screen appears. 
Is there any way to have a custom boot animation with Chrome os?

sebanc, 2020-05-03:
Hi, I have reviewed the kernel config a lot in the new testing release, could you try it ?

Nootboot64, 2020-05-03:
Yes, luckily there's a way to dual boot cause I'm using windows now.

Nootboot64, 2020-05-03:
Sorry for doing a complete 180 on this, but I just cannot figure out how to dual boot, and I don't want to spend 4 hours on this. Sorry.

Nootboot64, 2020-05-04:
Ok, I actually decided to install it, do you have any specific things I should test?
Also, is there a way to check what version of brunch I have?

sebanc, 2020-05-04:
@Nootboot64 There is currently no way to know which brunch version you currently have but that's something I will try to implement. If you have a doubt, update to the latest version by running `sudo chromeos-update -f <path to the latest brunch archive>`.

Try to see with the new version if you have regressions compared to your current build but for now it appears to be quite stable.

Nootboot64, 2020-05-04:
Sorry, but I rebooted and the android apps don't boot again. I hope I can help test with the next release

sebanc, 2020-05-09:
@Nootboot64 @Sarojpadhy Could you try booting with "add_efi_memmap" on the kernel command line (after cros_debug) ?

MiLKQC, 2020-05-09:
> @Nootboot64 @Sarojpadhy Could you try booting with "add_efi_memmap" on the kernel command line (after cros_debug) ?

Hi, I have the same problem, I added the command but it didn't work after rebooting.

sebanc, 2020-05-13:
Hi, regarding this issue, the fact that google play works on first boot and then stops working most likely means that there is an issue with your profile / Google account setup.

Could you try the "Clear cache & data from Google Play Services" mentioned here?
https://support.google.com/googleplay/answer/7512202?hl=en

nicogig, 2020-05-17:
Hello, I'm having the same issue as @Nootboot64. The log output I gathered is the same as his when using cat and grep. Using the "add_efi_memmap" did not fix it, as well as resetting the sync options.

As a last attempt, I tried deleting the Play Store and reinstalling it from Chrome OS's settings, but now it does not install anymore ("Something went wrong" error shows up).

If it helps, the target device is a Lenovo Thinkpad Helix 2 (Intel Core M-5Y71) with 8GB RAM and 256GB SSD. The image used is rummus. It is currently booting with the options "enable_updates" (updated to r81, the problem was in r80 as well), "suspend_s3", and "advanced_als".

sebanc, 2020-05-17:
General question: Have you setup chromeos on several devices or just one ? Just to see if it could be related to that

nicogig, 2020-05-17:
> General question: Have you setup chromeos on several devices or just one ? Just to see if it could be related to that

Just one, the one specified above.

After removing Play Services, I tried rebooting a couple of times and Play Store seems to be up and running again. Will let you know if the issue comes up again.

harun911, 2020-05-17:
I encountered the same problem today on my laptop. I googled around a bit and found out that even official chromebooks sometimes have this problem. What works is, to let the OS sit and wait at the login screen for 2 minutes and then login. I think it probably has to do with the android emulator/environment not being fully booted up yet when logging in to fast.

nicogig, 2020-05-19:
> I encountered the same problem today on my laptop. I googled around a bit and found out that even official chromebooks sometimes have this problem. What works is, to let the OS sit and wait at the [...]

This seems to be it! I've been messing with my setup for the past day and just leaving it rest for a few minutes at the login screen fixed my issues.

khanhtranngoccva, 2020-05-21:
Well I don't know why the container keeps suffering from sigterm.
INFO arc-container[2462]: [INFO:run_oci.cc(871)] Sending signal 15 to container android-run_oci

sebanc, 2020-05-22:
Hi, thanks for the investigations you have done of this, the 2 minutes delay before bootup technique is interesting and I tried something to fix it in this release:
https://github.com/sebanc/brunch-testing/releases

Could you give it a try and see if you still need the 2 minutes delay to have android apps working ?

nicogig, 2020-05-22:
> Hi, thanks for the investigations you have done of this, the 2 minutes delay before login workaround is interesting and I tried something to fix it in this release:
> https://github.com/sebanc/brunch-testing/releases
> 
> Could you give it a try and see if you still need the 2 minutes delay to have android apps working ?

Tried it and looked in straight after boot, without waiting the 2 minutes. Play Store launched right away!

harun911, 2020-05-22:
I don't have this laptop at my disposal anymore, so I can't test your fix sebanc. I'll try to install the whole thing on a different laptop later (i5 6200u, nvidia mx150). What I remember from my findings with the other laptop, was that after the rebuilding of the root file, it always worked instantly. The problem started appearing every consecutive (re)boot after. Maybe you can try to validate that too @nicogig?

nicogig, 2020-05-22:
> I don't have this laptop at my disposal anymore, so I can't test your fix sebanc. I'll try to install the whole thing on a different laptop later (i5 6200u, nvidia mx150). What I remember from my [...]

I did a couple of reboots to be on the safe side, and Android Apps are working as expected without performing the workaround.

khanhtranngoccva, 2020-05-28:
The order of execution of fscrypt and cgroup might be why waiting for 2 minutes work. Can you check these screenshots?
[image]
[image]

closed 2020-06-26 by sebanc

karthikdhanavath, 2020-10-15:
please help me playstore not opening
[image]

goxtozo, 2021-07-15:
> My bad, `grep` in chrosh shell is very limited, the command below should be enough to show any issues:
> `cat /var/log/messages | grep Android`

[image]

Mine play store is also note working giving error 7 and my wifi is also not working I have a realtek 8822ce wifi adapter please help me sir

Icey67, 2022-02-02:
[image]
Everytime when i start ChromeOS, Android container errors keep appearing and will disappear by relogging but i'm sick of it,
Is there anyway i can permanently fix this?

[4 more replies]