sebanc/brunch #75: Frequent video playback studder/freeze

jhedfors, 2020-04-27:
Often during video playback (YouTube and other media player), the playback will get caught in a loop for several seconds where it will loop the last fraction of a second. All input will be frozen until a few seconds later which it then will continue normally.  

Please let me know if I can provide additional information to troubleshoot this.

lfom, 2020-04-27:
What are the CPU specs? Are you running brunch from USB flash disk?

jhedfors, 2020-04-27:
Dual Boot on a Dell laptop. Intel Core i5-7500 CPU @ 2.5GHz.

lfom, 2020-04-27:
So you are running from the img file on an internal disk? Does your computer have a dual GPU? If yes, did you enable the internal Intel chip?

jhedfors, 2020-04-27:
It has integrated Intel HD Graphics 620.  Is this the option you are refering to? -disable_intel_hda?

lfom, 2020-04-27:
No, it is for Intel HD audio. Then maybe graphics acceleration wasn't enabled for some reason, but I am not sure how to test it...

jhedfors, 2020-04-27:
Thanks for your help. Also, this is probably unsupported at this time, as had switched to the Dev channel  (84 build).... sorry, I should have mentioned that.

lfom, 2020-04-27:
Ah, yes... If you are using a dev build, maybe it's why. Video works fine here with rammus v80. Anyway, you can check if hardware acceleration is enabled with: 
`chrome://gpu/`

jhedfors, 2020-04-27:
Cool - thanks! Since you are not having that in 80, i will close this issue. As far as I can understand it, it appears taht hardware acceleration is on:

Canvas: Hardware accelerated
Flash: Hardware accelerated
Flash Stage3D: Hardware accelerated
Flash Stage3D Baseline profile: Hardware accelerated
Compositing: Hardware accelerated
Multiple Raster Threads: Enabled
Out-of-process Rasterization: Hardware accelerated
OpenGL: Enabled
Hardware Protected Video Decode: Unavailable
Rasterization: Hardware accelerated
Skia Renderer: Disabled
Video Decode: Hardware accelerated
Vulkan: Disabled
WebGL: Hardware accelerated
WebGL2: Hardware accelerated

jhedfors, 2020-04-27:
Closing as this issue is in DEV 84 build, and not in Stable 80.

jhedfors, 2020-04-29:
FYI, this is happeng in BETA v83 as well.... not sure if I want to roll back to 80 :)

jhedfors, 2020-05-02:
Just went back to Stable channel (v81), and the studdering/freezing is still happening. I assume it is hardware related.

sebanc, 2020-05-05:
Could you try the latest release ?

jhedfors, 2020-05-07:
I had switched back to Beta 83.0.4103.31 with the latest Brunch, and I just experienced a freeze, but this time it actually crashed the computer and restarted.  I will go back to test the latest brunch on the Stable release (I have both installed as separate images) and report back.

sebanc, 2020-05-08:
Hi, 

I have fixed audio issues which could cause that in the new release, could you try it ?

jhedfors, 2020-05-08:
I am installling the latest Brunch now... thanks. I will report back.

jhedfors, 2020-05-09:
Thank you. I will let you know if that resolves my issue.

sebanc, 2020-05-14:
I will try to think of something. What are the options or kernel command line arguments that you currently use ?

Also, could you verify what's the state of your cpu workload when the freezing occurs with  the"COG" extension ?

jhedfors, 2020-05-14:
I using only the enable-updates option. 

I will install COG extension, but it does not appear that the app shows any history, so I am not sure if I can get any usefull information unless I have it open all the time and visible - but I will check it out!

jhedfors, 2020-05-15:
As I suspected, COG does not help in this scenario, as the system is frozen (probably due to system resources being maxed out) prohibiting me from seeing the COG data. After the freezing stops, COG shows normal.

jhedfors, 2020-07-29:
Any suggestion to what log I can look at to see what could be causing this issue? 
file:///var/log/

It does not look like it is the OS, as it also happens while running Linux Mint.  I am wondering if it is a problem with the system RAM perhaps?

closed 2020-10-29 by jhedfors