sebanc/brunch #76: "Attempt to improve support for devices with dual gpu."

hacker1024, 2020-04-28:
I saw this in the latest testing release notes - would you mind elaborating? Is there laptop NVIDIA support?

sebanc, 2020-04-28:
Hi, the dual gpus (intel/nvidia or intel/amd) are theoretically supported. However, they might need some specific kernel command line arguments to correctly boot (for the intel gpu to be recognized as the main framebuffer device). Just to be clear, the other gpu is not used in chromeos.

There is one report of laptop with only an AMD gpu working with "grunt" recovery image but i am still unsure of the general AMD devices compatibility.

Laptops with only nvidia gpus are not supported as there is no chromeos recovery image with nvidia drivers currently.

hacker1024, 2020-04-28:
Thanks. Luckily my laptop (Intel UHD  620 (Whiskey Lake) + NVIDIA GeForce MX150) worked even before that release

closed 2020-04-28 by hacker1024