sebanc/brunch #77: Chrome Os Is Not Booting

MhFoysal, 2020-04-28:
My old pc have intel processor.UEFI is not support on my pc.
is there any brunch which is not require uefi firmwire.
after i installed chrome os then i restarted but i got a message which is no operating system found.

lfom, 2020-04-28:
UEFI is a requirement, as it is stated in the main ReadMe page. I doubt you can boot it on any BIOS computer since the kernel needs UEFI. You can try Androidx86, I think it still supports legacy boot.

MhFoysal, 2020-04-28:
> UEFI is a requirement, as it is stated in the main ReadMe page. I doubt you can boot it on any BIOS computer since the kernel needs UEFI. You can try Androidx86, I think it still supports legacy boot.

lfom, 2020-04-28:
As I said, I don't think so: UEFI requirement is a ChromiumOS/ChromeOS requirement. All ChromeOS devices are UEFI as far as I know.

MhFoysal, 2020-04-28:
My Device Is Not Chromebook It's a pc.
Whish is not Support UEFI.

sebanc, 2020-05-11:
Hi,

Actually someone apparently found a way to make brunch work on legacy bios devices using the linux dual boot method:
1) Install a linux distro on your device
2) Follow linux dual boot method

MhFoysal, 2020-05-11:
How Can You Just Tell Me The Details

sebanc, 2020-05-11:
1) Install ubuntu for exemple (refer to online resources)
2) Follow the instructions listed here:
https://github.com/sebanc/brunch#dual-boot-chromeos-from-your-hdd

If while you are attempting this you don't understand a specific step, ask here

MhFoysal, 2020-05-11:
Do You Have Any Source Which Source Can Help To Me To Create Dual Boot In
Linux Operating System.

darethehair, 2020-05-13:
I am trying to pursue this for the sake of being able to install Brunch onto an older BIOS laptop, but I have run into a GRUB2 error.

First, some basic background:

1) Preparation for mounting '/dev/sda4' for Brunch image installation:
```
darren@asusk501:~/Downloads$ mkdir -p ~/tmpmount
darren@asusk501:~/Downloads$ sudo mount /dev/sda4 ~/tmpmount

darren@asusk501:~/Downloads$ sudo fdisk -l /dev/sda
Disk /dev/sda: 232.9 GiB, 250059350016 bytes, 488397168 sectors
Disk model: WDC WD2500BPVT-2
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disklabel type: dos
Disk identifier: 0xb1453a34

Device     Boot     Start       End   Sectors   Size Id Type
/dev/sda1            2048     69631     67584    33M  b W95 FAT32
/dev/sda2  *        69632    274431    204800   100M ef EFI (FAT-12/16/32)
/dev/sda3          274432 209989631 209715200   100G 83 Linux
/dev/sda4       209989632 488396799 278407168 132.8G 83 Linux
```

2) Install Brunch image onto '/dev/sda4' partition (successfully): 
```
$ sudo bash chromeos-install.sh -src chromeos_12739.111.0_samus_recovery_stable-channel_mp-v3.bin -dst ~/tmpmount/chromeos.img -s 100
Creating image file
0+0 records in
0+0 records out
0 bytes copied, 0.000275558 s, 0.0 kB/s
WARNING: Primary GPT header is invalid
WARNING: Secondary GPT header is invalid
335FD367-2345-0248-AD79-7848E4B9C56A
       start        size    part  contents
           0           1          PMBR (Boot GUID: 335FD367-2345-0248-AD79-7848E4B9C56A)
           1           1          Pri GPT header
           2          32          Pri GPT table
...
ChromeOS installed.
To boot directly from this image file, add the lines between stars to either:
- A brunch usb flashdrive grub config file (then boot from usb and choose boot from disk image in the menu),
- Or your hard disk grub install if you have one (refer to you distro's online resources).
********************************************************************************
menuentry "ChromeOS (boot from disk image)" {
        img_part=/dev/sda4
        img_path=/chromeos.img
        search --no-floppy --set=root --file $img_path
        loopback loop $img_path
        linux (loop,gpt7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
                cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path
        initrd (loop,gpt7)/initramfs.img
```

3) Update '/etc/grub.d/40_custom' file (for Linux LMDE distro custom grub entries) with above code and then run:

```
$ sudo update-grub
Generating grub configuration file ...
Found theme: /boot/grub/themes/linuxmint/theme.txt
Found linux image: /boot/vmlinuz-4.19.0-9-amd64
Found initrd image: /boot/initrd.img-4.19.0-9-amd64
Found linux image: /boot/vmlinuz-4.19.0-8-amd64
Found initrd image: /boot/initrd.img-4.19.0-8-amd64
done
```

4) After reboot, 'ChromeOS' boot option is available, but when picked:

```
error: no such partition
error: you need to load the kernel first
```

A bit of poking inside GRUB reveals no problems:

```
grub> ls (hd0,msdos4)/
lost+found/ chromeos.img
```

So why is it complaining about a non-existent partition?  Is there something wrong with the generated GRUB2 entries?

A bigger question: CloudReady is able to install/boot on this old BIOS laptop just fine -- what trick does it use to be able to do that?  The Brunch 'USB stick' doesn't work at all on machines like that -- I suspect that it is because they are not set up to handle boots on BIOS -- just UEFI?

sebanc, 2020-05-13:
@darethehair Thanks for the feedback, the guy who posted the solution did not mention this and I did not think about this but the dual boot script probably has to be amended a bit.

I would say "linux (loop,gpt7)" has to be replaced by "linux (loop,7)".
so from
```
        linux (loop,gpt7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
                cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path
        initrd (loop,gpt7)/initramfs.img
```
to
```
        linux (loop,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
                cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path
        initrd (loop,7)/initramfs.img
```

If it does not work, could you press "c" at your grub screen, then run "loopback loop (hd0,msdos4)/chromeos.img" and post the result of "ls" command ?

darethehair, 2020-05-13:
Changing 'loop,gpt7' references to 'loop,7' still results in the same message of 'no such partition'

Running the 'loopback' line that you suggested results in no messages at all (good or bad).  Following that by 'ls' results in (for me):
```
(loop) (hd0) (hd0,msdos4) (hd0,msdos3) (hd0, msdos2) (hd0, msdos1)
```

sebanc, 2020-05-13:
Thanks for trying it, the issue is that the loop device is created without partitions detected. I removed in  recent releases a gpt/mbr compatibility features which I considered unnecessary as mbr was not supported, maybe this method would work with old releases which still had this compatibility feature.

Could you try the same with "Brunch r80 k4.19 stable 20200329" release ?

sebanc, 2020-05-13:
If it works I will re-add this feature

darethehair, 2020-05-13:
Hmmm...unless I have made a mistake, it looks like no change at all:

```
darren@asusk501:~/Downloads$ tar zxvf brunch_r80_k4.19_stable_20200329.tar.gz
chromeos-install.sh
efi_legacy.img
efi_secure.img
rootc.img
darren@asusk501:~/Downloads$ mkdir -p ~/tmpmount
darren@asusk501:~/Downloads$ sudo mount /dev/sda4 ~/tmpmount
darren@asusk501:~/Downloads$ sudo bash chromeos-install.sh -src chromeos_12739.111.0_samus_recovery_stable-channel_mp-v3.bin -dst ~/tmpmount/chromeos.img -s 100
Creating image file
0+0 records in
0+0 records out
0 bytes copied, 0.00016926 s, 0.0 kB/s
WARNING: Primary GPT header is invalid
WARNING: Secondary GPT header is invalid
4A40431B-5F5D-D542-81FC-B1B5D4384478
       start        size    part  contents
           0           1          PMBR (Boot GUID: 4A40431B-5F5D-D542-81FC-B1B5D4384478)
           1           1          Pri GPT header
           2          32          Pri GPT table
    19255304   190459848       1  Label: "STATE"
                                  Type: Linux data
                                  UUID: 99D96750-38C5-2647-A8AF-A48DA0D2FC4B
     2117638       65536       2  Label: "KERN-A"
                                  Type: ChromeOS kernel
                                  UUID: 1FECBDEF-656F-7B47-953E-799F3F5F1CDE
                                  Attr: priority=15 tries=15 successful=0 
    10866696     8388608       3  Label: "ROOT-A"
                                  Type: ChromeOS rootfs
                                  UUID: 6ED16807-1E00-0344-AF0C-2EDF043DDCCC
     2183174       65536       4  Label: "KERN-B"
                                  Type: ChromeOS kernel
                                  UUID: C277F482-D6E1-9F42-A119-4388500983EC
                                  Attr: priority=0 tries=15 successful=0 
     2478088     8388608       5  Label: "ROOT-B"
                                  Type: ChromeOS rootfs
                                  UUID: 28397AB7-6E65-C149-8670-E564848101FE
       16448           1       6  Label: "KERN-C"
                                  Type: ChromeOS kernel
                                  UUID: D0F75C86-8D6E-3B4D-9FDB-FD88FE062D85
                                  Attr: priority=0 tries=15 successful=0 
       16456     2097152       7  Label: "ROOT-C"
                                  Type: ChromeOS rootfs
                                  UUID: E262DBDE-2BAE-6A40-A078-0F9999F360FE
     2248712       32768       8  Label: "OEM"
                                  Type: Linux data
                                  UUID: FCAF8430-FF05-224D-B3CA-0F1DC79678D3
     2113608           1       9  Label: "reserved"
                                  Type: ChromeOS reserved
                                  UUID: 03E84288-3334-8747-9FBE-9145F01F019A
     2113609           1      10  Label: "reserved"
                                  Type: ChromeOS reserved
                                  UUID: F08E07C6-70A2-CE48-8928-E8301FB829EA
          64       16384      11  Label: "RWFW"
                                  Type: ChromeOS firmware
                                  UUID: 966A2396-763A-3742-B1F0-81B5C3B8C6B9
     2412552       65536      12  Label: "EFI-SYSTEM"
                                  Type: EFI System Partition
                                  UUID: 4A40431B-5F5D-D542-81FC-B1B5D4384478
   209715167          32          Sec GPT table
   209715199           1          Sec GPT header
Writing partition 1
10.0MiB 0:00:00 [52.6MiB/s] [  <=>                                                                                                                                      ]
Writing partition 2
16.0MiB 0:00:00 [59.8MiB/s] [  <=>                                                                                                                                      ]
Writing partition 3
2.00GiB 0:01:16 [26.6MiB/s] [                                                              <=>                                                                          ]
Writing partition 4
[... 33 lines cut ...]
darren@asusk501:~/Downloads$ cat /etc/grub.d/40_custom 
#!/bin/sh
exec tail -n +3 $0
# This file provides an easy way to add custom menu entries.  Simply type the
# menu entries you want to add after this comment.  Be careful not to change
# the 'exec tail' line above.
menuentry "ChromeOS (boot from disk image)" {
	img_part=/dev/sda4
	img_path=/chromeos.img
	search --no-floppy --set=root --file $img_path
	loopback loop $img_path
	linux (loop,gpt7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
		cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path
	initrd (loop,gpt7)/initramfs.img
}

darren@asusk501:~/Downloads$ sudo update-grub
Generating grub configuration file ...
Found theme: /boot/grub/themes/linuxmint/theme.txt
Found linux image: /boot/vmlinuz-4.19.0-9-amd64
Found initrd image: /boot/initrd.img-4.19.0-9-amd64
Found linux image: /boot/vmlinuz-4.19.0-8-amd64
Found initrd image: /boot/initrd.img-4.19.0-8-amd64
done
```

sebanc, 2020-05-13:
It's normal for the output to be the same however the grub loop device should have shown partitions.

Sorry for making you waste time on this, I will try to get some info from the guy who managed this and will revert back.

darethehair, 2020-05-13:
I am only too glad to help with this, so I consider it time well spent!  If BIOS capability was made available for Brunch, it would greatly expand the usefulness and happiness of those using it! :)

sebanc, 2020-05-13:
Well you seem really motivated by this, the issue with hybrid boot is that I don't really understand it and I have no device which supports legacy boot.

If you want to try, I am attaching here an attempt at legacy boot which I cannot test and is unlikely to succeed.
[testmbr.zip](https://github.com/sebanc/brunch/files/4623641/testmbr.zip)

How does it work:
1) Extract the latest brunch release in a directory
2) Extract both files in the archive in the same directory (it will replace the original chromeos-install.sh and add a mbr.img)
3) add "-m" argument when you run the chromeos-install.sh script

Tell me what happens when you try to boot the usb (Do you see syslinux text ?, Do you see grub ?, Does it boot ?)

darethehair, 2020-05-13:
I did most of what you suggested, except I continued to install directly to my '/dev/sda4' partition as I have previously.  In your last statement, it sounds like you wanted (?) me to install to a USB stick instead?  In any case, with what I did, no difference that I can see -- even if the installer produces new 'MBR-style' messages:
```
darren@asusk501:~/Downloads$ tar zxvf brunch_r81_k4.19_stable_20200510.tar.gz
chromeos-install.sh
efi_legacy.img
efi_secure.img
rootc.img
darren@asusk501:~/Downloads$ unzip testmbr.zip 
Archive:  testmbr.zip
replace chromeos-install.sh? [y]es, [n]o, [A]ll, [N]one, [r]ename: y
  inflating: chromeos-install.sh     
  inflating: mbr.img                 
darren@asusk501:~/Downloads$ mkdir -p ~/tmpmount
darren@asusk501:~/Downloads$ sudo mount /dev/sda4 ~/tmpmount
[sudo] password for darren:       
darren@asusk501:~/Downloads$ sudo bash chromeos-install.sh -m -src chromeos_12739.111.0_samus_recovery_stable-channel_mp-v3.bin -dst ~/tmpmount/chromeos.img -s 100
Creating image file
0+0 records in
0+0 records out
0 bytes copied, 0.00019892 s, 0.0 kB/s
WARNING: Primary GPT header is invalid
WARNING: Secondary GPT header is invalid
F93BF764-889A-8543-ADD7-6C20DD916345
       start        size    part  contents
           0           1          PMBR (Boot GUID: F93BF764-889A-8543-ADD7-6C20DD916345)
           1           1          Pri GPT header
           2          32          Pri GPT table
    19255304   190459848       1  Label: "STATE"
                                  Type: Linux data
                                  UUID: 9C3341C6-B849-6A42-A971-8E38770A5BB4
     2117638       65536       2  Label: "KERN-A"
                                  Type: ChromeOS kernel
                                  UUID: A6EFC688-14E6-BF41-A235-45373250BB30
                                  Attr: priority=15 tries=15 successful=0 
    10866696     8388608       3  Label: "ROOT-A"
                                  Type: ChromeOS rootfs
                                  UUID: 22FEBE61-4AA7-8C49-882E-C37A51C486FF
     2183174       65536       4  Label: "KERN-B"
                                  Type: ChromeOS kernel
                                  UUID: BD288B32-0A40-6745-BC5F-D69E9B5682A7
                                  Attr: priority=0 tries=15 successful=0 
     2478088     8388608       5  Label: "ROOT-B"
                                  Type: ChromeOS rootfs
                                  UUID: 62192269-77C1-744C-A338-5572C92E1F1D
       16448           1       6  Label: "KERN-C"
                                  Type: ChromeOS kernel
                                  UUID: 745933EC-1310-2D4E-9E39-F85B923E23D9
                                  Attr: priority=0 tries=15 successful=0 
       16456     2097152       7  Label: "ROOT-C"
                                  Type: ChromeOS rootfs
                                  UUID: 1AC61EFC-A414-954D-91C4-E3769BB7A879
     2248712       32768       8  Label: "OEM"
                                  Type: Linux data
                                  UUID: 65F3A569-C34E-6E4F-A2C6-649C9EDE7C8A
     2113608           1       9  Label: "reserved"
                                  Type: ChromeOS reserved
                                  UUID: 141B25AB-80FD-8C4C-8469-F108C68FD8A3
     2113609           1      10  Label: "reserved"
                                  Type: ChromeOS reserved
                                  UUID: 0AA85EED-E1C8-2D4B-A037-AF879395EDA7
          64       16384      11  Label: "RWFW"
                                  Type: ChromeOS firmware
                                  UUID: 6651D19B-29FF-944C-9288-9EF97A42EA6B
     2412552       65536      12  Label: "EFI-SYSTEM"
                                  Type: EFI System Partition
                                  UUID: F93BF764-889A-8543-ADD7-6C20DD916345
   209715167          32          Sec GPT table
   209715199           1          Sec GPT header
Writing partition 1
10.0MiB 0:00:00 [35.0MiB/s] [  <=>                                                                                                                                      ]
[... 74 lines cut ...]
To boot directly from this image file, add the lines between stars to either:
- A brunch usb flashdrive grub config file (then boot from usb and choose boot from disk image in the menu),
- Or your hard disk grub install if you have one (refer to you distro's online resources).
********************************************************************************
menuentry "ChromeOS (boot from disk image)" {
	img_part=/dev/sda4
	img_path=/chromeos.img
	search --no-floppy --set=root --file $img_path
	loopback loop $img_path
	linux (loop,gpt7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
		cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path
	initrd (loop,gpt7)/initramfs.img
}
********************************************************************************
darren@asusk501:~/Downloads$ sudo umount ~/tmpmount
darren@asusk501:~/Downloads$ sudo update-grub
Generating grub configuration file ...
Found theme: /boot/grub/themes/linuxmint/theme.txt
Found linux image: /boot/vmlinuz-4.19.0-9-amd64
Found initrd image: /boot/initrd.img-4.19.0-9-amd64
Found linux image: /boot/vmlinuz-4.19.0-8-amd64
Found initrd image: /boot/initrd.img-4.19.0-8-amd64
done

```

sebanc, 2020-05-13:
Sorry, I should have been clearer, this one is an attempt for full mbr support, could you try it on usb ? dual boot will come after if it works

darethehair, 2020-05-13:
When the install is directed to a USB stick, I get some messages that I don't like, but it completes OK.  The USB stick is a valid boot option, but when selected just displays a blinking underline cursor :(
```
darren@asusk501:~/Downloads$ time sudo bash chromeos-install.sh -m -src chromeos_12739.111.0_samus_recovery_stable-channel_mp-v3.bin -dst /dev/sdb
[sudo] password for darren:       
All data on device /dev/sdb will be lost, are you sure ? (type yes to continue) yes
umount: /dev/sdb: not mounted.
umount: /dev/sdb1: not mounted.
umount: /dev/sdb10: not mounted.
umount: /dev/sdb11: not mounted.
umount: /dev/sdb12: not mounted.
umount: /dev/sdb2: not mounted.
umount: /dev/sdb3: not mounted.
umount: /dev/sdb4: not mounted.
umount: /dev/sdb5: not mounted.
umount: /dev/sdb6: not mounted.
umount: /dev/sdb7: not mounted.
umount: /dev/sdb8: not mounted.
umount: /dev/sdb9: not mounted.
08E28634-1005-3A4B-B8BD-F7C133A835BF
       start        size    part  contents
           0           1          PMBR (Boot GUID: 08E28634-1005-3A4B-B8BD-F7C133A835BF)
           1           1          Pri GPT header
           2          32          Pri GPT table
    19255304    11022280       1  Label: "STATE"
                                  Type: Linux data
                                  UUID: 64960A13-B795-1048-A831-1A57ADE08A3B
     2117638       65536       2  Label: "KERN-A"
                                  Type: ChromeOS kernel
                                  UUID: 8CACEE31-E4B9-6740-A4AB-1EDF0676D568
                                  Attr: priority=15 tries=15 successful=0 
    10866696     8388608       3  Label: "ROOT-A"
                                  Type: ChromeOS rootfs
                                  UUID: B221EAAF-B8E8-D94F-B6F2-E4BB241764F2
     2183174       65536       4  Label: "KERN-B"
                                  Type: ChromeOS kernel
                                  UUID: E15C7E24-2AD2-F941-ADC5-9FC57AF52A89
                                  Attr: priority=0 tries=15 successful=0 
     2478088     8388608       5  Label: "ROOT-B"
                                  Type: ChromeOS rootfs
                                  UUID: BA497BD7-DC06-F848-AD9E-B921DA45B1E2
       16448           1       6  Label: "KERN-C"
                                  Type: ChromeOS kernel
                                  UUID: 02B882E3-4846-5C44-B714-77AAE0F63631
                                  Attr: priority=0 tries=15 successful=0 
       16456     2097152       7  Label: "ROOT-C"
                                  Type: ChromeOS rootfs
                                  UUID: ABB066C4-4D73-C843-90B4-53C5C06A321F
     2248712       32768       8  Label: "OEM"
                                  Type: Linux data
                                  UUID: 97D18040-3CD5-9946-88A5-0C41527B9B6E
     2113608           1       9  Label: "reserved"
                                  Type: ChromeOS reserved
                                  UUID: 5B781A21-7BF0-7B4F-9B32-00BB564F569D
     2113609           1      10  Label: "reserved"
                                  Type: ChromeOS reserved
                                  UUID: AE394D83-CDAE-134C-AD2C-3A374DAE67B4
          64       16384      11  Label: "RWFW"
                                  Type: ChromeOS firmware
                                  UUID: EB5C4CF2-5731-3146-AB2B-917C067B44D7
     2412552       65536      12  Label: "EFI-SYSTEM"
                                  Type: EFI System Partition
                                  UUID: 08E28634-1005-3A4B-B8BD-F7C133A835BF
    30277599          32          Sec GPT table
    30277631           1          Sec GPT header
Writing partition 1
10.0MiB 0:00:00 [57.8MiB/s] [=========================================================================================================================>] 100%            
Writing partition 2
16.0MiB 0:00:00 [56.8MiB/s] [=========================================================================================================================>] 100%            
Writing partition 3
2.00GiB 0:03:32 [9.65MiB/s] [=========================================================================================================================>] 100%            
[... 7 lines cut ...]
1.00GiB 0:01:10 [14.5MiB/s] [=========================================================================================================================>] 100%            
Writing partition 8
16.0MiB 0:00:00 [67.8MiB/s] [=========================================================================================================================>] 100%            
Writing partition 9
 512 B 0:00:00 [2.17MiB/s] [==========================================================================================================================>] 100%            
Writing partition 10
 512 B 0:00:00 [2.38MiB/s] [==========================================================================================================================>] 100%            
Writing partition 11
8.00MiB 0:00:00 [53.0MiB/s] [=========================================================================================================================>] 100%            
Writing partition 12
08E28634-1005-3A4B-B8BD-F7C133A835BF
Checking that no-one is using this disk right now ... FAILED

This disk is currently in use - repartitioning is probably a bad idea.
Umount all file systems, and swapoff all swap partitions on this disk.
Use the --no-reread flag to suppress this check.

sfdisk: Use the --force flag to overrule all checks.
32.0MiB 0:00:00 [67.0MiB/s] [=========================================================================================================================>] 100%            
ChromeOS installed.

real	16m4.476s
user	0m0.409s
sys	0m17.198s
```

sebanc, 2020-05-14:
Thanks for trying and for the detailled report :)

Could you give this one a try ? (using the exact same procedure)
[testmbr2.zip](https://github.com/sebanc/brunch/files/4626715/testmbr2.zip)

darethehair, 2020-05-14:
Upon reboot, still a blinking underline cursor :(

```
darren@asusk501:~/Downloads$ unzip testmbr2.zip
Archive:  testmbr2.zip
replace chromeos-install.sh? [y]es, [n]o, [A]ll, [N]one, [r]ename: A
  inflating: chromeos-install.sh     
  inflating: mbr.img                 
darren@asusk501:~/Downloads$ time sudo bash chromeos-install.sh -m -src chromeos_12739.111.0_samus_recovery_stable-channel_mp-v3.bin -dst /dev/sdb
All data on device /dev/sdb will be lost, are you sure ? (type yes to continue) yes
umount: /dev/sdb: not mounted.
umount: /dev/sdb10: not mounted.
umount: /dev/sdb11: not mounted.
umount: /dev/sdb12: not mounted.
umount: /dev/sdb2: not mounted.
umount: /dev/sdb3: not mounted.
umount: /dev/sdb4: not mounted.
umount: /dev/sdb5: not mounted.
umount: /dev/sdb6: not mounted.
umount: /dev/sdb9: not mounted.
BCFDDE5A-5912-E74B-85D9-BDD4815AE273
Checking that no-one is using this disk right now ... OK

Disk /dev/sdb: 14.4 GiB, 15502147584 bytes, 30277632 sectors
Disk model: DataTraveler 3.0
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 0A9792D3-3AC3-F340-B541-BACCCDE3E63D

Old situation:

Device        Start      End  Sectors  Size Type
/dev/sdb1  19255304 30277583 11022280  5.3G Microsoft basic data
/dev/sdb2   2117638  2183173    65536   32M ChromeOS kernel
/dev/sdb3  10866696 19255303  8388608    4G ChromeOS root fs
/dev/sdb4   2183174  2248709    65536   32M ChromeOS kernel
/dev/sdb5   2478088 10866695  8388608    4G ChromeOS root fs
/dev/sdb6     16448    16448        1  512B ChromeOS kernel
/dev/sdb7     16456  2113607  2097152    1G ChromeOS root fs
/dev/sdb8   2248712  2281479    32768   16M Microsoft basic data
/dev/sdb9   2113608  2113608        1  512B ChromeOS reserved
/dev/sdb10  2113609  2113609        1  512B ChromeOS reserved
/dev/sdb11       64    16447    16384    8M unknown
/dev/sdb12  2412552  2478087    65536   32M EFI System

Partition table entries are not in disk order.

>>> Script header accepted.
>>> Created a new DOS disklabel with disk identifier 0xf3e56cff.
The old gpt signature may remain on the device. It is recommended to wipe the device with wipefs(8) or sfdisk --wipe, in order to avoid possible collisions.

/dev/sdb1: Created a new partition 1 of type 'W95 FAT32 (LBA)' and of size 32 MiB.
Partition #1 contains a vfat signature.
/dev/sdb2: Created a new partition 2 of type 'GPT' and of size 512 B.
/dev/sdb3: Done.

New situation:
Disklabel type: dos
Disk identifier: 0xf3e56cff

Device     Boot   Start     End Sectors  Size Id Type
/dev/sdb1  *    2412552 2478087   65536   32M  c W95 FAT32 (LBA)
/dev/sdb2             1       1       1  512B ee GPT

Partition table entries are not in disk order.

The partition table has been altered.
Calling ioctl() to re-read partition table.
[... 52 lines cut ...]
2.00GiB 0:03:42 [9.19MiB/s] [=========================================================================================================================>] 100%            
Writing partition 4
16.0MiB 0:00:00 [33.5MiB/s] [=========================================================================================================================>] 100%            
Writing partition 5
2.00GiB 0:03:40 [9.30MiB/s] [=========================================================================================================================>] 100%            
Writing partition 6
 512 B 0:00:00 [2.57MiB/s] [==========================================================================================================================>] 100%            
Writing partition 7
1.00GiB 0:01:23 [12.2MiB/s] [=========================================================================================================================>] 100%            
Writing partition 8
16.0MiB 0:00:00 [57.2MiB/s] [=========================================================================================================================>] 100%            
Writing partition 9
 512 B 0:00:00 [2.41MiB/s] [==========================================================================================================================>] 100%            
Writing partition 10
 512 B 0:00:00 [ 946KiB/s] [==========================================================================================================================>] 100%            
Writing partition 11
8.00MiB 0:00:00 [54.3MiB/s] [=========================================================================================================================>] 100%            
Writing partition 12
32.0MiB 0:00:00 [71.5MiB/s] [=========================================================================================================================>] 100%            
ChromeOS installed.

real	15m0.871s
user	0m0.471s
sys	0m16.536s
```

Inotamira, 2020-05-15:
So, I'm going to partially weigh in on this, because I'm having issues getting it to boot, but in a completely different way. What it sound like you're trying to do, is use Grub2 as a pass-through, which is completely doable, I've got it half working.

The trouble seems to come from the weird method that the default ChromeOS (or Brunch?) looks for drives, as default grub seems to have no idea what you're talking about. I'm using Coreboot, and by simply modifying the grub.cfg on (ahci0,gpt12) to point to (ahci0,gpt7) instead of that weird $drive,7 thing you've got going on, it starts booting.

My issue is that it then fails to make some several dozen directories and falls into a "self_repair" mode that does nothing. Hopefully that'll help with the Grub bootstrapping.

sebanc, 2020-05-15:
@darethehair Could you try the below package when you have the time ?
[testmbr3.zip](https://github.com/sebanc/brunch/files/4632813/testmbr3.zip)

@Inotamira Could you provide some background on what you are attempting ? Do you use a normal brunch release ? are you using one of the package I posted here ? Are you trying to boot via single or dual boot procedure ?

Inotamira, 2020-05-15:
> @Inotamira Could you provide some context on what you are attempting ? Do you use a normal brunch release ? are you using one of the package I posted here ? Are you trying to boot via single or dual boot procedure ?

So, it's a bit complicated, but I'm using Coreboot with the Grub payload. Basically I'm booting it from grub and nothing else, no BIOS, no UEFI, and that's over simplifying, but close enough. I'm single booting, installed from mint as recommended in a guide. Installs fine, I load it up through grub, but I have to modify the grub.cfg in partition 12 for my Grub bootloader to start the system.

The system seems to start doing it's thing, tries to make some directories, it flies past really fast, but fails to do so, and from what I can tell, it's because the partitions aren't where it expects them to be. I'm not sure if that's because Grub2 doesn't understand the regex command, which is why I have to edit the boot parameters every time, or simply because they aren't names in a typical way that a UEFI system would do it.

sebanc, 2020-05-15:
@Inotamira As far as I understand your device can use gpt partitions which should be enough for brunch to boot. I think you have 2 separate issues:
- The brunch grub config is not working with your coreboot grub payload: brunch is supposed to be booted using grubx64.efi on partition 12, with your specific setup you have to adapt grub as you properly did.
- brunch will not boot on your device: Probably another issue related to your hardware.
I don't mind trying to help you but please create another issue explaining exactly your hardware and what you are trying to do as it seems not related to this one.

B3ATDROP3R, 2020-05-15:
I know i'm not the same person that was helping before but when I ran the latest zip you provided I got a blinking cursor. I have also tried booting it using PLOP boot loader but it gives the same result.

sebanc, 2020-05-15:
@B3ATDROP3R Thanks for giving it a try. I will try to think of something else

sebanc, 2020-05-15:
Well, actually I have just seen in issue #143 that old processors like intel core 2 duo do not have the necessary cpu instructions to boot a recent chromeos image even if legacy boot would be implemented.
I will add to the main page that "devices with Intel core 2 duo and below are not supported".

My question would now be, are there devices without efi boot which could potentially boot chromeos ?

B3ATDROP3R, 2020-05-15:
The device I was attempting to use is a Dell studio 1569 with an Intel i5 430m or something like that. It is some time of i5 processor so idk if that would work or not but maybe.

sebanc, 2020-05-15:
yes that one would probably work

B3ATDROP3R, 2020-05-15:
I think I've gotten fydeos + chromify to work on that system before but I'm not entirely sure. Also I'm booting from a sata ssd if that matters too.

darethehair, 2020-05-15:
> @darethehair Could you try the below package when you have the time ?
> [testmbr3.zip](https://github.com/sebanc/brunch/files/4632813/testmbr3.zip)
> 
Sorry, as B3ATDROP3R mentioned, the latest attempt just resulted in the same flashing underline cursor...

sebanc, 2020-05-15:
I will keep trying other things.

B3ATDROP3R, 2020-05-15:
I have an idea that might work to make it boot on legacy/mbr. I am going to try it then report back.

B3ATDROP3R, 2020-05-15:
It isn't even getting to boot grub. I tried even using the one from fydeos where its supposed to show a menu and i still get a blinking cursor. I think it has something to do with it being a GPT partition table and not MBR. ZI am gonna try some stuff and see what happens.

B3ATDROP3R, 2020-05-16:
This is as far as I have been able to get. 
[image]

sebanc, 2020-05-16:
Today's attempt::
[testmbr4.zip](https://github.com/sebanc/brunch/files/4638049/testmbr4.zip)

There is an additional file but the process remains exactly the same.

darethehair, 2020-05-16:
> Today's attempt::
> [testmbr4.zip](https://github.com/sebanc/brunch/files/4638049/testmbr4.zip)
> 
> There is an additional file but the process remains exactly the same.

A bit of progress!  Got the SYSLINUX boot prompt:

```
SYSLINUX 6.03 EDD 2014-10-06 Copyright (C) 1994-2014 H. Peter Anvin et al
boot:
```
Not sure what else I could try/type at this point.  I think this means that no config file is present?

sebanc, 2020-05-16:
That's a start :)

I think it does not like my syslinux config, could you mount partition 12 and replace /syslinus/syslinux.cfg with the below text ?

```
PROMPT 1
TIMEOUT 10

label grub2lnx
  menu label Grub2
  kernel lnxboot.img
  append initrd=core.img
```

darethehair, 2020-05-16:
More activity! This is what I did:
```
darren@frodo3:/usr/local/bin$ sudo mkdir /mnt/sdc12
darren@frodo3:/usr/local/bin$ sudo mount /dev/sdc12 /mnt/sdc12
darren@frodo3:/usr/local/bin$ cd /mnt/sdc12
darren@frodo3:/mnt/sdc12$ ls
efi  syslinux
darren@frodo3:/mnt/sdc12$ cd syslinux
darren@frodo3:/mnt/sdc12/syslinux$ ls
core.img  default.cfg  grub.cfg  ldlinux.c32  ldlinux.sys  lnxboot.img  README  root.A.cfg  root.B.cfg  syslinux.cfg  usb.A.cfg  vmlinuz.A  vmlinuz.B
darren@frodo3:/mnt/sdc12/syslinux$ sudo vi syslinux.cfg
darren@frodo3:/mnt/sdc12/syslinux$ cat syslinux.cfg
PROMPT 1
TIMEOUT 10

label grub2lnx
  menu label Grub2
  kernel lnxboot.img
  append initrd=core.img
darren@frodo3:/mnt/sdc12/syslinux$ cd
darren@frodo3:~$ sudo umount /dev/sdc12
```
This was the result on booting:
```
boot:
No DEFAULT or UI configuration directive found!
```

(infinite loop of this message)

sebanc, 2020-05-16:
Try this config:
```
PROMPT 1
TIMEOUT 10
DEFAULT Grub2

label Grub2
  menu label Grub2
  kernel lnxboot.img
  append initrd=core.img
```

darethehair, 2020-05-16:
Interesting!  Now I get a grub prompt!

`grub>`

sebanc, 2020-05-16:
Try to modify /syslinux/grub.cfg with:
```
configfile $root/efi/boot/grub.cfg
```

darethehair, 2020-05-16:
No change.  Still just the prompt.  if I type 'ls' I just get:

`(proc)`

sebanc, 2020-05-16:
Thanks it means that grub is not properly loaded. I have to think about it but it is clearly good progress already.

sebanc, 2020-05-16:
New test image:
[testmbr6.zip](https://github.com/sebanc/brunch/files/4638552/testmbr6.zip)

Added a missing module from grub

darethehair, 2020-05-16:
Getting closer!  Still just a grub prompt, but now 'ls' shows a lots of partitions:

```
grub> ls

(proc) (hd0) (hd0,gpt12) (hd0,gpt11) (hd0,gpt10) (hd0,gpt9) (hd0,gpt8) (hd0,gpt7) (hd0,gpt6) (hd0,gpt5) (hd0,gpt4) (hd0,gpt3) (hd0,gpt2) (hd0,gpt1) (hd1) (dh1,msdos4) (hd1,msdos3) (hd1,msdos2) (hd1,msdos1)
```
Yes, I typed this one at a time :)

sebanc, 2020-05-16:
> Try to modify /syslinux/grub.cfg with:
> 
> ```
> configfile $root/efi/boot/grub.cfg
> ```

Try the above

darethehair, 2020-05-16:
Just the same grub prompt.  Something flashed by but I have no idea what it said.  Not sure if it was important.

sebanc, 2020-05-16:
Well, for now let's hardcode it, we will see that later:
```
configfile (hd0,gpt12)/efi/boot/grub.cfg
```

sebanc, 2020-05-16:
Actually could you confirm if you have /efi/boot/grub.cfg, I think I forgot to include it...

sebanc, 2020-05-16:
Nevermind, it is there, you can try:

> Well, for now let's hardcode it, we will see that later:
> 
> ```
> configfile (hd0,gpt12)/efi/boot/grub.cfg
> ```

darethehair, 2020-05-16:
Nope :( But I can confirm that the location you mention seems valid:

```
grub> ls (hd0,gpt12)/efi/boot
grub.cfg
```
It almost looks like it is ignoring your config file (?).

sebanc, 2020-05-16:
Could you try to type in directly "configfile (hd0,gpt12)/efi/boot/grub.cfg" ?

darethehair, 2020-05-16:
It reacts!  I just the familiar grub "*ChromeOS" boot entry!  However, when I select it, I get:

```
error: disk `,7' not found
error: you need to load the kernel first

Press any key to continue...
```

darethehair, 2020-05-16:
I temporarily overrode the grub boot lines and changed "$disk" to "hd0", and Brunch started to boot!  However it ran into a problem:

`brunch:  Chromeos was not found on any device. fallback to shell..`

sebanc, 2020-05-16:
Ok so 2 issues:
1) I don't know where grub expect the default config to be
2) There is the need for a different boot script

Nevertheless, if you want, you can copy the below text in /syslinux/grub.cfg and use "configfile grub.cfg" when you see grub, it should attempt to boot:
```
set timeout=2

menuentry "ChromeOS" {
  linux (hd0,gpt7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug
  initrd (hd0,gpt7)/initramfs.img
}
```

sebanc, 2020-05-16:
Ah sorry missed your previous comment, could you post a short video of the boot process ?

darethehair, 2020-05-16:
I hope it is OK, but rather than do exactly what you suggested, I repeated what I had already done (i.e. edited the grub command line to replace $disk).  I used my camera to record, converted the MOV file to AVI, and uploaded to Dropbox:

https://www.dropbox.com/s/n3xm76xtol47naw/brunch_bios.avi?dl=0

[60 more replies]

closed 2020-06-02 by sebanc