sebanc/brunch #78: Camera not working - MacBookAir 5, 2

Comstock2K, 2020-04-28:
Hi,

Installed latest build r81 on MacBookAir 5, 2. Everything is working fine apart from the camera.

dmesg is reporting the following

[    1.383229] uvcvideo: Found UVC 1.00 device FaceTime HD Camera (Built-in) (05ac:8510)
[    1.428276] uvcvideo 1-1.1:1.0: Entity type for entity Processing 3 was not initialized!
[    1.432055] uvcvideo 1-1.1:1.0: Entity type for entity Camera 1 was not initialized!
[    1.434506] input: FaceTime HD Camera (Built-in):  as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.1/1-1.1:1.0/input/input4

usb-devices is showing the following

T:  Bus=01 Lev=02 Prnt=02 Port=00 Cnt=01 Dev#=  3 Spd=480 MxCh= 0
D:  Ver= 2.00 Cls=ef(misc ) Sub=02 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=05ac ProdID=8510 Rev=80.25
S:  Manufacturer=Apple Inc.
S:  Product=FaceTime HD Camera (Built-in)
S:  SerialNumber=CCGC7E04Q2DWC8C0
C:  #Ifs= 3 Cfg#= 1 Atr=80 MxPwr=500mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=0e(video) Sub=01 Prot=00 Driver=uvcvideo
I:  If#= 1 Alt= 0 #EPs= 0 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo
I:  If#= 2 Alt= 0 #EPs= 0 Cls=ff(vend.) Sub=00 Prot=00 Driver=(none)

Any idea to get it running?

Thank you for your dedication so far!

Attached dmesg
[dmesg.txt](https://github.com/sebanc/brunch/files/4545257/dmesg.txt)

Comstock2K, 2020-04-29:
Camera seems to be working on samus recovery.

sebanc, 2020-05-03:
Hi, just to say that i am working on the camera issue but have not found a proper fix for new recovery images yet.

sebanc, 2020-06-26:
The camera issue has been included in #331 "Main issues on which help is needed". I am therefore closing individual issues related to camera not working with rammus recovery.

closed 2020-06-26 by sebanc