sebanc/brunch #79: Drawing problems on boot

F1F7Y, 2020-04-28:
[image]
When booting it draws this (image), from multiple attempts I 100% know that this all is text. The image was taken on the first boot when it was building, yet it does this on every boot before it shows the 'chrome' splashscreen.

I'm running rammus v80 and the latest brunch v80 testing. My notebook is the HP ProBook 430 g2. The same happened on the stable version of brunch v80.
A note-worthy thing is also that secureboot doesn't like the certificate and so it is turned OFF, I never was able to boot with secureboot and so I don't know if this screen is drawn with with it.

My notebook doesn't have a fullHD (it's 1366 x 768) screen wich I feel like is causing it to draw like this.

sebanc, 2020-04-28:
Hi,

Could you try to add `vt.handoff=1` to the kernel command line (after cros_debug) by running "sudo edit-grub-config" ?

F1F7Y, 2020-04-28:
[image]
This is how the edited file looks like, just making sure I did what you wanted me to try. It didn't help, the messed up text still shows as normal.

juicypop, 2020-04-28:
You could try "nomodeset" in the kernel line

Actually i didn't tried this yet on Chrome os but it solved kind of these glitches in the past for me

lfom, 2020-04-28:
If you see the GRUB menu normally, then you can try to add this line just below "ChromeOS":
`set gfxpayload=keep`

F1F7Y, 2020-04-29:
[image]
I've tried both `nomodeset` (my notebook wasn't able to boot with this) and `set gfxpayload=keep` yet the text is still messed up, my question is if there is a way to tell the bootloader the screens resolution, maybe that could fix the problem.
I'm gonna try and educate myself on this matter, thanks for the replies, if you have any other suggestions please post them here, thanks.

F1F7Y, 2020-04-29:
Also something important, the GRUB menu desplays normally, it's the section between the grub menu and chrome os splashscreen.

lfom, 2020-04-29:
Then the video mode isn't set correctly in ChromeOS... I don't know if it would work, but you can manually set the video mode adding something like `video=1024x768` to the kernel line.

sebanc, 2020-04-30:
Could you also try adding "vga=normal" to the kernel command line ?

F1F7Y, 2020-04-30:
[image]
`vga=normal` didn't do anything.

F1F7Y, 2020-04-30:
After searching aroud I found a comment made by @sebanc in an issue around google play and android apps not working suggesting to add `console=` in the kernel line after cross_debug wich worked! There is still a distorted `_` blinking, but at least I won't get people staring and making comments about it. Thanks to everyone who made suggestions.

closed 2020-04-30 by F1F7Y

sebanc, 2020-05-03:
FYI, someone with the same issue as yours (actually an HP Probook also) solved it by going to the bios and setting Advanced > boot options > boot mode > UEFI Hybrid (with CSM).

Hope it helps