sebanc/brunch #80: Incompatible with Dev version 84

EstPC13, 2020-04-28:
Hey! Just wanted to know that installing the new dev update leads to a boot loop, so anyone on the dev channel should switch to beta or stable.

Beta v83 works fine, though.

Just saving everyone the trouble of having to reinstall.

jhedfors, 2020-04-29:
It must be device dependant, as it was working just fine for me, with exception of some freezing/studdering that I was requently experiencing with video playback.  I am back on Beta myself due to that issue.

Comstock2K, 2020-04-29:
I had the same issue as OP, boot loop. Will wait for updated stable.
Anybody know what would be the issue if we update ChromeOS without Brunch framework?

sebanc, 2020-04-29:
Hi,

I think it is related to a change google recently made which impacted many recovery images.

@EstPC13 and @Comstock2K, could you post your laptop models ? (specifically the CPU will be important)

Comstock2K, 2020-04-29:
Laptop model Apple MacBookAir 5,2
`CPU0: Intel(R) Core(TM) i5-3427U CPU @ 1.80GHz (family: 0x6, model: 0x3a, stepping: 0x9)`

sebanc, 2020-04-29:
Thanks, that confirms what I thought:
Google has recently added new cpu instructions (Movbe instructions if I am not mistaken) to the rammus devices and many other boards which are now therefore only supported starting from Intel Gen 4 devices.

For users with CPU < Intel Gen 4, the new recommended recovery image is therefore "samus".

Note: a known issue with "samus" is that you have to be in guest mode to open "chrome://flags"

sebanc, 2020-04-29:
Could you both check that you can update to dev when using samus recovery ?

Comstock2K, 2020-04-29:
Sorry, but do I have to reinstall it again from scratch or can I do it from current rammus recovery?

sebanc, 2020-04-29:
Download the "samus" recovery and extract it.

Run "sudo bash chromeos-update -r <path_to_the_extracted_samus_recovery>"

Comstock2K, 2020-04-29:
Thanks.
Updated to samus, however the "Change Channel" is greyed out and thus am unable to change it.

sebanc, 2020-04-29:
Indeed, there is an ownership issue with samus (related to the use of TPM1.2), nevertheless you can switch channel with command line (`sudo update_engine_client --channel=dev-channel -update` if I remember correctly) and change flags from guest mode.

The choice is yours to stay on rammus but you won't be able to update it anymore.

Comstock2K, 2020-04-29:
You are correct @sebanc, I tried to update to beta on rammus, even that caused a boot loop.
Started from scratch, updated to samus, installed brunch r81, enrolled into beta (r83 - using the shell command) and all is working good. I don't have the owner issue either anymore, not sure why...
Even my camera is working now...!

Happy chappy!
Thanks for your help.

sebanc, 2020-04-29:
Ah I kind of remember now that updating solved this for me too at some point when I was testing recovery images.

Still, to be confirmed by others

sebanc, 2020-04-29:
Note: ownership (chrome flags and changing update channels notably) only work on dev channel currently.
To update to dev channel:
`sudo update_engine_client --channel=dev-channel -update`

EstPC13, 2020-05-01:
I’ve returned and done everything!

Latest dev update works with samus, but now I can’t sign in to my old username; is there any way to powerwash from Terminal?

My CPU is a Pentium NE56R, so not new at all.

Comstock2K, 2020-05-01:
Click on add new user, then add the **same** user details as what you are trying to login with, it will overwrite your current user. I had the same issue.

jhedfors, 2020-05-01:
Have you tried Alt+Ctrl+Shift+R, then tapping restart?

EstPC13, 2020-05-01:
I tried both of these things, and got the following results:

1. When adding the same details, it just infinitely loaded.
2. Holding the Powerwash command on restart did nothing, sadly.

I'm now setting it up again, fresh install built from samus, I'll let you know if I have any issues!

EstPC13, 2020-05-01:
I’ve know powerwashed my dev install, and I’m stuck at the “please wait” screen after signing into my Google account.

I have no idea what’s going on; dev mode is currently disabled on grub and updates and advanced brightness are enabled, it might be the first?

EstPC13, 2020-05-01:
EDIT: That was exactly it! Everything works smoothly after re-enabling cros_debug; let’s hope I can disable it once again once I set everything up (I use Netflix a lot).

closed 2020-06-02 by sebanc