sebanc/brunch #81: Kernel headers?

hacker1024, 2020-04-29:
Hi @sebanc,
I'm trying to build [this kernel module](https://github.com/rohitpid/Linux-Magic-Trackpad-2-Driver) to get my Apple Magic Mouse 2 working.
This is proving difficult, as I can't find your kernel headers.
When I use other headers, the module won't load: 
```
insmod: ERROR: could not insert module /lib/modules/4.19.98-brunch-sebanc/kernel/drivers/hid/hid-magicmouse.ko: Invalid module format
```
Which, as far as I can tell, is because I didn't use the right headers.

Would you mind sharing the kernel headers? TIA.

sebanc, 2020-04-29:
Hi,

Since a few versions ago, the kernel headers are stored under "/usr/src/linux-headers-<kernel_version>".

hacker1024, 2020-04-29:
Awesome! Thanks.

hacker1024, 2020-04-29:
I also need the .config file to build modules

sebanc, 2020-04-29:
Give me the link to your driver and i will try to include it directly in the kernel, it will be easier

sebanc, 2020-04-29:
Sorry did not remember that the link was already included, i am going to look at it

hacker1024, 2020-04-29:
Great, thanks.
I just built it myself against mainline 4.19.98 sources, but I get some errors upon loading:

```
$ sudo insmod hid-magicmouse.ko
insmod: ERROR: could not insert module hid-magicmouse.ko: Invalid module format

$ cat /var/log/messages
...
2020-04-29T22:40:39.482462+10:00 WARNING kernel: [30722.739228] hid_magicmouse: loading out-of-tree module taints kernel.
2020-04-29T22:40:39.482490+10:00 NOTICE kernel: [30722.739569] hid_magicmouse: module verification failed: signature and/or required key missing - tainting kernel
2020-04-29T22:40:39.482494+10:00 ERR kernel: [30722.739703] module: x86/modules: Skipping invalid relocation target, existing value is nonzero for type 1, loc 00000000936cbde6, val ffffffffc0b0442f

$ sudo cp hid-magicmouse.ko /lib/modules/4.19.98-brunch-sebanc/kernel/drivers/hid/hid-magicmouse.ko
$ sudo modprobe hid-magicmouse
insmod: ERROR: could not insert module hid-magicmouse.ko: Invalid module format
```

sebanc, 2020-05-03:
Hi, this driver has been included in the new testing release.

hacker1024, 2020-05-03:
Thanks, I'll check it out!

hacker1024, 2020-05-03:
Would upgrading brunch delete other non-brunch partitions on the drive? (I accept there's a chance that something could go wrong, but will it definitely?)

hacker1024, 2020-05-03:
The install script looks safe enough, I'll try it.

hacker1024, 2020-05-03:
I just tried it;
- [x] The module loads
- [x] The module appears to recognise the mouse
```
[   82.375198] Bluetooth: core.c:hidp_init() HIDP (Human Interface Emulation) ver 1.2
[   82.375211] Bluetooth: sock.c:hidp_init_sockets() HIDP socket layer initialized
[   82.420985] magicmouse 0005:004C:0269.0008: unknown main item tag 0x0
[   82.421042] input: hacker1024's Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.0/bluetooth/hci0/hci0:256/0005:004C:0269.0008/input/input27
[   82.421235] magicmouse 0005:004C:0269.0008: input,hidraw2: BLUETOOTH HID v1.00 Mouse [hacker1024's Mouse] on 98:3b:8f:81:83:40
```
- [ ] The mouse scrolling still doesn't work

If I read from `/dev/input/input20`, moving the mouse produces data, but swiping my finger to scroll does not. I don't know where to go from here in terms of debugging.

Thanks for all your work so far, @sebanc. I don't expect you to be able to do much more than that, as debugging would no doubt require specific hardware (and time!), and there are probably few people who use this mouse with brunch - it's probably not worth it.

sebanc, 2020-05-03:
Indeed, I will not be able to help more than this, from here you can try modifying /etc/gesture/40-touchpad-cmt.conf.

By the way, there is a specific config for apple magic mouse 1 in /etc/gesture/40-touchpad-cmt.conf, you might just need to change the device vendorid/productid to match yours.

If you manage to make it work, I can include the specific config.

hacker1024, 2020-05-03:
Is the `evdev` driver included in brunch? I think I need to tell the system to use that.

http://sneclacson.blogspot.com/2016/09/using-apple-magic-mouse-with-ubuntu-1604.html

sebanc, 2020-05-03:
No, you have to use the chromeos cmt driver, however the equivalent config to the one you point for cmt is in /etc/gesture/40-touchpad-cmt.conf.

sebanc, 2020-05-15:
My mistak, actually the evdev driver is included, you can try to change "cmt" to "evdev" in /etc/gesture/40-touchpad-cmt.conf

sebanc, 2020-05-18:
I have just found this repo from someone who ported the cmt driver to linux, it has a lot of cmt driver configuration exemples:
https://github.com/hugegreenbug/xf86-input-cmt/tree/master/xorg-conf

Hope it will help

sebanc, 2020-06-02:
I don't think I can help further on this issue so I will close it, don't hesitate to re-open it if needed

closed 2020-06-02 by sebanc