sebanc/brunch #82: Storage

abdouakil, 2020-04-29:
Good morning,
first of all thanks for your hard work, I succesfully triple booted windows, linux and chrome OS. While instaling, I gave chrome OS 100Gb of my hdd using gparted, but after installing it here is in the screenshot down how much storage it says I have. 
[image]
Thank you.

treypop, 2020-04-29:
How did you install Chrome OS to the hdd?

abdouakil, 2020-04-29:
> How did you install Chrome OS to the hdd?

i have windows in my ssd, linux and chrome os in my hdd
i followed this tutorial:
https://www.youtube.com/watch?v=FEY9MtznYGU&list=WL&index=24&t=0s

lfom, 2020-04-29:
AFAIK brunch creates a image to boot from the internal disk, not actual partitions as it does when installing to an external USB drive, because ChromeOS partitioning is very complicated, what makes multiple boot from the same disk very hard. So you are seeing the space related to this image, and not the partition. You must modify the install script to make the partition bigger.

PS: be very careful when you download such scripts without a direct link to the source so you can check what is does...

sebanc, 2020-04-30:
Hi, the solution is to remove the chromeos.img you have created through the script and recreate it using the official instructions (with the -s option to specify the image size).

I do not support the script you have used, therefore this issue should be opened on the github repository of the guy who made this script.

closed 2020-04-30 by sebanc

jhedfors, 2020-04-30:
To add to what @Ifom said... While I think that while the YouTube poster is genuinely trying to help -creating his own install script outside of the official process has caused a lot of confusion...

abdouakil, 2020-04-30:
Thank you guys for your replies, I appreciate your hard work. I thought the youtube guy had something to do with the owner that why I used his script. Anyways thanks for brining ChromeOS to our laptops, going to remove the chromeos.img and follow the official instructions.