sebanc/brunch #83: docker error

titeya, 2020-04-29:
Hi!
Docker on linux doesn't want start containers :
`Error response from daemon: OCI runtime create failed: container_linux.go:349: starting container process caused "exec: \"/portainer\": stat /portainer: no such file or directory": unknown`

Docker info return :
`WARNING: No swap limit support`

I tried to add the following key in grub.conf but without success:
> cgroup_enable=memory swapaccount=1

Any idea?
TIA

titeya, 2020-05-01:
Salut @Seban,

Your kernel doesn't include swap management (CONFIG_MEMCG_SWAP).
After activating and recompiling the kernel with this option, docker works correctly.

`sudo modprobe configs && zgrep MEMCG /proc/config.gz`

`CONFIG_MEMCG=y`
`CONFIG_MEMCG_SWAP=y`
`CONFIG_MEMCG_SWAP_ENABLED=y`
`CONFIG_MEMCG_KMEM=y`

Can you include this option in your next release ?

sebanc, 2020-05-01:
Salut @titeya 

I have completely reviewed my kernel config for the next release and I can confirm those configs are all activated.

I still need to do a bit of testing before releasing it but it should not take very long.

closed 2020-05-01 by titeya

sebanc, 2020-05-03:
Salut @titeya, je viens de publier la nouvelle version qui devrait résoudre le problème avec Docker.

titeya, 2020-05-03:
Je confirme, problème résolu 👍 
Merci pour cette nouvelle release en tout cas !

robertoandrade, 2022-08-11:
getting a similar error running docker within crostini on brunch/chromeos 103 with 5.10 kernel:

```
docker: Error response from daemon: failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: error setting cgroup config for procHooks process: bpf_prog_query(BPF_CGROUP_DEVICE) failed: function not implemented: unknown.
ERRO[0000] error waiting for container: context canceled
```
any ideas on how to get around it?