sebanc/brunch #84: Touchpad

abdouakil, 2020-04-29:
Hi,
So I'm using an HP probook 430 G5, i7 8550u and I cant use the touchpad, it only works when I click (I cant move the cursor), I have to sign out and then log in again to make it work. Reboot does not make it work.

abdouakil, 2020-05-01:
Update: I tried the r80 and r81 builds and none of them fixed my problem.

sebanc, 2020-05-01:
Hi,

Could you attach a dmesg to this issue ?

To get the dmesg, run "dmesg > ~/Downloads/dmesg.txt" in crosh shell and attach dmesg.txt from your downloads folder to this issue.

abdouakil, 2020-05-01:
Hello,
this is a dmesg i got running on r80 build.
(running r80 build because i got the issue on the r81 build has the same problem).
Thanks
[dmesg.txt](https://github.com/sebanc/brunch/files/4565317/dmesg.txt)

abdouakil, 2020-05-02:
Update:
I tried an other recovery different from rammus and it has the same problem.
I tried configuring 40-touchpad-cmt.conf and 20-mouse.conf or deleting them but none of this works, (even after deleting these files I could use my touchpad only after I sign out and then log in back).

sebanc, 2020-05-02:
2 things to try:

1) If you add to the kernel command line "psmouse.synaptics_intertouch=1", does it work correctly ?

1) enter "sudo rmmod psmouse" and then "sudo modprobe psmouse" in crosh shell after boot (if it works, it is possible to automate it)

abdouakil, 2020-05-02:
Good afternoon @sebanc,

unfortunately none of the 2 things you mentioned worked.

abdouakil, 2020-05-02:
UPDATE: 
So I tried to run these two commands and it seems that now i can use my touchpad, but using more than one finger seems broken still (e.g two finger scrolling).

sudo modprobe -r psmouse
sudo modprobe psmouse synaptics_intertouch=0 proto=exps

sebanc, 2020-05-02:
This should be almost the same as adding "psmouse.synaptics_intertouch=0" to the kernel command line, have you tried it ?

abdouakil, 2020-05-02:
I tried but it didnt work. The way i did is i added "psmouse.synaptics_intertouch=0" after cros_debug using edit grub config from chromeos in crosh shell. The other way i did is to add it after cros_debug from grub customizer. but I'm not sure if this is the right way to add a command to the kernel command line.Unfortunately none of them worked.

abdouakil, 2020-05-02:
UPDATE:

To make this change permanent, I created a file such as touchpad.conf under /etc/modprobe.d/, and I put the following line in it:
options psmouse proto=imps

The only problem that still exist is the multi finger gestures. I hope there is a way to fix it.

Zece88, 2020-05-04:
Hi @sebanc,

I've tried latest version r81 (test build), and only touchpad not working. It accepts keypres, but can't move around, same goes for gestures.

My laptop is a Asus 556UB, I believe it uses the Elan driver (not sure).

I can say that using a linux build: linuxmint-19.3-cinnamon-64bit, touchpad is working almost perfect. Maybe you could extract driver from there for me?.

Best regards

sebanc, 2020-05-04:
@abdouakil For each device that you have with a name like /dev/input/mouseX, could you post the output of "udevadm info /dev/input/mouseX" ?

@Zece88 Your issue seems to be different, could you open another issue and attach a dmesg ?

abdouakil, 2020-05-05:
Hi @sebanc,

for mouse0:

P: /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-7/i2c-SYNA3067:00/0018:06CB:8265.0002/input/input13/mouse0
N: input/mouse0
S: input/by-path/pci-0000:00:15.1-platform-i2c_designware.1-mouse
E: DEVLINKS=/dev/input/by-path/pci-0000:00:15.1-platform-i2c_designware.1-mouse
E: DEVNAME=/dev/input/mouse0
E: DEVPATH=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-7/i2c-SYNA3067:00/0018:06CB:8265.0002/input/input13/mouse0
E: ID_INPUT=1
E: ID_INPUT_TOUCHPAD=1
E: ID_PATH=pci-0000:00:15.1-platform-i2c_designware.1
E: ID_PATH_TAG=pci-0000_00_15_1-platform-i2c_designware_1
E: ID_SERIAL=noserial
E: MAJOR=13
E: MINOR=32
E: POWERD_ROLE=internal_touchpad
E: POWERD_TAGS=usable_when_laptop usable_when_display_off role_internal_touchpad
E: POWERD_TAGS_USABLE=usable_when_laptop usable_when_display_off
E: SUBSYSTEM=input
E: TAGS=:powerd:
E: USEC_INITIALIZED=49685246

for mouse1:

P: /devices/platform/i8042/serio3/input/input25/mouse1
N: input/mouse1
S: input/by-path/platform-i8042-serio-3-mouse
E: DEVLINKS=/dev/input/by-path/platform-i8042-serio-3-mouse
E: DEVNAME=/dev/input/mouse1
E: DEVPATH=/devices/platform/i8042/serio3/input/input25/mouse1
E: ID_INPUT=1
E: ID_INPUT_MOUSE=1
E: ID_PATH=platform-i8042-serio-3
E: ID_PATH_TAG=platform-i8042-serio-3
E: ID_SERIAL=noserial
E: MAJOR=13
E: MINOR=33
E: SUBSYSTEM=input
E: TAGS=:powerd:
E: USEC_INITIALIZED=69467039

sebanc, 2020-05-05:
Could you try adding " i8042.noloop i8042.nomux i8042.nopnp i8042.reset" to the kernel command line (after cros_debug) ?

Taken from: https://wiki.gentoo.org/wiki/Synaptics

abdouakil, 2020-05-05:
Hi @sebanc,

Unfortunately, " i8042.noloop i8042.nomux i8042.nopnp i8042.reset" couldn't fix my problem.

sebanc, 2020-05-05:
Just to verify, could you send me a dmesg with "console= i8042.noloop i8042.nomux i8042.nopnp i8042.reset" added to the kernel command line ? I am not sure your modifications are correctly taken into account due to a previous comment on "console=" which should definitely prevent text from being displayed on the screen.
Not sure the other arguments help for the touchpad or not though.

abdouakil, 2020-05-05:
Hi @sebanc,

Here is a dmesg as requested 
[dmesg.txt](https://github.com/sebanc/brunch/files/4583650/dmesg.txt)

sebanc, 2020-05-06:
ok, that's perfect !

Let's try something else, could you replace those kernel command line parameters which do not help with "module_blacklist=i8042" ?
If it does not work, could you also try "module_blacklist=i2c_hid" ?
And also "module_blacklist=i2c_hid,i8042".

abdouakil, 2020-05-06:
Goodmorning @sebanc,

I tried the three options you mentionned but none of them worked. Here are the three dmesg:
[module_blacklist=i2c_hid,i8042.txt](https://github.com/sebanc/brunch/files/4586394/module_blacklist.i2c_hid.i8042.txt)
[module_blacklist=i2c_hid.txt](https://github.com/sebanc/brunch/files/4586395/module_blacklist.i2c_hid.txt)
[module_blacklist=i8042.txt](https://github.com/sebanc/brunch/files/4586396/module_blacklist.i8042.txt)
Thanks for your hard work. I dont know if this gonna help but I forgot to mention that the mouse is working perfectly on linux mint.

sebanc, 2020-05-09:
I changed something in the new release, to be honest I don't think it will help but could you try it if you have the time ?

abdouakil, 2020-05-09:
Hi @sebanc,

Unfortunately I tried updating brunch to the leatest release and it didnt help.

sebanc, 2020-05-09:
Could you try adding "module_blacklist=hid_rmi" to the kernel command line ?

sebanc, 2020-05-09:
And also playing with "psmouse.synaptics_intertouch" value while hid_rmi is blacklisted.

sebanc, 2020-05-09:
Try booting also with "module_blacklist=hid_multitouch"

abdouakil, 2020-05-09:
Hi @sebanc,

I tried all of them but no luck.

sebanc, 2020-05-10:
Well, the touchpad is correctly recognized but won't work and nothing makes a difference. Also, I verified my kernel config against cloudready one.

Your best shot is to try to make a specific config for it, such as the examples you have in /etc/gesture folder but there is not much more I can do.

Last idea, do you have an option to change touchpad mode in bios ?

abdouakil, 2020-05-12:
Hi @sebanc,

I think I will give a shot to another OS for this moment, and maybe I will came back after a while. Thank you for all the support. I will close this thread for now until I get some luck next time.

closed 2020-05-12 by abdouakil