sebanc/brunch #85: Few things not working HP Pro tablet 608 g1

xp1ode, 2020-04-30:
First of all thank you for brunch, it's made my tablet that much better.  The only options I had were Ubuntu 20.04 and Windows, Ubuntu is not so bad but i really wanted a touchscreen OS to run on this tablet.  But here are the things that are not working for me and was wondering if anyone can help
Camera
Microphone
Sound
Screen rotation

Things that do work
Touch screen
Wifi
BT

Does anyone know how to get the notes app to show up on the lock screen again?  For some reason it was working when I ran the system off the SD before installing it on the actual device.  Now that is running off the device's storage I don't get notes on the lock screen.

sebanc, 2020-04-30:
Hi,

Could you attach a dmesg to this issue ? (go to crosh shell, run "dmesg > ~/Downloads/dmesg.txt" and attach the dmesg.txt from your downloads folder)

xp1ode, 2020-04-30:
Sure can, here you are...

https://drive.google.com/open?id=1lRYqT24npKuefMBAtupySuH6GkyF1iMv

sebanc, 2020-05-01:
As far as I can see the cameras are not supported under linux.

I think the accelerometer should work but your device must not be in tablet mode, could you run "sudo edit-grub-config" and add "options=force_tablet_mode" after cros_debug ?

For sound/mic, i will try something in the next release.

xp1ode, 2020-05-01:
Dude you're awesome, thank you.  I turned on the tablet mode and I get the option under displays for auto-rotation but it's not working.  Is there anything else I may be able to try to get it working?

sebanc, 2020-05-01:
I saw in your dmesg that the sensors module is correctly loaded so in theory rotation should work, can you try to reboot and see if it works ?

xp1ode, 2020-05-01:
Just tried it, no dice.

sebanc, 2020-05-03:
Hi, could you try the new testing release ?

xp1ode, 2020-05-03:
I really appreciate you taking the time to reply to my posts.  Just updated to your latest release and everything is still the same.  Auto-rotation doesn't work and neither audio or Mic.  I have tried Ubuntu 20.04 and Audio and screen rotation works on it.  Would I be able to pull off any info from there that might help?

sebanc, 2020-05-03:
I may have wrongly identified the audio driver, could you attach another dmesg so that i can verify if the driver has been loaded ?

xp1ode, 2020-05-03:
Here you are

https://drive.google.com/open?id=1OxlE9EQqLJHtc-E5Wg5wwg6xNXA4UC-2

sebanc, 2020-05-09:
Could you try the new release, I tried to improve the sound fix ?

xp1ode, 2020-05-09:
Installed but still no audio unfortunately.

sebanc, 2020-05-09:
Could you post another dmesg ?

Also what's the output of "aplay -l" in crosh shell ?

xp1ode, 2020-05-09:
https://drive.google.com/open?id=1BhvrrQtrkLjGDIf6zHdvvxu0hVunF8Sz (dmesg)
https://drive.google.com/open?id=1e32I33iiI8Dr2DitTdT3V8ozdumkYVbe (aplay -l)

Here you are.

sebanc, 2020-05-10:
Could you add "snd.slots=bytcht_cx2072x,snd_hdmi_lpe_audio" to the kernel command line, check if you have sound and if not provide the result "aplay -l" again ?

xp1ode, 2020-05-10:
https://drive.google.com/open?id=18XzZ5UrEKRNeSzUfbpMWP6QBXxhOxhCt

same outcome, I think this tablet might just be trash.

sebanc, 2020-05-10:
I am pretty sure we are not far from getting it to work. Try to add "module_blacklist=snd_hdmi_lpe_audio" to the kernel command line, check if you have sound and if not provide the result "aplay -l" again ?

xp1ode, 2020-05-10:
https://drive.google.com/open?id=127S60mU7ZVnzIsyDjvEXu5FyqP5jQv2P

Here you are.

sebanc, 2020-05-10:
This time it looks good, it is still not working ?

xp1ode, 2020-05-10:
nope, tried a few videos online and still no audio

sebanc, 2020-05-10:
Does "dmesg | grep fw_sst" return something ?

xp1ode, 2020-05-10:
no, it doesnt return anything

sebanc, 2020-05-10:
Ok, I verified all the steps in here and it seems ok:
https://github.com/heikomat/linux/tree/cx2072x/cx2072x_fixes_and_manual

What is the output of "alsaucm open bytcht-cx2072x" ?

xp1ode, 2020-05-10:
i get nothing from that command, but now when i boot up and right before the chromeos logo with the white background pops up, i get this weird sound from the tab.  As if the speakers want to output something but its more like a "clack" sound. Also after booting into the tablet i get no audio. 

BTW i can't thank you enough for how much time you are spending on this.

sebanc, 2020-05-11:
Unfortunately, not having output on "alsaucm open bytcht-cx2072x" is normal, I was hoping to see some error.

For now, I am out of ideas, your best chance would be to try playing with "alsamixer" settings.

xp1ode, 2020-05-11:
Thank you i really appreciate your help.

djamesrock, 2020-05-15:
Can something in this bug fix help?
https://bugzilla.kernel.org/show_bug.cgi?id=115531#c122

sebanc, 2020-05-16:
@djamesrock That could have been related to this but the solution provided is for an ASUS E200HA and according to issue #144 the only thing not working on that laptop is the touchpad so it looks like the included sound driver is working as expected.

However, the driver / ucm files might not be compatible with all devices equipped with this chip:
https://github.com/heikomat/linux/issues/12

I am not sure how to move on with this issue. Btw, do you also have the same issue ?

swarshall, 2020-05-16:
Hi, I want to tell you that I have no sound on my Asus E200HA in addition to the trackpad problem .. I just don't need it. Sorry

> @djamesrock Cela aurait pu être lié à cela, mais la solution fournie est pour un ASUS E200HA et selon le problème [# 144,](https://github.com/sebanc/brunch/issues/144) la seule chose qui ne fonctionne pas sur cet ordinateur portable est le pavé tactile, donc il semble que le pilote audio inclus fonctionne comme prévu.
> 
> Cependant, les fichiers driver / ucm peuvent ne pas être compatibles avec tous les appareils équipés de cette puce:
> [heikomat / linux # 12](https://github.com/heikomat/linux/issues/12)
> 
> Je ne suis pas sûr de savoir comment passer à ce problème. Btw, avez-vous également le même problème?

Hi, I want to tell you that I have no sound on my Asus E200HA in addition to the trackpad problem .. I just don't need it. Sorry

sebanc, 2020-05-16:
@swarshall Thanks for correcting my misunderstanding on sound working.

@djamesrock Thanks for pointing this, the solution might be there.

@xp1ode Could you download the file below, backup your /usr/share/alsa/ucm directory and replace it with the ucm2 folder from the downloaded archive ?
https://github.com/alsa-project/alsa-ucm-conf/archive/master.zip

djamesrock, 2020-05-16:
> @djamesrock It could have been related to this but the solution provided is for an ASUS E200HA and according to issue #144 the only thing not working on that laptop is the touchpad so it looks like the included sound driver is working as expected.
> 
> However, the driver / ucm files might not be compatible with all devices equipped with this chip:
> [heikomat/linux#12](https://github.com/heikomat/linux/issues/12)
> 
> I am not sure how to move on with this issue. Btw, do you also have the same issue ?

Yes I also have the trackpad and audio not working, I have a HP x2 10-n113dx Detachable PC. 
Trackpad I can get around because of the touchscreen. The audio is my real pain. The audio does work via bluetooth. If the audio issue here is fixed I believe it will also fix my problem. I have the exact same audio hardware.

sebanc, 2020-05-16:
@djamesrock Regarding the link you posted:
- I would say that the kernel patches are correctly applied as the driver is correctly loaded.
- For ucm files, it seems the ones that the ones which were mentioned in your link are not the same ones that I included so I hope it is that.
- I have not yet checked the alsa-lib version in chromeos, though I hope it is sufficient as I cannot do much about it.

djamesrock, 2020-05-16:
> @djamesrock Regarding the link you posted:
>
> * I would say that the kernel patches are correctly applied as the driver is correctly loaded.
> * For ucm files, it seems the ones that the ones which were [...]

Thank you for your reply, I triend the master zip. It did not solve it for me. Thanks for your time, I will just try what gets suggested over here and will report if it works.

sebanc, 2020-05-24:
Could you try the test release below ? It might fix an issue due to a delay before the sound module is available.
https://github.com/sebanc/brunch-testing/releases

If it does not work,, Could you also try to add to the kernel command line "snd.slots=,snd-hda-codec-hdmi" (replace "snd-hda-codec-hdmi" with your hdmi codec, I don't have access to the logs anymore).
Verify with "aplay -l" that the real audio card is first and hdmi is second.

kbnhvn, 2020-06-11:
Is now sound working on HP pro 608 g1?

sebanc, 2020-06-14:
According to #309 it seems the sound issue is fixed.

djamesrock, 2020-06-16:
> According to #309 it seems the sound issue is fixed in latest build.

I have to confirm you sir sebanc are a genius. Sound working on my HP Pavilion 10-n113dx x2 Detachable PC.

xp1ode, 2020-06-16:
Awesome stuff!!! I can't believe it actually works. Thank you so much for this. Would you happen to know why the Mic is not working? If it's tied to the same cx2072x, should it be working or is it a whole different set of driver code?

sebanc, 2020-06-16:
The microphone issue is probably the general one reported in #56 that I am not managing to fix. There are some workarounds to fix it mentioned in #56.

kbnhvn, 2020-06-16:
I have mic working with same soundcard with this brunch, on HP pro 608 g1

xp1ode, 2020-06-17:
@sebanc tried the suggestions on #56 but they did not work.  The command amixer scontrols only gives me "Simple Mixer control 'Master',0" and nothing more. 

@kbnhvn are you using samus or rammus?  I'm currently on samus, wondering if that might be an issue!? Also what BIOS version are you using for your tablet, I have this feeling that me updating to the latest one might have been a huge mistake and that may be what's causing all these problems.  I've been following some of the Androidx86 threads and some individuals have success with some very similar specs/hardware to this one but I have had nothing but issues.

kbnhvn, 2020-06-17:
It worked with rammus, and I switched yo Octopus which seems more stable for my device and it still works.
I don't know which bios version, don't have tablet with me now but it's an old version. I've read newer versions can make touchscreen issues.

xp1ode, 2020-06-17:
Just tried both Rammus and Octopus and neither one has Mic working for me. I'm almost 100% convinced that the upgraded BIOS is what has given me so much problems. Too bad it's not possible to downgrade! Thank you both for the help.

closed 2020-06-26 by sebanc