sebanc/brunch #86: Lenovo yoga thinkpad 11E 5th Gen

Lucaacer, 2020-04-30:


sebanc, 2020-04-30:
Hi,

Could you attach a dmesg to this issue ? (go to crosh shell, run "dmesg > ~/Downloads/dmesg.txt" and attach the dmesg.txt from your downloads folder)

Lucaacer, 2020-04-30:
Here it is, thanks

[dmesg.txt](https://github.com/sebanc/brunch/files/4557772/dmesg.txt)

Lucaacer, 2020-04-30:
Just one more thing, I can easily connect to the hotspot of my smartphone but, if I choose 'mobile devices' and the my phone, the system crashes.

sebanc, 2020-04-30:
Ah sorry I should have mentionned that the dmesg needed to be taken just after a reboot, could you please attach another one ?

Lucaacer, 2020-04-30:
Here it is, thanks

[dmesg.txt](https://github.com/sebanc/brunch/files/4557940/dmesg.txt)

sebanc, 2020-04-30:
Unfortunately, only the hid-sensors have been ported to be compatible with chromeos and your device uses the specific bmc150 accelerometer module.
I am sorry but unless someone with the right hardware ports this sensor module, rotation will not work in chromeos.

For the other issue, could you run "sudo edit-grub-config" and add "options=iwlwifi_backport" to the kernel command line (after cros_debug) ?

Lucaacer, 2020-04-30:
Hi, I edited the grub, but the system still crashes when choosing my device. The phone actually activates the hot spot and the pc gets connected, but after 3 or 4 seconds everything crashes.

Not a big issue, anyway, since I can easily activate the hot spot on the phone and then connect the pc flawlessly.

I will have a look if there is a patch floating around for the accelerometer. Btw, I have noticed that, if I fold the screen and then choose a fixed orientation (lets say 90 or 270 degree) the system keep that orientation every time I fold it again. So when I use the hardware keyboard, the screen is horizontal and when I use the device in tablet mode, it automatically puts the screen to 90 degree. A little useful workaround.

Thanks again.

Lucaacer, 2020-04-30:
It looks this one is the module https://github.com/torvalds/linux/blob/master/drivers/iio/accel/bmc150-accel-i2c.c

I will try to compile from crostini, uname -r gives as output 4.19.87 just like your kernel, it should be fine:)

sebanc, 2020-05-02:
Unfortunately, the module is already included so that is not the issue. The issue is that chromeos uses a very specific sensors format and that the sensor module would have to be aligned with the chromeos  specific format to be usable.

Lucaacer, 2020-05-02:
Thanks for replying, I was going to download the kernel sources, you saved me some time:)

I want to be sure I got it, do you mean the module should be written from scratch to comply to the chromeos format? Doesn't any chromebook use such accelerometer? It sounds weird, I will look for some info, maybe I can find the correct module somewhere, though it's a minimal issue.

Have a good day.

sebanc, 2020-05-02:
What i am saying is that as far as I understand, even if the exact same sensor was included into a chromebook, it would have a different firmware and would be managed by the cros-ec-sensors module located in the kernel source at "/drivers/iio/common/cros_ec_sensors/".

In your case, to make the sensor work, you would therefore need to modify "/drivers/iio/accel/bmc150-accel-i2c.c" to have it match the behavior of "/drivers/iio/common/cros_ec_sensors/".

juicypop, 2020-05-03:
@Lucaacer we're in the same boat, my accelerometer is a Kionix one and I'm facing exact the same problem that my driver isn't compatible with cros_ec_sensors
Please let me know if you're modifying your driver and have success, I had no success yet ;(

Lucaacer, 2020-05-03:
I am sorry Juicypop@, but it's beyond my skills and it's not that easy even finding some info:(

juicypop, 2020-05-03:
Yeah, just like my situation, beyond knowledge and lack of useful information :(

Lucaacer, 2020-05-06:
I am writing here just to keep things in order. First of all, everything is very stable on my device, so I am not complaing at all, just experimenting.

This is my grub

set timeout=2

menuentry "ChromeOS brunch" {
  regexp --set disk "(^.+)(,gpt)" $root
  linux ($disk,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug
  options=enable_updates
  initrd ($disk,7)/initramfs.img
}

As you can see, I just changed the name to ChromeOS brunch and added options=enable_updates after cros_debug. I changed the name to be sure the grub had actually been updated and I can now see it has, so also options=enable_updates should work.

I tried this config both from a live usb and from the ssd and I can't get any native update from 80 something, though rammus should be on 81.0.4044.127. Moving to beta or dev doesn't change anything and I can even go back and forward from a channel to another without a powerwash (which should not be feasible by design).

I have even updated to the latest testing framework on my ssd, but nothing seems to change. Btw, updating the framework on my "dirty"  installation (i.e. without mounting the .img) did give a weird output, but everything did get rebuilt in the right way though.

What's wrong in my grub? Did I make any typo? Do I have to remove cros_debug?

Everything on my devices is cloud bound, so I won't give a damn if anything gets nuked:)

Thanks a lot.

Lucaacer, 2020-05-09:
Sorry for being so dumb, I had added options=enable_updates in the line below, instead of placing  in the kernel command line. Update flawlessly.

sebanc, 2020-06-26:
This issue has been included in #331 "Main issues on which help is needed". I am therefore closing individual issues related to sensors other than Intel ISH not working in chromeos.

closed 2020-06-26 by sebanc

hotelbuddy-online, 2021-12-19:
I got the accelerometer working in my Lenovo Ideapad 5 by using Volteer chrome os image.