sebanc/brunch #87: HP pavilion x360 i5-10210u realtek rtl8822ce wifi

senssopy, 2020-04-30:
HELLO, I'm happy to success of booting chrome os by usb because of your brunch.

Since Dec (2019), I tried to boot chrome os following chromefy (not yours).
I can't boot, any way..

Continuously googling...
I found sebanc, I think thank you so much!

My computer is HP pavilion x360 i5-10210u realtek rtl8822ce wifi.

I surprised it detailedly provide even a touch driver including many things.

I also want to use wifi internet by my computer.
I googled so many times.

I found here ↓↓↓

https://github.com/lwfinger/rtlwifi_new

I tried ↓↓↓ (using linux terminal application, not ctrl + alt + f2)

git clone http://github.com/lwfinger/rtlwifi_new.git -b rtw88
cd rtlwifi_new
make
sudo make install

cd ~/rtlwifi_new
git pull
make
sudo make install

It didn't work.

Could I use my com's wifi?

juicypop, 2020-04-30:
You could try adding "options=iwlwifi_backport" to the kernel after cros_debug
Just run "sudo edit-grub-config" in crosh shell, revert it if messes up something or doesn't work at all
If you can't boot with this option you can edit the kernel line also hitting e while grub comes up with Chrome OS Option

senssopy, 2020-05-03:
Thx for your interest. I tried but it didn't work. It didn't mess up, too. ^ ^

sebanc, 2020-05-03:
Hi, this driver has been included in the new testing release, however I do not have the hardware so it needs testing.

Could you add "options=rtl8822" to the kernel command line and see if it works ?

senssopy, 2020-05-04:
It didn't work.

https://drive.google.com/open?id=17f69gEbBUwylKtOxnKGPg0H4t7wVzPcF

And I tried to install my wifi driver rtl8822ce, terminal error is next.
1. no such file or directory
2. make compile error 2
(make[1]: *** /lib/modules/4.19.87-07826-g55ab70aff690/build: no such file or directory. [make file:2291: modules] error 2 compile make driver error: 2)

Command sudo apt-get install linux-headers-$(uname -r) didn't work.

(Next is another driver install site)
https://github.com/fanguanqun/RTL8822CE_WiFi_linux
https://github.com/juanro49/rtl88x2ce-dkms

sebanc, 2020-05-04:
Hi, I think the dmesg you have provided is from crostini, could you provide one from crosh shell while booting with "options=rtl8822" ?

Also, could you boot without "options=rtl8822", run the 2 below commands in crosh shell and tell me what is the output ?

```
insmod /lib/modules/rtwpci.ko
insmod /lib/modules/rtw88.ko
```

senssopy, 2020-05-05:
When before the computer boot in front of the chrome os,
I pressed c or e, https://drive.google.com/file/d/1-KAngIG0o3UMEuMNjrdwCCXfhVpA8pNq/view?usp=sharing

and boot..

crosh shell dmesg

https://drive.google.com/file/d/1AKe9EJNrvn8faaJPgAEOoJXuvKXVE_B6/view?usp=sharing

and typed sudo 1 and sudo 2

We trust you have received the usual lecture from the local System
Administrator. It usually boils down to these three things:

2nd type..
localhost / # sudo insmod /lib/modules/rtwpci.ko
insmod: ERROR: could not insert module /lib/modules/rtwpci.ko: Unknown symbol in module

chronos@localhost / $ sudo insmod /lib/modules/rtwpci.ko
insmod: ERROR: could not insert module /lib/modules/rtwpci.ko: Unknown symbol in module

senssopy, 2020-05-05:
Hi, I finally get a this, dmesg.txt

https://drive.google.com/file/d/1aA8uaYcAlMDsWRKoeuipYjoaI-R12NPF/view?usp=sharing

sebanc, 2020-05-08:
Hi,

Could you try the new release ? wifi should work out of the box without the need for "options=rtl8822".

If it does not work, could you send me another dmesg ?

senssopy, 2020-05-09:
Thx, wifi is ordinarily, perfectly work! I surprised it. I didn't do anything. Just install & sudo resize-data.

Thank you very much ^ ^ .

closed 2020-05-09 by senssopy