sebanc/brunch #89: Medion Akoya E3216 - Kionix Accelerometer not working

juicypop, 2020-04-30:
I installed brunch r80 and I'm using rammus r80.

So far everything seems to work as it should accept one thing: my Accelerometer a Kionix kxcj9 (as shown in Windows Device Manager) 

@sebanc already had a look into this and couldn't help me out as it appears that the driver isn't compatible with cros_common_ec
If I mixed something up please correct me here. 

Maybe someone else has a clue how to port the driver so that it actually starts working. 

I think that this accelerometer could also be found in other brand laptops and maybe someone got it already working, but I haven't found anything yet. 

In the meantime I will collect useful information in this issue which could help solving this problem.

juicypop, 2020-05-01:
@sebanc could you maybe have once more time to look into the link I found? It looks pretty decent and I ask myself whether this could be implemented into brunch?

sebanc, 2020-05-01:
Hi,

Unless I am mistaken, the driver you point at is not a kernel driver but an embedded controller driver. As seen in the code, the chromebook sensor is handled by the motionsense sensor driver located at "/drivers/iio/common/cros_ec_sensors/" in the kernel.

Whereas your computer's sensor is managed by "/drivers/iio/accel/kxcjk-1013.c" in the kernel.

Even though the hardware might be similar, they are handled in a completely different way. As far as I know, the only way to make it work would be to modify "/drivers/iio/accel/kxcjk-1013.c" in order for its output to be aligned with "/drivers/iio/common/cros_ec_sensors/".

juicypop, 2020-05-02:
Ok I think I don't get it yet how things work. I also can't understand how there can be non working modules in the kernel if they aren't yet compatible with the cros_ec_sensors

sebanc, 2020-05-02:
> Ok I think I don't get it yet how things work. I also can't understand how there can be non working modules in the kernel if they aren't yet compatible with the cros_ec_sensors

Chromeos kernel is based on the linux kernel, it includes all linux drivers + some specific chromeos drivers. Those sensors are disabled by google in the kernel config but I enabled them as I base my config on ubuntu.

sebanc, 2020-06-26:
This issue has been included in #331 "Main issues on which help is needed". I am therefore closing individual issues related to sensors other than Intel ISH not working in chromeos.

closed 2020-06-26 by sebanc