sebanc/brunch #91: Booting inside and from Virtualbox impossible?

jordanvanbergen, 2020-04-30:
I tried to get this installed inside Virtualbox. Everything works and if I start the virtual machine I see it booting and as soon as the white screen with the Chrome OS shows it "stops" CTRL + ALT + F2 will get me to the command prompt so I see it running and I can login with chronos. 

So the thing definitely boots and works but no graphic screen. 

It will save me and others a lot of time if someone can tell if this could or will ever work using it inside VirtualBox so booting this from VirtualBox?

jordanvanbergen, 2020-04-30:
By The Way: I have this running on my Lenovo Ideapad Yoga 13 from 8 years ago and it's running fast, reliable and with no problems until now (only internal mic not working).

sebanc, 2020-04-30:
Hi,

I don't know VirtualBox enough to answer that 100%.

My opinion is that in your case the display stops because Chromeos does not support VirtualBox drm drivers and even if they were included, a lot of things would probably not work (android apps, linux...).

Now, I don't know if it can be done with virtualbox but if a virtual machine would expose the i915 card as passthrough device, it could probably work.

closed 2020-05-18 by sebanc