sebanc/brunch #93: Not booting after partition shrink

hacker1024, 2020-04-30:
Hi, I just shrank my data partition with GParted, and created a new one for Windows  data (number 13) Now, GRUB won't boot Chrome OS:

```
error: unknown filesystem.
error: you need to load the kernel first.
```

As far as I can tell, my data and all the other partitions are still there, and readable from the GRUB shell. I didn't change any numbers. What's going on? (The kernel seems to be on partition 3, and my data on 12).

hacker1024, 2020-04-30:
Partition 7 is a raw kernel image, version ~ 4.4.200. Partition 6 is around 4.4.100.

Grub seems to be trying to load the kernel from 7/kernel, which makes no sense - the only partition that has that is 3, ROOT-C. If I changes the 7s to 3s in the boot script, it boots, but:

```
brunch: scanning device /dev/nvme0n1
brunch: scanning device /dev/sda
brunch: Chromeos was not found on any device, fallback to shell...
```

It's on /dev/sda, not the other disk

hacker1024, 2020-04-30:
`fdisk -l` in this shell shows no partitions other than sda1...

lfom, 2020-04-30:
User data partition (also called "stateful partition" IIRIC, more or less like "home" in Linux) is encrypted: you can't see its contents when mounted from the image (or disk in a real installation). And to make things more complicated, the script that mounts the partitions during boot uses UUID, so if you change the disk ID for any reason, it won't mount anymore.

hacker1024, 2020-04-30:
The GRUB script doesn't use UUIDs in this case, though, hence why changes to the partition layout seemed to break it. It uses a regex expression to choose a partition relative to the ESP, as far as I can tell.

hacker1024, 2020-04-30:
Furthermore, the stateful partition is not completely encrypted. Many parts of it aren't. It can be mounted, but the encrypted directories just look like they contain gibberish.

lfom, 2020-05-01:
UUID in GRUB is used for brunch boot, that is where the magic happens (ROOT-C)...

By the way (not related), but the last time I checked there was no ROOT-B, I saw only ROOT-A and ROOT-A1... Maybe something needs to be updated in the install process.

hacker1024, 2020-05-01:
I fixed the problem by re-numbering the partitions. It looks like Windows randomly decided to number them in order...

I took a partition table backup with sfdisk: sfdisk -d /dev/sda > sda.ptable

I then copied the table backup, and changed the numbers to the correct order.

Then I flashed the new partition table: sfdisk /dev/sda < sda.ptable

And everything's fine (although brunch did rebuild the rootfs for some reason)

hacker1024, 2020-05-01:
I'd love to write this up on the wiki, but that's not used anymore. @sebanc, can we get a wiki on the new site?

sebanc, 2020-05-01:
Hi,

ChromeOS has a specific partitioning system with only 12 partitions which should remain untouched. The dual boot method is the proper way to dual boot with windows, creating other partitions after ChromeOS ones is dangerous and not a good idea in my opinion. Please don't post this method anywhere.

closed 2020-05-01 by sebanc

hacker1024, 2020-05-01:
The partition wasn't for Windows itself, I just wanted a new one to store files/apps.

I suspect this could happen on a pure Chrome OS disk, though, if it's simply accessible to a Windows install on another one. I'd post it as a fix in case of emergencies.

sebanc, 2020-05-01:
In general, even if the partition is not for windows, it is not a good idea to change the partitioning scheme. Many scripts in ChromeOS just call the `rootdev` command to get the main block device and manipulate partitions by numbers.

There is a high risk that some chromeos scripts will fail due to an incorrect number of partitions on the main block device or behave incorrectly.