sebanc/brunch #100: Brunch keeps resizing rootfs like it's the first boot

aviwad, 2020-05-01:
Hi there! Wonderful piece of software you've written. Unfortunately, running this on my Acer Swift 3 (6th gen Intel i3 cpu only), has led me to a few sharp edges. 

Using your latest r81 Brunch version with Rammus v80 software, and installing and running ChromeOS on my pendrive leads me to wait 1 hour as it resizes Rootfs on first boot. As you stated, this is expected. 

However, what is not expected is this happens on my 3rd/4th reboot at random. infact, it's started happening on every bootup. 

I was planning to replace my current Linux install on my HDD, but I'm worried if this rootfs resizing keeps happening on HDD as well, it will kill my harddrive with the constant read/writes every bootup.

sebanc, 2020-05-02:
Hi,

Brunch will rebuild the rootfs in the following situations:
- You update chromeos or brunch.
- You change the "options=" or "cros_debug" kernel command line parameters.
- You change machine.
- You modify something on partition 7.

Could you verify that you were not in one of those cases ?

If not, before brunch starts rebuilding the rootfs, you should see shortly one of those 3 messages:
- brunch: update detected
- brunch: new install detected
- brunch: device/framework change detected

Could you tell me which message is displayed when it rebuilds without reason ?

aviwad, 2020-05-02:
Ahhh it rebuilds the rootfs whenever i mess with the "options" thing. Although, why? What if I just want to try an option to test blacklisting my wifi module, why would that require the entire rootfs getting resized?

sebanc, 2020-05-02:
It rebuilds when using the "options" because those options are specific rootfs patches.

"options=" should only be used with those:
https://github.com/sebanc/brunch#framework-options

Else you can just pass things to the kernel command line without "options=", for exemple to blacklist a module, just add "module_blacklist=<name_of_the_module>" to the kernel command line.

aviwad, 2020-05-02:
Ahh yeah. To check if my wifi driver will be fixed with a blacklist, i always use options= side by side to fix my suspend. I think I will just make the "options=" permanent in the grub.cfg. 

thank you, this makes sense!

closed 2020-05-02 by aviwad