sebanc/brunch #101: Wi-Fi networks can't be found after wakeup from suspend

aviwad, 2020-05-01:
Hello! Sorry to bug with another bug. Wonderful piece of software, but I've encountered another bug. 

On my Acer Swift 3 (i3 6th gen, no other GPU), I first noticed my laptop wouldn't properly suspend on ChromeOS (Brunch r81 + Rammus 80). I fixed this by adding the following to my grub config:
```mem_sleep_default=deep```

However, this has led to another new bug: the system can't find any wifi networks after waking up from suspend. On the latest Linux kernel on Fedora 32, my wifi works perfectly fine after/before suspend, so I fear it may be because of some driver mismatch in Brunch.

My wifi card is: qualcomm atheros qca6174

Thank you!

sebanc, 2020-05-03:
Hi, could you the new testing version and see if you still have the issue ?

aviwad, 2020-05-04:
It made it much worse :cry: 
in the new testing version, unless i do ```options=suspend_s3``` it gets hung on a black screen in suspend. and with that option enabled, it takes 10 seconds to go to suspend, but never wakes up again, and freezes. 

so because of this I cannot actually test if the wifi works after suspend because it either never suspends, or freezes after suspend

sebanc, 2020-05-04:
Those suspend issues are quite difficult to debug but I can still try, could you attach a dmesg to this issue ? (go to crosh shell, run "dmesg > ~/Downloads/dmesg.txt" and attach the dmesg.txt from your downloads folder)

sebanc, 2020-05-05:
Also, could you try to replace the board.bin file in /lib/firmware/ath10k/QCA6174/hw3.0/ with the one present in this archive ?
[board.zip](https://github.com/sebanc/brunch/files/4579801/board.zip)

aviwad, 2020-05-05:
[dmesg.txt](https://github.com/sebanc/brunch/files/4580181/dmesg.txt)

1. here's the dmesg i ran before i swapped the board.bin file

2. even after replacing that board.bin file, it's still the same. it takes 10 seconds to go to suspend, and once i wake it up, the screen is on, and it takes me to the locked out page, but in a second it goes black, the cursor is still there, and after 5 seconds the display shuts off and it's frozen. then I REISUB it to rebooting.

sebanc, 2020-05-06:
Could you try adding "acpi=noirq" to the kernel command line ?

aviwad, 2020-05-06:
Hi. After doing that, the computer slowed down a lot (even the chrome bootup animation was taking 30 seconds), and none of my builtin peripherals were working (keyboard + touchpad)

sebanc, 2020-05-06:
ok, could you try booting with "pci=nomsi" added to the kernel command line ? (without "acpi=noirq")

aviwad, 2020-05-06:
after doing that, there was no noticeable difference.still took 10 seconds to suspend, and after waking up, worked for 3 seconds before black screen + cursor hung. only this time, the backlight remained on, but it still froze

sebanc, 2020-05-19:
Hi,

Just to say that I observed some weird behavior with the current chromiumos kernel source regarding networking and suspend for which I have not been able to clearly identify the origin of the issue.
I expect that this is caused by a bad commit which will be corrected in the next kernel release. I will update this issue once a release with a newer kernel is available.

sebanc, 2020-05-20:
If you have the time, could you try the release I have just uploaded in the "brunch-testing" repo (not this one), I have changed settings related to usb which might have an impact on your issue.

aviwad, 2020-05-21:
will try in a week

aviwad, 2020-05-26:
hi, i can't find the brunch-testing repo. it's not on your profile. can you link to it please?

sebanc, 2020-05-26:
It's a private repo, you have to accept something in an email you received from github, then the link is there:
https://github.com/sebanc/brunch-testing/releases

aviwad, 2020-05-26:
ahh I somehow didn't see the repo invite. accepted it, and now im downloading the repo. will let you know the results by tomorrow :smile:

sebanc, 2020-06-02:
Just to let you know that I have actual reports of suspend issues fixed with the current "brunch-testing"

aviwad, 2020-06-05:
It works now, but now sometimes the keyboard doesn't work after suspend, and my laptop doesn't shut down. But I see there's an even newer testing release, so I'll try that and let you know if any other problems exist

aviwad, 2020-06-15:
my only problem is the touchpad tap to click latency, which i've addressed in a separate bug. I'm closing this now, thank you!

closed 2020-06-15 by aviwad