sebanc/brunch #94: Bluetooth headset sounds terrible (SCO audio codec)

hacker1024, 2020-05-01:
Using my Bluetooth buds in [HSP](https://en.wikipedia.org/wiki/List_of_Bluetooth_profiles#Headset_Profile_(HSP)) for online calls sounds much worse than in other OSes.

They're QCY-T5s, and they support SCO, SBC, AAC, and LDAC.

cedricju, 2020-05-01:
One of my customer told me the same but I was on an old build ..
Will do s test on the latest build...
C

sebanc, 2020-05-01:
Hi,

I cannot test this as I don't have bluetooth earphones but would the below fix it ?
https://support.google.com/chromebook/thread/1499544?hl=en

hacker1024, 2020-05-01:
Unfortunately, not.
The reason the above fix works in that scenario is because when switching from the Bluetooth mic, the system stops using the headset mode and starts using the headphone mode - which sounds better, because there's more bandwidth available for the sound output (as there's no input).

On every OS, switching to heaphone mode will improve the sound quality.

I, however, need to use the headset microphone. In the mic on mode, other OSes still sound better than Chrome OS in the same mode.

yutouyes, 2020-05-05:
Try to enable newblue. It works for me
chrome://flags/#newblue

hacker1024, 2020-05-08:
Turns out this is a widespread Linux bug.

https://bugs.chromium.org/p/chromium/issues/detail?id=843048
https://bugs.launchpad.net/ubuntu/+source/pulseaudio/+bug/1838151

Shamw0w, 2020-09-22:
I recently got a pair of Jabra Elite 85h and managed to pair them with my Chromebook, and I'm sure when I first did they were working well for calls (I think it might have been on Google Meet and Microsoft Teams) but recently I haven't been able to use them on a few other platforms (Starleaf, GotoMeeting and Zoom). I'm not sure why they worked initially, and maybe it was the difference in platforms.