sebanc/brunch #96: External drives and sd cards are not recognized

keweminer, 2020-05-01:
This is great, thanks for all the hard work.

I'm running a Dell 5558 and everything runs pretty much as expected, but I have two issues. External hard drives are not recognized, nor is the sd card reader (Realtek RTS5170). I can work around the sd card issue with an adapter. But, I haven't found a solution for the external drive issue. Any thoughts?

sebanc, 2020-05-01:
Hi,

Which brunch release are you using ? Could you try the latest testing if it is not the one your are using ? I have fixed an issue related to this recently.

keweminer, 2020-05-01:
I am using Brunch r80 k4.19 testing 20200413. Would the latest be Brunch r81 k4.19 testing 20200419?

keweminer, 2020-05-01:
SD Cards and External Hard drives both work now, after using Brunch r81 k4.19 testing 20200419.

Thank you.

closed 2020-05-01 by keweminer