sebanc/brunch #98: Black screens on certain apps

B3nji27, 2020-05-01:
Hi

I have successfully followed the install guide for Brunch on an old Dell Optiplex Micro 3020, i3 processor, and it has loaded fine. The only issue I have is certain apps (e.g. Gmail, Playstore) if in full screen, will flash black and freeze regularly, and it seems to be on animations. I can't see anybody has this issue so wonder if it's just me? I have tested several Chrome OS builds when creating the USB, (Rammus and Fizz, which is a desktop as is the Dell, so thought may be more compatible) and it still is happening. This also happens whether installed to the SSD or using the bootable USB stick.

I'm sure it's just a flag i have to switch off or something, but any guidance appreciated

Thanks

sebanc, 2020-05-01:
Hi, this issue is supposed to be fixed in the latest testing version, could you try it if is not already the case ?

B3nji27, 2020-05-01:
Hi Sebanc

Thanks for replying. I'm using the latest 80 stable version. Should i use one of the testing versions?

sebanc, 2020-05-01:
Yes, try latest r81 testing version.

B3nji27, 2020-05-01:
OK I will try. I missed the release notes about fixing flickering Google Apps, and figured I should just use the stable version. I will try this, thanks you

B3nji27, 2020-05-01:
All good, thanks for the help. Serves me right for going for the stable version i guess haha

closed 2020-05-01 by B3nji27