sebanc/brunch #104: Instagram android app upload not working.....

jordanvanbergen, 2020-05-03:
Is it okay to post specific android app issues as well?

Instagram app is working really nice but I tried to add a story using a video today. Everything seems to work, the video place, you can add stickers, polls, etc. And when you are done you hit DONE and the story shows on the top left but with an exclamation mark and if you click that it shows Can't upload retry. 

Retried with LAN connection, different internet connections and wifi but it won't upload:

https://imgur.com/zP5kQVp

I hoped I could use a workstation like chrome OS to use social media apps like instagram. Still need my smartphone for now.

sebanc, 2020-05-03:
Hi,

On which brunch version does this issue appears ? I tried to make the brunch kernel closer to the standard chromeos kernel in latest release, could you try it ?

jordanvanbergen, 2020-05-03:
> Hi,
> 
> On which brunch version does this issue appears ? I tried to make the brunch kernel closer to the standard chromeos kernel in latest release, could you try it ?

On this release: https://github.com/sebanc/brunch/releases/tag/r81-k4.19-testing-20200503
How can I check if I really have this version running?

sebanc, 2020-05-03:
Ok, that's the newest then, there is no way currently to know which brunch version you are using, that's a good point, i will try to add that.

Which chromeos version and channel are you using ?

jordanvanbergen, 2020-05-03:
> Ok, that's the newest then, there is no way currently to know which brunch version you are using, that's a good point, i will try to add that.
> 
> Which chromeos version and channel are you using ?

rammus v80: 

So from https://cros-updates-serving.appspot.com/ this one: https://dl.google.com/dl/edgedl/chromeos/recovery/chromeos_12739.111.0_rammus_recovery_stable-channel_mp-v2.bin.zip

sebanc, 2020-05-03:
Last question, had you tried it with a previous release ? do you know if it was working ?

jordanvanbergen, 2020-05-03:
> Last question, had you tried it with a previous release ? do you know if it was working ?

No sadly not. So I can only assume the same happened. So if someone has the older version before today maybe they could try. 

Is it possible to revert the new brunch release to the old? So just do the same as installing a new release (updating) but pointing to an older brunch release? 

Maybe if this is possible I can try this later today. For now I assume the same applies for older versions but that's just an assumption.

sebanc, 2020-05-03:
Yes, normally you can update to a previous version without issue. Could you try it when you have the time (the previous testing version should be a good exemple) ?

Thanks !

jordanvanbergen, 2020-05-03:
> Yes, normally you can update to a previous version without issue. Could you try it when you have the time (the previous testing version should be a good exemple) ?
> 
> Thanks !

That's the only possibility I guess as stable version 80 had the black flickering issues and you where not able to use the android apps this way. So the only one to test is I assume the previous one:  https://github.com/sebanc/brunch/releases/tag/r81-k4.19-testing-20200419

jordanvanbergen, 2020-05-03:
> Yes, normally you can update to a previous version without issue. Could you try it when you have the time (the previous testing version should be a good exemple) ?
> 
> Thanks !

Reverted back to this release: https://github.com/sebanc/brunch/releases/tag/r81-k4.19-testing-20200419

Same problem as with the latest release. Can't upload when uploading an Instagram Story with video. If you hit the download button while creating the Instagram Story you get an error message as well. 

So both version have this issue. I won't go back to the stable release as those android app don't really work because of the black flickering issues.

sebanc, 2020-05-03:
No worries, that's perfect, I have just changed the kernel in latest release so it was just to know if it was related to that change or not. There is no reason that stable would have worked anyway as it was mostly the same kernel as the previous testing.

I will try to see if I can find something that could cause that and update this issue when i have news.

sebanc, 2020-05-04:
Hi, I have verified that all configs were correctly included in the kernel. I think it could be a kernel/chromeos version issue as the kernel is the r81 version. Could you add "options=enable_updates" to the kernel command line and update to r81 stable chromeos version ?

jordanvanbergen, 2020-05-04:
> Hi, I have verified that all configs were correctly included in the kernel. I think it could be a kernel/chromeos version issue as the kernel is the r81 version. Could you add "options=enable_updates" to the kernel command line and update to r81 stable chromeos version ?

I had this enabled and I have a Chrome OS r81.xxxx on my lenovo ideapad yoga 13. How do I check if this is a stable ChromeOS version?

sebanc, 2020-05-04:
Ok, if you are on r81, it is stable version, i will try to think about something else

lfom, 2020-05-04:
> Ok, that's the newest then, there is no way currently to know which brunch version you are using, that's a good point, i will try to add that.
> 
> Which chromeos version and channel are you using ?

Actually you can using `uname -a`, it will show the date when the kernel was built, and it is linked to the release build.

jordanvanbergen, 2020-05-14:
> Actually you can using `uname -a`, it will show the date when the kernel was built, and it is linked to the release build.

This uname -a is working nicely:

uname -a
Linux localhost 4.19.113-brunch-sebanc #1 SMP PREEMPT Sun May 10 13:29:05 CEST 2020 x86_64 Intel(R) Core(TM) i5-3317U CPU @ 1.70GHz GenuineIntel GNU/Linux

You see the date 10 May in it as well so this is the latest stable release. 

So nice to see you can check which version you have.

jordanvanbergen, 2020-05-14:
Issue not being able to share a story on Instagram app still exists. FYI

spygency1, 2023-09-22:
NEED URGENT HELP?!!

NOTE:MESSAGE ONLY THIS ACCOUNT
#🇬🇧Expert Here To Assist In Any Cyber Issues General_Hacking_Recovery_Spy_Blackmail_Stolen,Lost BTC_Hacked Phones ☎️ 📥
🌎Facebook hacking - recovery
🌎Instagram hacking - recovery
🌎Snapchat hacking - recovery
🌎Lost money(investment scams) recovery
🌎Cheating partner spy (online)
🌎Crypto wallet hacking
🌎Phone hacks📱
🌎Message hacks & tweaks

For swift response send a message to my social media accounts, we are active 24/7 to solve all your Online ETHICAL HACKING issues 

INSTAGRAM 📥 spygency.1
EMAIL 📧 spygency@gmail.com