sebanc/brunch #105: Huawei matebook hz-w09..problems..

geofrey89, 2020-05-04:
Thanks sebanc for your development is am now able to run Chrome OS on my Huawei matebook...I have  encountered the following problems..
1. Camera not working
2. Power button not working i.e when I press I don't see ant options..
3.if volume up, down and finger print scanner works I will be happy....thanks . 
I have tried. R80 stable, r81 testing latest .

sebanc, 2020-05-04:
Hi,

Could you attach a dmesg to this issue ? (go to crosh shell, run "dmesg > ~/Downloads/dmesg.txt" and attach the dmesg.txt from your downloads folder)

geofrey89, 2020-05-04:
[dmesg.txt](https://github.com/sebanc/brunch/files/4574660/dmesg.txt)
am using cave... it has similar cpu like mine..,,

sebanc, 2020-05-04:
Hi,

1- For the camera, I could not find any resources on its compatibility with linux and it is not detected at all by the kernel so I would say chances are low that I can make it work.

2- and 3- I can probably try something in my next release except for fingerprint scanner which is too specific in chromeos.

4- Could you unzip the attached file and copy both files in /lib/firmware/brcm/
[firmwares.zip](https://github.com/sebanc/brunch/files/4575659/firmwares.zip)

geofrey89, 2020-05-04:
Is there a way to turn off the verbose mode

sebanc, 2020-05-04:
Add "console=" to the kernel command line (after cros_debug), you will have a blinking cursor instead.

geofrey89, 2020-05-04:
I have added the two files. But still the Bluetooth is on but can not locate any device

sebanc, 2020-05-04:
Could you attach another dmesg ?

geofrey89, 2020-05-04:
[dmesg.txt](https://github.com/sebanc/brunch/files/4576441/dmesg.txt)

lfom, 2020-05-04:
My device running latest brunch has an Intel Bluetooth and it has them same problem: it turns on, scans but finds nothing. I was able to pair and connect using `bluetoothctl` using the shell tho. Also, when scanning, WiFi disconnects. I didn't find anything in dmesg or /var/log/messages that would given any hint about the cause... But I think it is not related to drivers, maybe it's something else when brunch installs, since normally when the kernel installs the drivers correctly I am able to use BT normally...

geofrey89, 2020-05-05:
i added console= but still verbose mode is present

geofrey89, 2020-05-05:
hi sebanc
here is another another dmesg. this time round am using nocturne ..can it assist in camera issue..?
[dmesg.txt](https://github.com/sebanc/brunch/files/4581372/dmesg.txt)

sebanc, 2020-05-08:
Hi,

I applied a patch in the new release which i hope will help with your buttons.

For bluetooth, what device are you trying to pair ?

The problem with the camera is that it has no driver in the linux kernel so trying a different image will unfortunately not help.

geofrey89, 2020-05-08:
Thanks sebanc.
for bluetooth i was trying to connect my phone mi8 lite.
I have updated to the new build..  
buttons are not responding.
here is the dmesg.
[dmesg.txt](https://github.com/sebanc/brunch/files/4601032/dmesg.txt)

lfom, 2020-05-08:
> for bluetooth i was trying to connect my phone mi8 lite.

You cannot connect to a phone directly (I was doing the same mistake), please see this thread/post:
https://github.com/sebanc/brunch/issues/67#issuecomment-624556402

sebanc, 2020-05-09:
Could you launch a linux live usb and provide the output of the below commands ? (you cannot use brunch for that as it overrides the product name):
```
cat /sys/class/dmi/id/product_name
cat /sys/class/dmi/id/sys_vendor
```

geofrey89, 2020-05-09:
Matebook HZ-W09
HUAWEI

geofrey89, 2020-05-09:
> Could you launch a linux live usb and provide the output of the below commands ? (you cannot use brunch for that as it overrides the product name):
> 
> ```
> cat /sys/class/dmi/id/product_name
> cat /sys/class/dmi/id/sys_vendor
> ```
Matebook HZ-W09
HUAWEI

sebanc, 2020-05-09:
Thanks, I had it wrong in my patch, hopefully the next release will give better results.

I will update this issue when the new release is out.

geofrey89, 2020-05-09:
> Thanks, I had it wrong in my patch, hopefully the next release will give better results.
> 
> I will update this issue when the new release is out.

Thanks sebanc..

sebanc, 2020-05-10:
Could you try the new stable release ?

geofrey89, 2020-05-10:
> Could you try the new stable release ?

i have updated ..
still the power button and volume buttons are not responding . here is the dmesg.
[dmesg.txt](https://github.com/sebanc/brunch/files/4605867/dmesg.txt)

sebanc, 2020-05-10:
Just found it, this one is on you ;)

> > Could you launch a linux live usb and provide the output of the below commands ? (you cannot use brunch for that as it overrides the product name):
> > ```
> > cat /sys/class/dmi/id/product_name
> > cat /sys/class/dmi/id/sys_vendor
> > ```
> 
> Matebook HZ-W09
> HUAWEI

MateBook HZ-W09 (with an uppercase B)

Well, next release...

sebanc, 2020-05-10:
The lines in red in the dmesg are normal I think for brcmfmac, i don't own such device but i always see those in the logs and it appears to work well

geofrey89, 2020-05-10:
> Just found it, this one is on you ;)
>
>
> MateBook HZ-W09 (with an uppercase B)
>
> Well, next release.

Sorry .. ok then

geofrey89, 2020-05-10:
Should have taken a photo .. I have re checked ..it should be uppercase B

sebanc, 2020-05-10:
No worries, it will just be for next release

geofrey89, 2020-05-10:
> No worries, it will just be for next release

Thanks in advance

sebanc, 2020-05-13:
Hi,

I created a new "brunch-testing" repository and uploaded a test release there, could you try it when you have the time ?

geofrey89, 2020-05-13:
Thanks
Will try it later in the day and give you feedback

geofrey89, 2020-05-13:
Thanks ... sebanc

> Hi,
> 
> I created a new "brunch-testing" repository and uploaded a test release there, could you try it when you have the time ?

Thanks sebanc..
I have updated to the testing version.. the buttons are working......there is the issue of battery indicating charging but instead it is discharging.  ...will send dmesg config shortly......

geofrey89, 2020-05-13:
[dmesg.txt](https://github.com/sebanc/brunch/files/4622792/dmesg.txt)

geofrey89, 2020-05-17:
hi @sebanc  found out the problem was the charging cable ., I replaced it and all is well....So far the buttons are working well,.
Thanks again.

Not working now 
    --camera~avstream 2500-- tried to research ,,I even tested some Linux distros. Non had the drivers for the camera..
Hope drivers will be available..
   --fingerprint

sebanc, 2020-05-18:
Unfortunately the camera and fingerprint will not work with chromeos, I cannot really help further with this

geofrey89, 2020-05-18:
> Unfortunately the camera and fingerprint will not work with chromeos, I cannot really help further with this

>@sebanc  thanks again for making this possible.

>Bluetooth update.  I paired my phone with Bluetoothctl 
>I can unlock my tablet  pc  when I unlock the phone..

geofrey89, 2020-05-18:
> Unfortunately the camera and fingerprint will not work with chromeos, I cannot really help further with this

Thanks @sebanc

closed 2020-05-22 by geofrey89