sebanc/brunch #106: Touchpad and gestures not working on Asus 556UB:

Zece88, 2020-05-04:
Hi @sebanc 
[dmesg.txt](https://github.com/sebanc/brunch/files/4576371/dmesg.txt)
As requested.

sebanc, 2020-05-05:
Hi,

Could you post the output of "udevadm info /sys/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-7/i2c-FTE1001:00/0018:0B05:0101.0007/input/input27" ?

Also, for each device that you have with a name like /dev/input/mouseX, could you post the output of "udevadm info /dev/input/mouseX" ?

Zece88, 2020-05-05:
Sorry for being a noob on crosh shell. Can you explain the command?. Thx.

sebanc, 2020-05-05:
No worries, another way:
- open crosh shell - type shell
- run "ls //dev/input/mouse*"
- For each line that you have in the output of the previous command run: "udevadm info <line_from_previous_command>"

Zece88, 2020-05-06:
Here you go:

chronos@localhost / $ udevadm info /dev/input/mouse0
P: /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.2/0003:046D:C52B.0003/0003:046D:101A.0007/input/input37/mouse0
N: input/mouse0
S: input/by-id/usb-Logitech_USB_Receiver-if02-mouse
S: input/by-path/pci-0000:00:14.0-usb-0:2:1.2-mouse
E: DEVLINKS=/dev/input/by-id/usb-Logitech_USB_Receiver-if02-mouse /dev/input/by-path/pci-0000:00:14.0-usb-0:2:1.2-mouse
E: DEVNAME=/dev/input/mouse0
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.2/0003:046D:C52B.0003/0003:046D:101A.0007/input/input37/mouse0
E: ID_BUS=usb
E: ID_INPUT=1
E: ID_INPUT_MOUSE=1
E: ID_MODEL=USB_Receiver
E: ID_MODEL_ENC=USB\x20Receiver
E: ID_MODEL_ID=c52b
E: ID_PATH=pci-0000:00:14.0-usb-0:2:1.2
E: ID_PATH_TAG=pci-0000_00_14_0-usb-0_2_1_2
E: ID_REVISION=1201
E: ID_SERIAL=Logitech_USB_Receiver
E: ID_TYPE=hid
E: ID_USB_DRIVER=usbhid
E: ID_USB_INTERFACES=:030101:030102:030000:
E: ID_USB_INTERFACE_NUM=02
E: ID_VENDOR=Logitech
E: ID_VENDOR_ENC=Logitech
E: ID_VENDOR_ID=046d
E: MAJOR=13
E: MINOR=32
E: POWERD_ROLE=external_input
E: POWERD_TAGS=usable_when_docked usable_when_laptop usable_when_tablet usable_when_display_off role_external_input
E: POWERD_TAGS_USABLE=usable_when_docked usable_when_laptop usable_when_tablet usable_when_display_off
E: SUBSYSTEM=input
E: TAGS=:powerd:
E: UDISKS_PRESENTATION_HIDE=0
E: USEC_INITIALIZED=43196040

chronos@localhost / $ udevadm info /dev/input/mouse1
P: /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.2/0003:046D:C52B.0003/0003:046D:4041.0009/input/input40/mouse1
N: input/mouse1
S: input/by-id/usb-Logitech_USB_Receiver-if02-mouse
S: input/by-path/pci-0000:00:14.0-usb-0:2:1.2-mouse
E: DEVLINKS=/dev/input/by-path/pci-0000:00:14.0-usb-0:2:1.2-mouse /dev/input/by-id/usb-Logitech_USB_Receiver-if02-mouse
E: DEVNAME=/dev/input/mouse1
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.2/0003:046D:C52B.0003/0003:046D:4041.0009/input/input40/mouse1
E: ID_BUS=usb
E: ID_INPUT=1
E: ID_INPUT_KEY=1
E: ID_INPUT_KEYBOARD=1
E: ID_INPUT_MOUSE=1
E: ID_MODEL=USB_Receiver
E: ID_MODEL_ENC=USB\x20Receiver
E: ID_MODEL_ID=c52b
E: ID_PATH=pci-0000:00:14.0-usb-0:2:1.2
E: ID_PATH_TAG=pci-0000_00_14_0-usb-0_2_1_2
E: ID_REVISION=1201
E: ID_SERIAL=Logitech_USB_Receiver
E: ID_TYPE=hid
E: ID_USB_DRIVER=usbhid
E: ID_USB_INTERFACES=:030101:030102:030000:
E: ID_USB_INTERFACE_NUM=02
E: ID_VENDOR=Logitech
E: ID_VENDOR_ENC=Logitech
E: ID_VENDOR_ID=046d
E: MAJOR=13
E: MINOR=33
E: POWERD_ROLE=external_input
E: POWERD_TAGS=usable_when_docked usable_when_laptop usable_when_tablet usable_when_display_off role_external_input
E: POWERD_TAGS_USABLE=usable_when_docked usable_when_laptop usable_when_tablet usable_when_display_off
E: SUBSYSTEM=input
E: TAGS=:powerd:
E: UDISKS_PRESENTATION_HIDE=0
E: USEC_INITIALIZED=43218499

chronos@localhost / $ udevadm info /dev/input/mouse2
P: /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.2/1-1.2:1.2/0003:046D:C52B.0006/0003:046D:4041.000A/input/input42/mouse2
N: input/mouse2
S: input/by-id/usb-Logitech_USB_Receiver-if02-mouse
S: input/by-path/pci-0000:00:14.0-usb-0:1.2:1.2-mouse
E: DEVLINKS=/dev/input/by-id/usb-Logitech_USB_Receiver-if02-mouse /dev/input/by-path/pci-0000:00:14.0-usb-0:1.2:1.2-mouse
E: DEVNAME=/dev/input/mouse2
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.2/1-1.2:1.2/0003:046D:C52B.0006/0003:046D:4041.000A/input/input42/mouse2
E: ID_BUS=usb
E: ID_INPUT=1
E: ID_INPUT_KEY=1
E: ID_INPUT_KEYBOARD=1
E: ID_INPUT_MOUSE=1
E: ID_MODEL=USB_Receiver
E: ID_MODEL_ENC=USB\x20Receiver
E: ID_MODEL_ID=c52b
E: ID_PATH=pci-0000:00:14.0-usb-0:1.2:1.2
E: ID_PATH_TAG=pci-0000_00_14_0-usb-0_1_2_1_2
E: ID_REVISION=2401
E: ID_SERIAL=Logitech_USB_Receiver
E: ID_TYPE=hid
E: ID_USB_DRIVER=usbhid
E: ID_USB_INTERFACES=:030101:030102:030000:
E: ID_USB_INTERFACE_NUM=02
E: ID_VENDOR=Logitech
E: ID_VENDOR_ENC=Logitech
E: ID_VENDOR_ID=046d
E: MAJOR=13
E: MINOR=34
E: POWERD_ROLE=external_input
E: POWERD_TAGS=usable_when_docked usable_when_laptop usable_when_tablet usable_when_display_off role_external_input
E: POWERD_TAGS_USABLE=usable_when_docked usable_when_laptop usable_when_tablet usable_when_display_off
E: SUBSYSTEM=input
E: TAGS=:powerd:
E: UDISKS_PRESENTATION_HIDE=0
E: USEC_INITIALIZED=43218544

chronos@localhost / $ udevadm info /dev/input/mouse3
P: /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-7/i2c-FTE1001:00/0018:0B05:0101.000C/input/input41/mouse3
N: input/mouse3
S: input/by-path/pci-0000:00:15.1-platform-i2c_designware.1-mouse
E: DEVLINKS=/dev/input/by-path/pci-0000:00:15.1-platform-i2c_designware.1-mouse
E: DEVNAME=/dev/input/mouse3
E: DEVPATH=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-7/i2c-FTE1001:00/0018:0B05:0101.000C/input/input41/mouse3
E: ID_INPUT=1
E: ID_INPUT_TOUCHPAD=1
E: ID_PATH=pci-0000:00:15.1-platform-i2c_designware.1
E: ID_PATH_TAG=pci-0000_00_15_1-platform-i2c_designware_1
E: ID_SERIAL=noserial
E: MAJOR=13
E: MINOR=35
E: POWERD_ROLE=internal_touchpad
E: POWERD_TAGS=usable_when_laptop usable_when_display_off role_internal_touchpad
E: POWERD_TAGS_USABLE=usable_when_laptop usable_when_display_off
E: SUBSYSTEM=input
E: TAGS=:powerd:
E: USEC_INITIALIZED=43342028

Zece88, 2020-05-12:
Hi @sebanc , you forgot about me :-) ?

sebanc, 2020-05-12:
Hi @Zece88 

No, don't worry :) I actually saw that I have missing config in my kernel which might be related to your issue, I could not add it directly in the stable release as I was not sure of its impact but I will include it in the next version and ask you to test.

Zece88, 2020-05-12:
Very nice. Thank you so much

sebanc, 2020-05-13:
Hi,

I created a new "brunch-testing" repository and uploaded a test release there, could you try it when you have the time ?

Zece88, 2020-05-13:
Hi,

Just tried the new test build.

**Working:** 
Left button click.
1 finger touch so I can click/mark objects.

**Still not working:**
Right click.
2 finger scroll.
Gestures.

**Also a thing I noticed:**
If I stress test the touchpad (clicking/touch) the screen goes black for a bit (some seconds) before pic is back.

sebanc, 2020-05-14:
Could you try adding "i8042.nomux=0" or "i8042.nomux=1" to the kernel command line ?

Zece88, 2020-05-14:
Hi again. How to do that?

sebanc, 2020-05-14:
Depends if you use the single boot or dual boot method:
- single boot: run "sudo edit-grub-config"  and add "i8042.nomux=0" after cros_debug
- dual boot: add "i8042.nomux=0" to the grub config you have created after cros_debug

Zece88, 2020-05-14:
Tried both now.

Added line after cros_debug, saved with ctrl o and enter, restarted.
Not sure it did anything.

**Working:**
*Left button click.
*1 finger touch so I can click/mark objects.

**Still not working:**
*Right click.
*2 finger scroll.
*Gestures.
*If I stress test the touchpad (clicking/touch) the screen goes black for a bit (some seconds) before pic is back.

[image]

sebanc, 2020-05-15:
Hi,

Could you run "sudo nano /etc/gesture/20-mouse.conf" and replace on line 5:
`    Driver          "cmt"`
by
`    Driver          "evdev"`

Then reboot and see if something changed.

jesusvito, 2020-05-15:
> Hi,
> 
> Could you run "sudo nano /etc/gesture/20-mouse.conf" and replace on line 5:
> ` Driver "cmt"`
> by
> ` Driver "evdev"`
> 
> Then reboot and see if something change

Hello... i've followed everything in this post but no changes... still not working right click and gestures

sebanc, 2020-05-15:
@jesusvito Hi, thanks for testing this, could you try this as well:

Rplace the first block in /etc/gesture/40-touchpad-cmt.conf with:
```
Section "InputClass"
    Identifier      "touchpad"
    MatchIsTouchpad "on"
    MatchDevicePath "/dev/input/event*"
    Driver          "evdev"
    Option          "Tap Minimum Pressure" "0"
    Option          "AccelerationProfile" "-1"
    Option          "Scroll Buttons" "0"
    Option          "Scroll Axes" "1"

    # CMT devices potentially process keyboard events
    Option          "XkbModel" "pc"
    Option          "XkbLayout" "us"
EndSection
```

jesusvito, 2020-05-15:
I'm on it! by the way i should tell you this is awesome just what i've been looking for... thank you very much for your support!

jesusvito, 2020-05-15:
still not working my friend...

sebanc, 2020-05-15:
Happy you like it, I am having a hard time understanding this touchpad issue.

Could you post the output of "lsmod" command in crosh shell ?

jesusvito, 2020-05-15:
sure thing just give me a sec

swarshall, 2020-05-15:
Hello guys, I followed all the instructions to operate the touchpad on Asus E200ha (it's the same hardware) but the gesture still does not work.

Thanks for brunch the rest works fine :)

jesusvito, 2020-05-15:
crosh> shell
chronos@localhost / $ lsmod
Module                  Size  Used by
cmac                   16384  0
rfcomm                 61440  4
wl                   6377472  0
uinput                 20480  0
btusb                  53248  0
btrtl                  20480  1 btusb
btbcm                  16384  1 btusb
btintel                24576  1 btusb
bluetooth             458752  34 btrtl,btintel,btbcm,btusb,rfcomm
ecdh_generic           20480  1 bluetooth
nf_nat_tftp            16384  0
nf_conntrack_tftp      16384  1 nf_nat_tftp
nf_nat_ftp             16384  0
nf_conntrack_ftp       16384  1 nf_nat_ftp
esp6                   20480  0
ah6                    16384  0
xfrm6_mode_tunnel      16384  0
xfrm6_mode_transport    16384  0
xfrm4_mode_tunnel      16384  0
xfrm4_mode_transport    16384  0
ip6t_REJECT            16384  0
ip6t_ipv6header        16384  0
psmouse               135168  0
sr_mod                 20480  0
cdrom                  36864  1 sr_mod
intel_rapl             20480  0
x86_pkg_temp_thermal    16384  0
intel_powerclamp       16384  0
cfg80211              577536  1 wl
snd_hda_codec_hdmi     49152  1
dell_wmi_descriptor    16384  0
acer_wmi               24576  0
sparse_keymap          16384  1 acer_wmi
wmi_bmof               16384  0
wmi                    24576  3 acer_wmi,wmi_bmof,dell_wmi_descriptor
snd_hda_codec_realtek    98304  1
snd_hda_codec_generic    69632  1 snd_hda_codec_realtek
snd_hda_intel          36864  3
snd_intel_nhlt         16384  1 snd_hda_intel
snd_hda_codec         110592  4 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec_realtek
snd_hda_core           65536  5 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek
snd_hwdep              16384  1 snd_hda_codec
mei_me                 24576  0
mei                    65536  1 mei_me
r8169                  73728  0
realtek                20480  1
zram                   24576  1
i915                 1474560  12
intel_gtt              20480  1 i915
video                  40960  2 acer_wmi,i915
joydev                 24576  0
input_leds             16384  0

THATS WHAT I GOT BRO

sebanc, 2020-05-15:
Some more things to try:

Replace the first block in /etc/gesture/40-touchpad-cmt.conf with:
```
Section "InputClass"
    Identifier      "touchpad"
    MatchIsTouchpad "on"
    MatchDevicePath "/dev/input/event*"
    Driver          "evdev"
    Option          "Tap Minimum Pressure" "0"
    Option          "AccelerationProfile" "-1"
    Option          "Scroll Buttons" "0"
    Option          "Scroll Axes" "1"
    Option          "Max Allowed Pressure Change Per Sec" "100000.0"
    Option          "Max Hysteresis Pressure Per Sec" "100000.0"

    # CMT devices potentially process keyboard events
    Option          "XkbModel" "pc"
    Option          "XkbLayout" "us"
EndSection
```

sebanc, 2020-05-15:
Another one, add "module_blacklist=psmouse" to the kernel command line

jesusvito, 2020-05-15:
> Another one, add "module_blacklist=psmouse" to the kernel command line

what do you mean? sudo edit-grub-conf. ?

sebanc, 2020-05-15:
> > Another one, add "module_blacklist=psmouse" to the kernel command line
> 
> what do you mean? sudo edit-grub-conf. ?

When it is just to try once, press "e" quickly when you see grub and add it manually after cros_debug

swarshall, 2020-05-15:
Sorry but it's not working for me

jesusvito, 2020-05-15:
> > > Another one, add "module_blacklist=psmouse" to the kernel command line
> > 
> > 
> > what do you mean? sudo edit-grub-conf. ?
> 
> When it is just to try once, press "e" quickly when you see grub and add it manually after cros_debug

WHAT IF I HAVE THE BROADCOM_WL OPTIONS ALREADT AFTER CROS_DEBUG?

sebanc, 2020-05-15:
either before or after options=broadcom_wl

jesusvito, 2020-05-15:
> either before or after options=broadcom_wl

got it

sebanc, 2020-05-15:
Again another one, replace the first block in /etc/gesture/40-touchpad-cmt.conf with:
```
Section "InputClass"
    Identifier      "touchpad"
    MatchIsTouchpad "on"
    MatchDevicePath "/dev/input/event*"
    Driver          "cmt"
    Option          "Tap Minimum Pressure" "0"
    Option          "AccelerationProfile" "-1"
    Option          "Scroll Buttons" "0"
    Option          "Scroll Axes" "1"

    # CMT devices potentially process keyboard events
    Option          "XkbModel" "pc"
    Option          "XkbLayout" "us"
EndSection
```

jesusvito, 2020-05-15:
> Another one, add "module_blacklist=psmouse" to the kernel command line

This one killed the remaining functionalities of the trackpad (touchpad didn't work at all) had to use a regular usb mouse... still no luck

jesusvito, 2020-05-15:
> Again another one, replace the first block in /etc/gesture/40-touchpad-cmt.conf with:
>
> ```
> Section "InputClass"
>     Identifier      "touchpad"
>     MatchIsTouchpad "on"
>     MatchDevicePath [...]

isn't "cmt" for apple mouse?

jesusvito, 2020-05-15:
> Again another one, replace the first block in /etc/gesture/40-touchpad-cmt.conf with:
> 
> ```
> Section "InputClass"
>     Identifier      "touchpad"
>     MatchIsTouchpad "on"
>     MatchDevicePath "/dev/input/event*"
>     Driver          "cmt"
>     Option          "Tap Minimum Pressure" "0"
>     Option          "AccelerationProfile" "-1"
>     Option          "Scroll Buttons" "0"
>     Option          "Scroll Axes" "1"
>     Option          "Max Allowed Pressure Change Per Sec" "100000.0"
>     Option          "Max Hysteresis Pressure Per Sec" "100000.0"
> 
>     # CMT devices potentially process keyboard events
>     Option          "XkbModel" "pc"
>     Option          "XkbLayout" "us"
> EndSection
> ```
YOU ARE DEFINETLY UP TO SOMETHING!!! TWO FINGERS AT THE SAME TIME ON THE TRACKPAD GIVES YOU "RIGHT CLICK" OPTIONS SCROLLING WORKING FINE WITH TWO FINGERS... NO MORE GESTURES THOUGH

sebanc, 2020-05-15:
Ah that's nice, finally some progress...

What are the remaining missing gestures ?

jesusvito, 2020-05-15:
> Ah that's nice, finally some progress...
> 
> What are the remaining missing gestures ?

as far as i care the only essential thing that was missing was just the right click options... also i believe the two finger scroll is "inverted" up goes up and down goes down... which i think it might be backwards not sure though... on windows up goes down and down goes up while scrolling

swarshall, 2020-05-15:
For me it's pointer movement

jesusvito, 2020-05-15:
> For me it's pointer movement

it does feel just a bit funny but pretty seamless to me...

if you mean speed you can adjust that on the regular settings / mouse Menu

swarshall, 2020-05-15:
> > For me it's pointer movement
> 
> it does feel just a bit funny but pretty seamless to me...
> 
> if you mean speed you can adjust that on the regular settings / mouse Menu

No I mean that when I move with my fingers the cursor does not move

jesusvito, 2020-05-15:
> Ah that's nice, finally some progress...
> 
> What are the remaining missing gestures ?

Just found out when you use australian mode you get the scrolling working right... up goes down and down goes up... Thank god for the guys down under...

Also, very important... I got a new gesture working... scrolling / swiping left to right or right to left with two fingers you get "back" or forward" fucntionalities... Awesome!

jesusvito, 2020-05-15:
> > > For me it's pointer movement
> > 
> > 
> > it does feel just a bit funny but pretty seamless to me...
> > if you mean speed you can adjust that on the regular settings / mouse Menu
> 
> No I mean that when I move with my fingers the cursor does not move

did you add this? module_blacklist=psmouse" to the kernel command line? because it disabled the trackpad for me :S

swarshall, 2020-05-15:
> > > > Pour moi, c'est le mouvement du pointeur
> > > 
> > > 
> > > cela me semble un peu drôle mais assez transparent pour moi ... 
> > > si vous voulez dire la vitesse, vous pouvez l'ajuster dans le menu des paramètres / souris standard
> > 
> > 
> > Non je veux dire que quand je bouge avec mes doigts le curseur ne bouge pas
> 
> avez-vous ajouté cela? module_blacklist = psmouse "à la ligne de commande du noyau? car il a désactivé le trackpad pour moi: S

I had to remove it I must put it back in edit-grub-config?

jesusvito, 2020-05-15:
remove that...

it should look like this

  v4.9-98-gf9ff902b /root/tmpgrub/efi/boot/grub.cfg  
set timeout=2

menuentry "ChromeOS" {
  regexp --set disk "(^.+)(,gpt)" $root
  linux ($disk,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug
  initrd ($disk,7)/initramfs.img

swarshall, 2020-05-15:
It's not that

sebanc, 2020-05-15:
@swarshall I think you have a totally different issue.

Could you open another issue and add attach a dmesg (run "dmesg > ~/Downloads/dmesg.txt in crosh shell and attach dmesg.txt from your downloads folder)

Also first make sure you are using the latest brunch release.

swarshall, 2020-05-15:
> @swarshall Je pense que vous avez un problème totalement différent.
> 
> Pourriez-vous ouvrir un autre problème et ajouter attacher un dmesg (exécutez "dmesg> ~ / Downloads / dmesg.txt dans le shell crosh et attachez dmesg.txt à partir de votre dossier de téléchargements)
> 
> Assurez-vous également que vous utilisez la dernière version de brunch.

Hi, I'm using the latest test version but can I switch to stable?

sebanc, 2020-05-15:
@swarshall yes please update first and then open the issue with a dmesg

sebanc, 2020-05-15:
@jesusvito I cannot apply this patch as it is right now in brunch as it may break things, could you confirm if replacing the first block with those 2 works the same:
```
Section "InputClass"
    Identifier      "touchpad"
    MatchIsTouchpad "on"
    MatchDevicePath "/dev/input/event*"
    Driver          "cmt"
    Option          "Tap Minimum Pressure" "0"
    Option          "AccelerationProfile" "-1"
    Option          "Scroll Buttons" "0"
    Option          "Scroll Axes" "1"

    # CMT devices potentially process keyboard events
    Option          "XkbModel" "pc"
    Option          "XkbLayout" "us"
EndSection

Section "InputClass"
    Identifier      "CMT for Logitech Touchpad"
    MatchUSBID      "046d:*"
    MatchIsTouchpad "on"
    MatchDevicePath "/dev/input/event*"
    Option          "Max Allowed Pressure Change Per Sec" "100000.0"
    Option          "Max Hysteresis Pressure Per Sec" "100000.0"
EndSection
```

jesusvito, 2020-05-15:
Zoom pinch in and punch out working fine with two fingers

jesusvito, 2020-05-15:
i'll try later on i should get some sleep hahaha i'm in Mexico. thanks for the all your help my friend... ill keep trying stuff out later. thanks m8

Zece88, 2020-05-15:
Hi,

I tried many of the suggestions. Still no changes :-(

**Working:**
Left button click.
1 finger touch so I can click/mark objects.

**Still not working:**
Right click.
2 finger scroll.
Gestures.

**Also a thing I noticed:**
If I stress test the touchpad (clicking/touch) the screen goes black for a bit (some seconds) before pic is back.

sebanc, 2020-05-15:
>

@Zece88 This one did not work ? You seem to have nearly the same touchpad and the same issue. Could you try on a clean install just in case ?

Zece88, 2020-05-15:
I'll try a clean install with r81 stable and try your suggestion again. Thanks for support

Zece88, 2020-05-15:
Just tried clean rr81 stable install and edited first block in /etc/gesture/40-touchpad-cmt.conf. 

Nothing changed i'm afraid.

**Working:**
Left button click.
1 finger touch so I can click/mark objects.

**Still not working:**
Right click.
2 finger scroll.
Gestures.

**Also a thing I noticed:**
If I stress test the touchpad (clicking/touch) the screen goes black for a bit (some seconds) before pic is back.

sebanc, 2020-05-16:
Hi @Zece88, thanks for testing on a clean install. Could you try this same config but changing "Max Allowed Pressure Change Per Sec" and "Max Hysteresis Pressure Per Sec" to other values like 1000 or 1000000 ?

Zece88, 2020-05-16:
Hi again,

Just tried , and no changes.

**Working:**
Left button click.
1 finger touch so I can click/mark objects.

**Still not working:**
Right click.
2 finger scroll.
Gestures.

**Also a thing I noticed:**
If I stress test the touchpad (clicking/touch) the screen goes black for a bit (some seconds) before pic is back.

sebanc, 2020-05-16:
It's really difficult for me to debug this issue as I don't have such hardware. On the kernel side, everything seems ok but the cmt touchpad driver probably needs some specific options.

I have just found this repo from someone who ported the cmt driver to linux:
https://github.com/hugegreenbug/xf86-input-cmt/tree/master/xorg-conf

Could you try some of the configs that are there, for example for 50-touchpad-cmt-elan.conf:
```
# Configure touchpads to use Chromium Multitouch (cmt) X input driver
Section "InputClass"
    Identifier      "touchpad elan"
    MatchIsTouchpad "on"
    MatchDevicePath "/dev/input/event*"
    Option          "Touchpad Stack Version" "1"
    Option          "Horizontal Resolution" "33"
    Option          "Vertical Resolution" "33"
    Option          "Integrated Touchpad" "1"
    Option          "Input Queue Delay" "0.017"
    Option          "Two Finger Vertical Close Distance Thresh" "35.0"
    Option          "Fling Buffer Suppress Zero Length Scrolls" "0"
EndSection
```
Take the "Option" lines:
```
    Option          "Touchpad Stack Version" "1"
    Option          "Horizontal Resolution" "33"
    Option          "Vertical Resolution" "33"
    Option          "Integrated Touchpad" "1"
    Option          "Input Queue Delay" "0.017"
    Option          "Two Finger Vertical Close Distance Thresh" "35.0"
    Option          "Fling Buffer Suppress Zero Length Scrolls" "0"
```

And add them to the main config in the first section of /etc/gesture/40-touchpad-cmt.conf.

Important: do not replace the full section, just add the options you want to try as the config misses some chromeos features

Zece88, 2020-05-16:
Good news!

I finally have it working (almost perfect) with your big help.

**What I did:**

Running R81 stable
Rammus recovery

`**50-touchpad-cmt-elan.conf:**

# Configure touchpads to use Chromium Multitouch (cmt) X input driver
Section "InputClass"
    Identifier      "touchpad elan"
    MatchIsTouchpad "on"
    MatchDevicePath "/dev/input/event*"
    Option          "Touchpad Stack Version" "1"
    Option          "Horizontal Resolution" "33"
    Option          "Vertical Resolution" "33"
    Option          "Integrated Touchpad" "1"
    Option          "Input Queue Delay" "0.017"
    Option          "Two Finger Vertical Close Distance Thresh" "35.0"
    Option          "Fling Buffer Suppress Zero Length Scrolls" "0"
EndSection

**40-touchpad-cmt-elan.conf:**

Take the "Option" lines:

    Option          "Touchpad Stack Version" "1"
    Option          "Horizontal Resolution" "33"
    Option          "Vertical Resolution" "33"
    Option          "Integrated Touchpad" "1"
    Option          "Input Queue Delay" "0.017"
    Option          "Two Finger Vertical Close Distance Thresh" "35.0"
    Option          "Fling Buffer Suppress Zero Length Scrolls" "0"

And add them to the main config in the first section of /etc/gesture/40-touchpad-cmt.conf.`

**Still not working:**
Only thing left is my right click. Right now it works when touch with 2x fingers.

harun911, 2020-05-16:
Zece88, absolute legend. I was fiddling with the settings for hours, when I came across this thread. You're suggested changes to those files actually brought movement to my not working touchpad. Thank you so much sebanc and Zece88.

sebanc, 2020-05-17:
@Zece88 Happy to see that it is fixed (when gestures are enabled, right click is disabled, it is normal in chromeos). However some work remains, you probably don't need the whole config but maybe just one or two lines to have it working. Could you try to figure out which lines are the minimum necessary for touchpad to work ?

@harun911 Same for you, the extra unneeded lines might be related to cursor latency. Also, could you post the output of the command "dmesg | grep input" in crosh shell ?

[13 more replies]

closed 2020-06-02 by sebanc