sebanc/brunch #107: Android and linux apps are blurry, not displaying in hdpi mode

Campbellding, 2020-05-05:
Successfully installed on my intel nuc8, everything works well so far but seems like the android and linux apps don't support hdpi mode. I'm using a 4k monitor and OS can change display size to e.g. 150% (like how you do in windows), but the android and linux app will be become blurry. 

I'm not sure it's the limitation of chromeOS as there's no issue on my real chromebook.

sebanc, 2020-05-06:
Hi,

I found some configs related to HDMI which should probably have been enabled, I hope it will fix your issue in the next release.

I have a totally unrelated question that maybe you could answer as you have a real chromebook. If you have the telegram android app installed on it, could you tell me if clicking telegram notifications opens the app or not ? (with brunch it is possible to use quick reply or dismiss but the app will not open)
If you don't use the telegram, you can ignore this question.

sebanc, 2020-05-09:
Hi, I have uploaded the new build, could you try it ?

Campbellding, 2020-05-09:
Hi @sebanc , thanks for updating the release. Unfortunately, it doesn't work. Here is my find out:

Clear:
- The chrome browser and all the web apps
-  The normal linux apps

Blurry:
- All the android apps
- The electron based linux apps, e.g. vscode, gitkraken..

[image]

Mou786Aha, 2022-06-19:
> Hi @sebanc , thanks for updating the release. Unfortunately, it doesn't work. Here is my find out:
> 
> Clear:
> 
> * The chrome browser and all the web apps
> * The normal linux apps, e.g. sublime text
> 
> Blurry:
> 
> * All the android apps
> * The electron based linux apps, e.g. vscode, gitkraken..
> 
> [image]

Any updates? Did you fix it?