sebanc/brunch #108: Fails to boot with 500GB Samsung T5

matthewcmead, 2020-05-05:
When creating the install directly to a 500GB Samsung T5, the drive fails to boot with a message that it can’t find chromeos. It seems to fail to scan sda.

When I build a 384GiB image file and dd it onto the Samsung T5 it works as expected. Using latest r81 release with an up to date r80 rammus recovery image.

Any ideas?

sebanc, 2020-05-05:
Hi,

Sorry but i don't understand your issue, I think I need clarification on the below points:
- "When creating the install directly to a 500GB" do you mean full hdd install or dual boot image ?
- "When I build a 384GiB image file and dd it onto the Samsung T5" Is it the same drive ? If not how are they different ?

matthewcmead, 2020-05-12:
The drive is the same in both cases.  If I build the image directly to /dev/sda (which is the block device for a 500GB Samsung T5), in other words, a full hdd install, the failure occurs.

When I build a 384GiB image file and dd it onto the *exact same 500GB Samsung T5* it boots and works fine.

The only difference is in the preparation of the output — in the working scenario, I build the image to a (smaller) intermediate file, and dd it to the block device.  In the non-working scenario, I build the image directly to the block device, and it fails to boot.

sebanc, 2020-05-12:
Hi,

Thanks for the clarification, I think I fully understand the issue now.

If I understand correctly, the install script without erros but after it would not boot ?

sebanc, 2020-06-02:
No news for a while on this issue so I will close it, don't hesitate to re-open it if needed

closed 2020-06-02 by sebanc

matthewcmead, 2020-06-17:
Yes, sorry for the delay.  The script finishes without errors but it will not boot.  As mentioned, if I make a 384GiB output *FILE*, then dd that to the Samsung T5, it does boot properly.