sebanc/brunch #109: [Feature Request] Grub boot splash

hussainalhaddad, 2020-05-05:
Hi Sebanc

Thanks you so much for your work and support!

Just wondering if it's possible to add the official boot splash to grub, or at least an option to make grub quiet and hide the log.

Thanks

sebanc, 2020-05-05:
Hi @hussainalhaddad 

For now, you can make boot quiet by adding "console=" to the kernel command line (after cros_debug).

Aside from this, it's not really in the top of my priority list to make the boot looks nicer, however if the community comes up with a nice boot splash, I don't mind implementing it.

GRUB should already contain the necessary modules in order to add a bootspash, so I will let this issue opened in hope that someone takes care of this.

clandestinz, 2020-05-14:
how about a spinning chrome icon?