sebanc/brunch #111: grub error

ghritik, 2020-05-06:
the error is like :> 
error: no such device: /chromeos.img
error: file ./chromeos.img not found 
error: disk `loop,gpt7` not found
error: you need to load the kernel first... 

plz how to fix this error ;;; i am currently using ubuntu 20.04
the grub code is -> 
--------------------
img_part=/dev/sda5
img_path=/chromeos.img
search --no-floppy --set=root --file $img_path
loopback loop $img_path
linux (loop,gpt7) /kernel boot=local noresume noswap loglevel=7 disablevmx=off \ cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path
initrd (loop,gpt7) /initramfs.img
____________

sebanc, 2020-05-06:
This error means that your chromeos file is either:
- not named "chromeos.img" (check if there is not a typo)
- not located at the root of the partition

Lucaacer, 2020-05-06:
Hi Sebanc, I don't mean to spam your thread, so please remove my post if you think it's off topic.

In my opinion, if someone is already a linux user, the easiest way of dual booting chrome OS and linux on the main drive is:

- installing chrome OS alone as per your instructions;
- booting a linux live usb (I used ubuntu 20.04) and resizing the bigger partition of chrome OS (stateful, if I'm not mistaken) with gparted;
- creating and formatting a new partition to ext4;
- rebooting once to let the framework rebuild chrome OS according to the smaller partition where it's now running;
- rebooting a live linux usb and installing ubuntu (in my case) in the new partition, it has got a GUI so it's very easy;
- removing the quiet grub attribute if required and when you reboot the device you should have the usual choices of ubuntu and even the one pointing to chrome OS (which you can actualy rename to chrome OS) which will run a separate grub and then let chrome OS start.

It might look a longer way, but does not require the user to understand img mounting.

At least, it looks easier in my opinion:)

sebanc, 2020-05-06:
Hi,

The "official" way is way more secure in my opinion as it keeps the official partition table structure untouched and uses a different block device so that chromeos won't touch linux / windows partitions.

You can use your method as well but I will not provide support for it as any script in chromeos can see that the partition layout is abnormal and perform any weird action.

ijinesh, 2020-05-06:
Hi

> This error means that your chromeos file is either:
> 
>     * not named "chromeos.img" (check if there is not a typo)
> 
>     * not located at the root of the partition

Hi Sebanc,

Thanks for your great work. I am facing the same problem as ghritik. There is no typo in the name or the location. Here's the script in Grub Customizer:

	img_part=/dev/mmcblk0p3
	img_path=/chromos.img
	search --no-floppy --set=root --file $img_path
	loopback loop $img_path
	linux (loop,gpt7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
		cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path
	initrd (loop,gpt7)/initramfs.img

Chromos.img is there in /Dev/mmcblk0p3.

Kindly help.

Also, I tried installing Chrome OS on USB and had encountered error while running command from terminal where recovery image was. Command was "sudo bash chromeos-install.sh -src rammus_recovery.bin -dst /dev/sda1
Error was:

	img_part=/dev/mmcblk0p3
	img_path=/chromos.img
	search --no-floppy --set=root --file $img_path
	loopback loop $img_path
	linux (loop,gpt7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
		cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path
	initrd (loop,gpt7)/initramfs.img

Lucaacer, 2020-05-06:
> Hi,
>
> The "official" way is way more secure in my opinion as it keeps the official partition table structure untouched and uses a different block device so that chromeos won't touch linux / windows [...]

Ok, thanks, that's the point!

To be true, when I resized the partition and rebooted, your script actually rebuilt everything flawlessly, but I haven't gone yet the update route, so you're very likely right. Anyway, I will be happy to act as a Guinea pig when a recovery for chromos 81 comes up and I will report:)

sebanc, 2020-05-06:
> Hi
>
>
> Hi Sebanc,
>
> Thanks for your great work. I am facing the same problem as ghritik. There is no typo in the name or the location. Here's the script in Grub Customizer:
>
> ```
> img_part=/dev/mmcblk0p3 [...]

Are you sure the image name is not "chromeos.img" with an "e" as in the example commands ?

> Also, I tried installing Chrome OS on USB and had encountered error while running command from terminal where recovery image was. Command was "sudo bash chromeos-install.sh -src rammus_recovery.bin [...]

The command ""sudo bash chromeos-install.sh -src rammus_recovery.bin -dst /dev/sda1" is incorrect, -dst should point to a block device and not a partition so /dev/sda instead of /dev/sda1.

ijinesh, 2020-05-06:
> Are you sure the image name is not "chromeos.img" with an "e" as in the example commands ?
Yes it is 'Chromos.img' and it is same name in the Grub as well as at /dev/mmcblk0p3.

Thanks for pointing the linux command error...I am not good at Linux. I will try.

Jeremy97L, 2020-05-06:
> This error means that your chromeos file is either:
> 
> * not named "chromeos.img" (check if there is not a typo)
> * not located at the root of the partition

I have a similar issue as OP, but the error I get when I want to boot my Chrome OS on partition /dev/sda6 is the following:

_error: unknown filesystem.
error: you need to load the kernel first_

The img file is correctly named chromeos.img. It is located at the root of the partition. I tried both ext4 and ntfs for this partition. I also tried with both secure boot on and off. The partition is around 250 GB. My laptop has UEFI enabled and I used the rammus recovery with the latest testing brunch and the latest stable one.

My script in grub is copied from the terminal and the following:

_menuentry "ChromeOS (boot from disk image)" {
        img_part=/dev/sda6
        img_path=/chromeos.img
        search --no-floppy --set=root --file $img_path
        loopback loop $img_path
        linux (loop,gpt7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
                cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path
        initrd (loop,gpt7)/initramfs.img
}_

I have both Windows 10 and Linux Mint installed on separate partitions. I tried the script on both my usb grub and my linux mint grub with the same error. If you need any more information please let me know.

ghritik, 2020-05-06:
> This error means that your chromeos file is either:
> 
>     * not named "chromeos.img" (check if there is not a typo)
> 
>     * not located at the root of the partition

i checked after you said that ... the name is chromeos.img ... and its in the root dir.. still getting those errors... i also tried reinstalling it ... still stucked on grub

sebanc, 2020-05-07:
After reading your comments, I had a doubt on whether the script was working correctly so I reproduced the exact same and it works for me.

If the images have correctly been created and are at the right location, it is most probably an issue with the grubx64.efi binary.

Could you try replacing the "grubx64.efi" binary from your linux distro (usually found in /boot/efi/EFI/<distro_name>/)  with the attached one ?
[grubx64.efi.zip](https://github.com/sebanc/brunch/files/4590852/grubx64.efi.zip)

Jeremy97L, 2020-05-07:
> After reading your comments, I had a doubt on whether the script was working correctly so I reproduced the exact same and it works for me.
>
> If the images have correctly been created and are at the [...]

Trying your grubx64.efi.zip resulted for me in:
_error: symbol 'grub file filters_all' not found
error: unknown filesystem.
error: you need to load the kernel first_

Anything else you'd recommend trying? If you manage to have some spare time, you should consider making a youtube tutorial for the dual-boot part

sebanc, 2020-05-07:
> Trying your grubx64.efi.zip resulted for me in:
> _error: symbol 'grub file filters_all' not found error: unknown filesystem. error: you need to load the kernel first_
>
> Anything else you'd recommend [...]

How did you create the image ? Windows or linux ? Which tool did you use to write it if it was created in windows ?

sebanc, 2020-05-07:
> Trying your grubx64.efi.zip resulted for me in:
> _error: symbol 'grub file filters_all' not found error: unknown filesystem. error: you need to load the kernel first_
>
> Anything else you'd recommend [...]

The partition on which the image reside is not correctly recognized by grub. How did you create it ?

Jeremy97L, 2020-05-07:
> How did you create the image ? Windows or linux ? Which tool did you use to write it if it was created in windows ?

I created my Chrome OS USB with Linux Mint which works just fine with the latest brunch and rammus recovery. I created the chromeos.img using Chrome OS on that USB. I downloaded all the required files on there and followed your instructions. It successfully made an .img of around 14GB on my /dev/sda6 partition. I put the grub info of the terminal into both my USB grub and Linux Mint grub, both give the same error.

> The partition on which the image reside is not correctly recognized by grub. How did you create it ?

The partition was made using the gparted tool found in Linux Mint. I tried both ext4 and ntfs and it is marked as a primary partition

sebanc, 2020-05-07:
In theory, it all seems perfect and it should work.

Could you launch the usb:
- press "c" when you see grub
- type "ls"
- identify your hdd in hd0...hdN
- type "ls (hdX,gpt6)"

and post the output in this issue ?

Jeremy97L, 2020-05-07:
> 
> 
> In theory, it all seems perfect and it should work.
> 
> Could you launch the usb:
> 
>     * press "c" when you see grub
> 
>     * type "ls"
> 
>     * identify your hdd in hd0...hdN
> 
>     * type "ls (hdX,gpt6)"
> 
> 
> and post the output in this issue ?

_ls_
(proc) (hd0) (hd0,gpt12) (hd0,gpt11) (hd0,gpt10) (hd0,gpt9) (hd0,gpt8) (hd0,gpt7) (hd0,gpt6)  (hd0,gpt5) (hd0,gpt4) (hd0,gpt3) (hd0,gpt2) (hd0,gpt1) (hd1) (hd1,gpt6) (hd1,gpt5) (hd1,gpt4)  (hd1,gpt3) (hd1,gpt2) (hd1,gpt1) 
_ls (hd1,gpt6)_
Partition hd1,gpt6: Filesystem type ntfs, UUID 5719D32F0EE4FBE8 - Partition start at 485429248KiB - Total size 247144448KiB

My partition for Chrome OS is indeed around the 250 GB so this should be the one.

sebanc, 2020-05-07:
what is the output of "ls (hd1,gpt6)/" ?

Jeremy97L, 2020-05-07:
> what is the output of "ls (hd1,gpt6)/" ? (there is just a slash at the end compared to the previous one)

_ls (hd1,gpt6)/_
$AttrDef $BadClus $Bitmap $Boot $Extend/ $LogFile $MFT $MFTMirr $RECYCLE.BIN/ $Secure $UpCase $Volume chromeos.img System Volume Information/

sebanc, 2020-05-07:
can you type "loopback loop (hd1,gpt6)/chromeos.img" ?

sebanc, 2020-05-07:
and then "ls" again

Jeremy97L, 2020-05-07:
> and then "ls" again

Putting in the loopback loop command returned nothing. Trying ls again resulted in:

(proc) (loop) (loop,gpt12) (loop,gpt11) (loop,gpt10) (loop,gpt9) (loop,gpt8) (loop,gpt7) (loop,gpt6) (loop,gpt5) (loop,gpt4) (loop,gpt3) (loop,gpt2) (loop,gpt1) (hd0) (hd0,gpt12) (hd0,gpt11) (hd0,gpt10) (hd0,gpt9) (hd0,gpt8) (hd0,gpt7) (hd0,gpt6) (hd0,gpt5) (hd0,gpt4) (hd0,gpt3) (hd0,gpt2) (hd0,gpt1) (hd1) (hd1,gpt6) (hd1,gpt5) (hd1,gpt4) (hd1,gpt3) (hd1,gpt2) (hd1,gpt1)

sebanc, 2020-05-07:
try "search --no-floppy --set=root --file /chromeos.img"
then "echo $root"

Jeremy97L, 2020-05-07:
> try "search --no-floppy --set=root --file /chromeos.img"
> then "echo $root"

The first command returned nothing, the second command returned "hd,gpt6"

sebanc, 2020-05-07:
try:
"linux (loop,gpt7)/kernel"
"initrd (loop,gpt7)/initramfs.img"

Jeremy97L, 2020-05-07:
> try:
> "linux (loop,gpt7)/kernel"
> "initrd (loop,gpt7)/initramfs.img"

Both commands return "error: unknown filesystem."

sebanc, 2020-05-07:
ok that's what is not normal, what's the output of "ls (loop,gpt7)" ?

Jeremy97L, 2020-05-07:
> ok that's what is not normal, what's the output of "ls (loop,gpt7)" ?

Partition loop,gpt7: No known filesystem detected - Partition start at 8228KiB - Total size 1048576KiB

What is this partition supposed to be?

sebanc, 2020-05-07:
rootc partition, where all brunch files are stored. It's the image that was not correctly generated for some reason.

Boot to chromeos on your usb, mount your ext4 partition and run:
"sudo bash chromeos-install -dst <path_to_the_destination_partition>/chromeos.img" (add -s argument if you want to specify the image size)

sebanc, 2020-05-07:
Btw, you might also be missing an "\" between "disablevmx=off" and "cros_secure":
```
linux (loop,gpt7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off
cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path
initrd (loop,gpt7)/initramfs.img
```

Jeremy97L, 2020-05-07:
> rootc partition, where all brunch files are stored. It's the image that was not correctly generated for some reason.
>
> Boot to chromeos on your usb, mount your ext4 partition and run:
> "sudo bash [...]

I did the following (btw, I have an ntfs partition instead of ext4 this time):
mkdir -p ~/tmpmount
sudo mount /dev/sda6 ~/tmpmount
sudo bash chromeos-install -dst /dev/sda6/chromeos.img

This returns ¨Device /dev/sda6/chromeos.img does not exist¨ 
If I install to sda instead of sda6 I will probably wipe my entire drive instead of install to a partition

> Btw, you might also be missing an \ between "disablevmx=off" and "cros_secure":

I checked this and this was already present in my USB grub

sebanc, 2020-05-07:
sudo bash chromeos-install -dst ~/tmpmount/chromeos.img

Jeremy97L, 2020-05-07:
> sudo bash chromeos-install -dst ~/tmpmount/chromeos.img

Whoops, my bad, anyway here is the output:

_Creating image file
0+0 records in
0+0 records out
0 bytes copied, 5.8408e-05 s, 0.0 kB/s
WARNING: Primary GPT header is invalid
WARNING: Secondary GPT header is invalid
B03179FC-DD45-E040-9D63-8A3E89E3E1FB
       start        size    part  contents
           0           1          PMBR (Boot GUID: B03179FC-DD45-E040-9D63-8A3E89E3E1FB)
           1           1          Pri GPT header
           2          32          Pri GPT table
    19255304    10104776       1  Label: "STATE"
                                  Type: Linux data
                                  UUID: 15F4D222-6BD8-DB47-B4CA-6B339F1D33E5
     2117638       65536       2  Label: "KERN-A"
                                  Type: ChromeOS kernel
                                  UUID: 3A8CD65D-2B0B-8244-B595-A372DF4B3B54
                                  Attr: priority=15 tries=15 successful=0 
    10866696     8388608       3  Label: "ROOT-A"
                                  Type: ChromeOS rootfs
                                  UUID: F1847D20-37A7-B54F-9877-721093B39B60
     2183174       65536       4  Label: "KERN-B"
                                  Type: ChromeOS kernel
                                  UUID: 5DB35A4A-CC97-6649-83FB-EF0D694E33E1
                                  Attr: priority=0 tries=15 successful=0 
     2478088     8388608       5  Label: "ROOT-B"
                                  Type: ChromeOS rootfs
                                  UUID: A492F6AE-98D8-5F4E-BD73-2AC50FDBE316
       16448           1       6  Label: "KERN-C"
                                  Type: ChromeOS kernel
                                  UUID: 5A89B961-50ED-9047-B778-E96DB16682F8
                                  Attr: priority=0 tries=15 successful=0 
       16456     2097152       7  Label: "ROOT-C"
                                  Type: ChromeOS rootfs
                                  UUID: 5ACDECD4-76D0-A047-8AC9-F2C86172D17B
     2248712       32768       8  Label: "OEM"
                                  Type: Linux data
                                  UUID: 49F6D553-9145-284A-82E1-CF71D4F651B4
     2113608           1       9  Label: "reserved"
                                  Type: ChromeOS reserved
                                  UUID: 61975B8F-EF0A-5241-B312-029BBB26BBBE
     2113609           1      10  Label: "reserved"
                                  Type: ChromeOS reserved
                                  UUID: 47EA0062-B106-AB49-9893-6D8F4F2291FC
          64       16384      11  Label: "RWFW"
                                  Type: ChromeOS firmware
                                  UUID: 205B7B8A-59AD-3747-B28B-F5C7AFB0F8B3
     2412552       65536      12  Label: "EFI-SYSTEM"
                                  Type: EFI System Partition
                                  UUID: B03179FC-DD45-E040-9D63-8A3E89E3E1FB
    29360095          32          Sec GPT table
    29360127           1          Sec GPT header
Writing partition 1
mke2fs 1.45.4 (23-Sep-2019)
Discarding device blocks: done                            
Creating filesystem with 1263097 4k blocks and 316368 inodes
Filesystem UUID: a9184a4c-32e0-494d-9355-e2e34d14b566
Superblock backups stored on blocks: 
        32768, 98304, 163840, 229376, 294912, 819200, 884736

Allocating group tables: done                            
Writing inode tables: done                            
Creating journal (16384 blocks): done
Writing superblocks and filesystem accounting information: done 

Writing partition 2
[... 11 lines cut ...]
Writing partition 8
16.0MiB 0:00:00 [ 100MiB/s] [==========================================================================>] 100%            
Writing partition 9
 512 B 0:00:00 [17.7KiB/s] [===========================================================================>] 100%            
Writing partition 10
 512 B 0:00:00 [16.7KiB/s] [===========================================================================>] 100%            
Writing partition 11
8.00MiB 0:00:00 [78.9MiB/s] [==========================================================================>] 100%            
Writing partition 12
32.0MiB 0:00:00 [ 101MiB/s] [==========================================================================>] 100%            
ChromeOS installed.
To boot directly from this image file, add the lines between stars to either:
- A brunch usb flashdrive grub config file (then boot from usb and choose boot from disk image in the menu),
- Or your hard disk grub install if you have one (refer to you distro's online resources).
********************************************************************************
menuentry "ChromeOS (boot from disk image)" {
        img_part=/dev/sda6
        img_path=/chromeos.img
        search --no-floppy --set=root --file $img_path
        loopback loop $img_path
        linux (loop,gpt7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
                cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path
        initrd (loop,gpt7)/initramfs.img
}
********************************************************************************_

sebanc, 2020-05-07:
yes, perfect, try booting it now

Jeremy97L, 2020-05-07:
> yes, perfect, try booting it now

It actually boots this time! The only problem is that I only have around 5 GB. Should I perform a ¨sudo resize-data¨ on the HDD Chrome OS installation and reboot?

Anyways, thank you so much for the quick and useful responses! What was the problem and how can we avoid others getting the same problem?

sebanc, 2020-05-07:
Unfortunately, I don't know what was the problem as it was at the time of creation of the previous image and I don't have the log.

For the size you need to specify -s option when you run the chomeos-install command (add "-s 50" if you want to give the image 50GB, chromeos takes 10GB so 40GB data will remain)
```
"sudo bash chromeos-install -dst <path_to_the_destination_partition>/chromeos.img" -s <image_size>"
```

So you have to recreate the image.

sebanc, 2020-05-07:
@ghritik @ijinesh Have you been able to solve your issues ?

Jeremy97L, 2020-05-07:
> Unfortunately, I don't know what was the problem as it was at the time of creation of the previous image and I don't have the log.
>
> For the size you need to specify -s option when you run the [...]

Alright, I now have all the space I needed, thanks again. The only small issue I have right now is that I can only boot the HDD Chrome OS via my USB grub and not the Linux Mint grub. Both have the exact same code which is this:

`menuentry "ChromeOS (boot from disk image)" {
        img_part=/dev/sda6
        img_path=/chromeos.img
        search --no-floppy --set=root --file $img_path
        loopback loop $img_path
        linux (loop,gpt7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
                cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path
        initrd (loop,gpt7)/initramfs.img
}
`
I used grub customizer in Linux Mint, created a new custom script, type ¨other¨ and inserted it

sebanc, 2020-05-07:
@Jeremy97L I don't know grub customizer. I imagine that linux mint work as ubuntu, so create a file named "/etc/grub.d/99_chromeos" with the below content:

```
#!/bin/sh
exec tail -n +3 $0
# This file provides an easy way to add custom menu entries.  Simply type the
# menu entries you want to add after this comment.  Be careful not to change
# the 'exec tail' line above.

menuentry "Chrome OS" {
img_part=/dev/sda6
img_path=/chromeos.img
search --no-floppy --set=root --file $img_path
loopback loop $img_path
linux (loop,gpt7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path
initrd (loop,gpt7)/initramfs.img
}
```

Then run:
"sudo chmod 0755 /etc/grub.d/99_chromeos"
"sudo update-grub"

If this config still does not boot, the grubx64.efi has not been correctly replaced by the brunch one.

ijinesh, 2020-05-07:
> @ghritik @ijinesh Have you been able to solve your issues ?

Hi Sebanc,
Instead of installing it directly on HDD as dual boot, I first installed on USD drive and then installed on HDD as single OS.

I will try to make a dual boot with Windows 10, but this seems too complicated for me to rectify the problem.

ijinesh, 2020-05-08:
> @ghritik @ijinesh Have you been able to solve your issues ?

Hi Sebanc,

I tired working with Soraka (HP Chromebook X2) image and wanted to install it on USB. However, I am facing weird problem now (see below). Is Brunch compatible with Soraka?

Also, in Rammus based ChromeOS bluetooth and SD Card isn't working. Any solution for the same? 

root@jinesh-T102HA:/media/jinesh/7732646b-6121-4583-8446-f5f1bf12bf18/ChromeOS# sudo bash chromeos-install.sh -src soraka_recovery.bin -dst /dev/sda
All data on device /dev/sda will be lost, are you sure ? (type yes to continue) yes
umount: /dev/sda: not mounted.
umount: /dev/sda10: not mounted.
umount: /dev/sda11: not mounted.
umount: /dev/sda12: not mounted.
umount: /dev/sda2: not mounted.
umount: /dev/sda3: not mounted.
umount: /dev/sda4: not mounted.
umount: /dev/sda5: not mounted.
umount: /dev/sda6: not mounted.
umount: /dev/sda7: not mounted.
umount: /dev/sda8: not mounted.
umount: /dev/sda9: not mounted.
BDBC10D2-F5C5-7D4D-832E-8B3C81700E58
       start        size    part  contents
           0           1          PMBR (Boot GUID: BDBC10D2-F5C5-7D4D-832E-8B3C81700E58)
           1           1          Pri GPT header
           2          32          Pri GPT table
    19255304    12046279       1  Label: "STATE"
                                  Type: Linux data
                                  UUID: 3C174AA6-CD76-9A4E-9E84-B100CABD880A
     2117638       65536       2  Label: "KERN-A"
                                  Type: ChromeOS kernel
                                  UUID: F16A39EC-E023-CA40-8547-091B7F2183EA
                                  Attr: priority=15 tries=15 successful=0 
    10866696     8388608       3  Label: "ROOT-A"
                                  Type: ChromeOS rootfs
                                  UUID: 2C62A443-8C4C-5C4D-AE9D-88A040C8BD94
     2183174       65536       4  Label: "KERN-B"
                                  Type: ChromeOS kernel
                                  UUID: 42EB935B-BC9A-E24F-AA2F-463532A86747
                                  Attr: priority=0 tries=15 successful=0 
     2478088     8388608       5  Label: "ROOT-B"
                                  Type: ChromeOS rootfs
                                  UUID: 6CB2FC3E-F377-E948-8942-65F54C0F5833
       16448           1       6  Label: "KERN-C"
                                  Type: ChromeOS kernel
                                  UUID: D05FDE0B-0303-1147-86ED-E2A2C06F8507
                                  Attr: priority=0 tries=15 successful=0 
       16456     2097152       7  Label: "ROOT-C"
                                  Type: ChromeOS rootfs
                                  UUID: 20C50230-99EC-504E-AF53-526F86391B88
     2248712       32768       8  Label: "OEM"
                                  Type: Linux data
                                  UUID: F1DB5973-1D31-E74B-A5C0-D99F5B488004
     2113608           1       9  Label: "reserved"
                                  Type: ChromeOS reserved
                                  UUID: 0B6E98D1-AFD7-EA47-8C0A-0DC42FF08ACB
     2113609           1      10  Label: "reserved"
                                  Type: ChromeOS reserved
                                  UUID: 93283787-28BF-B644-93B9-0AA6F26769DE
          64       16384      11  Label: "RWFW"
                                  Type: ChromeOS firmware
                                  UUID: 05582FCF-F3A5-0147-BD16-FE3FA19D85C1
     2412552       65536      12  Label: "EFI-SYSTEM"
                                  Type: EFI System Partition
                                  UUID: BDBC10D2-F5C5-7D4D-832E-8B3C81700E58
    31301598          32          Sec GPT table
    31301630           1          Sec GPT header
Writing partition 1
10.0MiB 0:00:00 [ 110MiB/s] [================================>] 100%            
Writing partition 2
16.0MiB 0:00:00 [ 118MiB/s] [================================>] 100%            
Writing partition 3
1.75GiB 0:00:14 [ 123MiB/s] [========================>         ] 74% ETA 0:00:04
pv: write failed: No space left on device
Writing partition 4
pv: write failed: No space left on device
Writing partition 5
pv: write failed: No space left on device
Writing partition 6
pv: write failed: No space left on device
Writing partition 7
pv: write failed: No space left on device
Writing partition 8
pv: write failed: No space left on device
Writing partition 9
pv: write failed: No space left on device
Writing partition 10
pv: write failed: No space left on device
Writing partition 11
pv: write failed: No space left on device
Writing partition 12
pv: write failed: No space left on device
ChromeOS installed.

Jeremy97L, 2020-05-08:
> Hi Sebanc,
>
> I tired working with Soraka (HP Chromebook X2) image and wanted to install it on USB. However, I am facing weird problem now (see below). Is Brunch compatible with Soraka?
>
> Also, in [...]

It looks like your USB doesn't have enough space for the installation. How much storage does it have? And is sda really your USB drive? For me it often is sdb instead

ijinesh, 2020-05-08:
> It looks like your USB doesn't have enough space for the installation. How much storage does it have? And is sda really your USB drive? For me it often is sdb instead

My USB drive is 16GB. Yes, it is SDA as shown in Gparted and Disks in Linux.

sebanc, 2020-05-08:
@ijinesh Hi, for some reason the write operation fails on some media and you need to manually reset it. There is no clearly identified method yet on how to solve this issue, those who have encountered it seems to have been able to fix it by deleting all partitions with fdsik, gparted should probably work too.

If you manage to fix it, I would be interested to know the method which worked for you.

Jeremy97L, 2020-05-08:
If you wish to remove all the partitions from your USB and make it a clean one partition drive, here is a tutorial for windows:
_hit windows-x and hit 'command prompt (admin)'
type 'diskpart' 
type 'list disk' 
type 'select disk #', with # being your flash drive
type 'clean' 
type 'create partition primary'
type 'format fs=fat32 quick' 
type 'active'
type 'assign' 
then finally type 'exit'_

Jeremy97L, 2020-05-08:
> @Jeremy97L I don't know grub customizer. I imagine that linux mint work as ubuntu, so create a file named "/etc/grub.d/99_chromeos" with the below content:
>
> ```
> #!/bin/sh
> exec tail -n +3 $0
> # This [...]

Sadly enough, your solution did not work. It does come up in the list of bootable options, but I still have the same error:
No such device: /chromeos.img
File '/chromeos.img' not found
Disk 'loop.gpt7' not found
You need to load the kernel first

Guess I'll have to keep using my USB grub for now and deal with it. On a whole other topic, are you aware of the compatibility with flatpak on chrome os using brunch? For me, many programs fail to load and show a blank screen like Signal and Discord.

ijinesh, 2020-05-08:
> @ijinesh Hi, for some reason the write operation fails on some media and you need to manually reset it. There is no clearly identified method yet on how to solve this issue, those who have [...]

Hi Sebanc,
I did reformat it and got it to a single partition and tried again. But it is same error again. I tried doing it multiple times with various permutation combination.

Do you think it could be due to Soraka image? 
@Jeremy97L - I used Gparted to clean all partitions which were getting created in the process.

sebanc, 2020-05-08:
@ijinesh I did test brunch with soraka at some point and it was working, also it fails on partition 3 at 1.75 GB whereas partition 3 is 4GB so it cannot be that.

I never really understood this issue try recreating the partition table, then creating only one ext4 partition which takes all the drive space and then try again chromeos-install.sh.

Another issue which looks like this one happens when the recovery image has not been unzipped. You should also verify that

ijinesh, 2020-05-08:
> @ijinesh I did test brunch with soraka at some point and it was working, also it fails on partition 3 at 1.75 GB whereas partition 3 is 4GB so it cannot be that.
>
> I never really understood this [...]

Recovery image was unzipped.

sebanc, 2020-05-09:
Ok after giving it a bit of though i have another suggestion.

Instead of installing directly to the device, create a chromeos image file and then flash it to the device using "dd" or another image flashing tool.

After chromeos first boot, you will have to use "sudo resize-data" to get your data partition to the correct size.

ijinesh, 2020-05-10:
> Ok after giving it a bit of though i have another suggestion.
>
> Instead of installing directly to the device, create a chromeos image file and then flash it to the device using "dd" or another image [...]

How do you create image file?

sebanc, 2020-05-10:
same command, just replace "-dst /dev/sdX" by "-dst chromeos.img"

natri23, 2020-05-13:
> @ijinesh Hi, for some reason the write operation fails on some media and you need to manually reset it. There is no clearly identified method yet on how to solve this issue, those who have [...]

I got the same problem yesterday on Solus with two very different USB. And the solution was to boot to another distro's live USB (PopOS in my case) and run the sh there. After that, I tried again in Solus and it works fine. 
I would guess for some reason the cgpt writes to cache instead of disk and outputs error when the cache is full

omminp, 2020-06-02:
If I can add a bit to this thread, I have simillar but not identical issue with dual boot.  In my case, I am using BIOS boot, and I have the chromeos.img file on partition sda6 of HDD.  I can boot to it from the USB stick grub as expected, but from the option inserted into Deepin grub on the HDD, it always fails as described here.  The error comes when it tries to execute the line 
linux (loop,gpt7)/kernel.......
and is 'no such partition'
The grub code is identical in each case, so the grub on the USB executes it without eror, but the Deepin grub does not.  I am using BIOS boot, is there is file that needs to be updated in Deepin to make it compatible?

francarl, 2020-06-09:
Created image from Brunch r83 k4.19 testing 20200528 and rammus 81 recovery and tried to configure Grub for loopback boot from image (which has size of 14GB).
This is grub configuration 
```
menuentry "ChromeOS (boot from disk image)" {
        img_part=/dev/sda1
        img_path=/cos.img
        search --no-floppy --set=root --file $img_path
        loopback loop $img_path
        linux (loop,gpt7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
                cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path
        initrd (loop,gpt7)/initramfs.img
} 
```
but grub fails with "error: no such partition". 
Some command that could help:

```
grub> ls
(hd0) (hd0,msdos1)
grub> ls (hd0,msdos1)
Partition hd0,msdos1:Filesystem type ext* ....
grub> ls (hd0,msdos1)/cos.img
cos.img
grub> loopback loop1 /cos.img
grub> ls
(loop1) (hd0) (hd0,msdos1)
grub> ls (loop1)
Device loop1: No known filesystem detected - Sector size 512B - Total size 14680064KiB
```

More commands from debian shell
```
root@debian:~# losetup -P -f /cos.img
root@debian:~# losetup
NAME       SIZELIMIT OFFSET AUTOCLEAR RO BACK-FILE DIO
/dev/loop0         0      0         0  0 /cos.img    0
root@debian:~# mount /dev/loop0p7 /mnt/Ch07/
root@debian:~# ll /mnt/Ch07/
total 11692
-rw-r--r-- 1 root root 3183722 May 28 09:21 initramfs.img
-rw-r--r-- 1 root root 8751664 May 28 09:21 kernel
drwxr-xr-x 6 root root    4096 May 28 09:21 lib
drwx------ 2 root root   16384 May 28 09:21 lost+found
drwxr-xr-x 2 root root    4096 May 28 09:21 packages
drwxr-xr-x 3 root root    4096 May 28 09:21 patches
drwxr-xr-x 3 root root    4096 May 28 09:21 usr
root@debian:~#
```

Also tried to generate image with '-l' flag for MBR with same result

gawalivaibhav, 2020-06-15:
I got the same error as mentioned on this page.

##Problem while booting HDD grub
error: no such device: /chromos.img
error: file ./chromeos.img not found
error: disk loop,gpt7 not found
error: you need to load the kernel first...

##Solustion 
   - I have disabled “Secure Boot” in BIOS, It resolved my issue.
for more detail please go through below link, It might help 
https://beebom.com/how-install-chrome-os-on-pc/

sebanc, 2020-06-26:
This issue covers too many different cases, I am closing it as individual issues will be easier to follow.

closed 2020-06-26 by sebanc

jwalcek, 2020-07-22:
Try adding "rmmod tpm" to grub.cfg. I had the same problem triple booting with W10 and Ubuntu on my XPS 18. I added [that to grub.cfg](https://askubuntu.com/questions/1186040/grub-command-loopback-loop-does-not-work-on-ubuntu-19-10), but not under menuentry like in the link. It was under one of the earlier entries; let me check and edit this post.

PeterParker2008, 2021-03-21:
Hi, I am not getting "boot from disk image" in the menuentry. I am getting:-
menuentry "ChromeOS" {
        rmmod tpm
        img_part=/dev/sda5
        img_path=/chromeos.img
        search --no-floppy --set=root --file $img_path
        loopback loop $img_path
        linux (loop,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
                cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path \
                console= vt.global_cursor_default=0 brunch_bootsplash=default quiet
        initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
}

menuentry "ChromeOS (debug mode)" {
        rmmod tpm
        img_part=/dev/sda5
        img_path=/chromeos.img
        search --no-floppy --set=root --file $img_path
        loopback loop $img_path
        linux (loop,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
                cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path
        initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
}

Please help me to solve this issue.

simplyrajatgupta, 2021-05-31:
Check this out 
**[https://github.com/sebanc/brunch/issues/1088]()**

Orangeskai, 2022-01-23:
> Try adding "rmmod tpm" to grub.cfg. I had the same problem triple booting with W10 and Ubuntu on my XPS 18. I added [that to grub.cfg](https://bugs.launchpad.net/ubuntu/+source/grub2/+bug/1851311/comments/62):
> 
> ```
> if loadfont /boot/grub/font.pf2 ; then
>  set gfxmode=auto
>  insmod efi_gop
>  insmod efi_uga
>  insmod gfxterm
>  terminal_output gfxterm
>  rmmod tpm
> fi
> ```
> 
> Mine didn't quite match in the above code, but it worked.

This is fix my problem thx I've been trying for hours

[1 more reply]