sebanc/brunch #113: ChromeOS freezes upon showing ChromeOS logo on Lenovo Yoga 11e 4th gen

GeorgGearlQ, 2020-05-06:
Hi. I'm having trouble getting ChromeOS to work on my Lenovo Yoga 11e 4th gen Celeron laptop. I've tried several different recoveries, including eve, pyro and rammus, and the last 4 releases of the brunch framework to build an img I could boot.

I've tried booting via USB and micro SD, with secure boot enabled and disabled, I've tried disabling devices such as Webcam and then boot, with no luck.

All the images works on my Lenovo L540 laptop, but whenever I try them on the yoga, it builds the root c, then boots, and freezes within 2 secs after showing the ChromeOS logo.

I hope there is some way to get it to work. But thanks for an awesome framework!

GeorgGearlQ, 2020-05-06:
[image]

sebanc, 2020-05-07:
Hi,

Thanks for the detailed description and the printscreen.

The printscreen situation is normal as eve recovery is built with cpu instructions which are not supported by your pc. However, the fact that it does not work with the 2 other recoveries (pyro and rammus) is not normal, my first idea would be the wifi driver failing.

Could you try booting try adding "options=iwlwifi_backport" to the kernel command line (after cros_debug) ?

If it does not work, could you try adding "module_blacklist=iwlwifi" to the kernel command line ?

GeorgGearlQ, 2020-05-07:
I've tried both now, no change sadly. I hope i did it right, i pressed "e" on the boot entry, added the lines each time, and pressed f10 to boot.

sebanc, 2020-05-08:
Seems fine, my only doubt is when you say "added the line", the text should be following cros_debug separated by a space.

Could you also try "module_blacklist=hid_sensor_cros_compat" ?

GeorgGearlQ, 2020-05-08:
Yeah, i worded it poorly, i did the way you just said. I tried to add the blacklist for the sensor, but that didnt work.

I forgot to mention, that at the moment the yoga gets stuck on the chromeos logo, the LED in the keyboard for the mute button lights up. I don't know if that is relevant or not.

sebanc, 2020-05-09:
The LED info is interesting, could you try booting with "module_blacklist=yoga_usage_mode,thinkpad_acpi" ?

GeorgGearlQ, 2020-05-09:
It still freezes, although now the LED for the mute button is not lighting up anymore.

sebanc, 2020-05-10:
Could you try the latest release ? I included a specific patch for your device, hope it works this time.

If it still does not work, could you try with "octopus" recovery image ?

GeorgGearlQ, 2020-05-11:
I'm sorry for all the trouble, but it still freezes, also with the octopus recovery image.

sebanc, 2020-05-12:
No worries, for now we have tried all the things i had in mind.

A more recent bios should not cause that unless its faulty in which case other distros would probably be affected too. Could you try to see if cloudready boots ?

Also, sometimes after a bios update you need to reset bios settings to default, you can also try it.

GeorgGearlQ, 2020-05-12:
I have reset my bios. I tried CloudReady and it works. The light on the mute button lid up for a second, then turned off and it started to load.

sebanc, 2020-05-12:
Could you send me a dmesg and an lsmod from Cloudready ?

To obtain them, run in crosh shell:
"dmesg > ~/Downloads/dmesg.txt"
"lsmod > ~/Downloads/lsmod.txt"

GeorgGearlQ, 2020-05-12:
Here they are.

[dmesg.txt](https://github.com/sebanc/brunch/files/4614170/dmesg.txt)
[lsmod.txt](https://github.com/sebanc/brunch/files/4614171/lsmod.txt)

sebanc, 2020-05-12:
Thanks ! I had a look at them, unfortunately i cannot directly identify what could be the issue with chromeos but I will try to increase generic kernel support in the next release.

sebanc, 2020-05-15:
I have just added you to the brunch-testing repo, when you have the time could you try the release that is there ?

GeorgGearlQ, 2020-05-15:
Sorry, but how do find it? I can't seem to figure out where to download it.

sebanc, 2020-05-15:
It's in another repository of my github account named "brunch-testing" instead of just "brunch", it's a private repository, you may need to validate something by email before you can see it.

GeorgGearlQ, 2020-05-15:
I have tried booting it now, and it freezes like the other releases.

sebanc, 2020-05-15:
Sorry to hear that, I hope that at some point I will figure it out

sebanc, 2020-05-24:
There are some changes in the latest build which might help if you want to try:
https://github.com/sebanc/brunch-testing/releases

GeorgGearlQ, 2020-05-24:
I tried it, sadly no change. I'm getting another windows touch device later today that I'm gonna try and install ChromeOS on instead, hopefully that device is gonna play nice with the brunch framework. 

I've tried every single chromium distro on the yoga, and only cloudready works, every other distro does the same thing and then freezes, so it seems to be a general issue with my yoga.

GeorgGearlQ, 2020-06-03:
I've bought a chromebook, but ill keep trying to install it on the Yoga once in a while with new updates.

sebanc, 2020-06-03:
Nice ! Enjoy your chromebook :)