sebanc/brunch #114: Issues with ChromeOS

PruvostMaxime, 2020-05-06:
Hello,succesfully installed ChromeOS so it's nice (was using Cloudready before)

And i have an issues with the mic not working with Google assistant

I've tried with my Bose Headphones,and on Google the voice search is working (but not assistant)
So it's like internal mic is not recognized,

what can i do?

PruvostMaxime, 2020-05-06:
Also, i need to add "options=broadcom_wl" to the GRUB for my wifi-card
There is a way to save it for not having to add it each time i power on the PC?

PruvostMaxime, 2020-05-07:
And also,there is a way to have the graphics card working? (GTX 1060 Mobile) because i can't use HDMI :/

PruvostMaxime, 2020-05-07:
And i can also add 2 details 
 - touchpad is very sensitive with "tap to click" , sometime it's click,but i just touched to move the cursor

- instant Tethering is not working,the "mobile data" is empty

sebanc, 2020-05-08:
Hi,

For the sensitive touchpad, you can try changing the below value in /etc/gesture/40-touchpad-cmt.conf:
Option          "Tap Minimum Pressure" "X"

Regarding mobile data and the cast icon i think i had an issue with those in one build that was corrected since. In the latest build, at least cast works fine for me but i canot say for mobile data as i don't use it.

I am working on the mic issue but without much success for now.

PruvostMaxime, 2020-05-08:
Hi,thanks for answer ^^

I'll check for the touchpad.

For the build,i'm using the 81.0.4044.127 (rammus)

And for the mic, i followed your advise of making a bootscript so it's now working,i can use Google Assistant and other things . (Issue  #56 )

I'll edit this post after testing touchpad.

ericma15, 2020-05-12:
Hi ,I have the same problem. After update brunch with r81-0510 with chromeos 81. I got mobile data ico and cast icon disapear. Do you solve it? In previous version, it can work. That's r80 with chrome80

sebanc, 2020-06-26:
As far as I know, there is no issue related to mobile data and casting in recent releases. Feel free to re-open this issue if it is still relevant.

closed 2020-06-26 by sebanc