sebanc/brunch #115: External monitor issue

zeroxul, 2020-05-06:
When I connect an external monitor (HDMI),  the videos on Youtube and Twitch don't start playing.
if i Switch the audio output back on the internal speakers, the videos play.
But when a new video starts, it hangs again, and you always have to switch the audio output back and forth.
Don't know if chrome os is at fault, But I don't have it with windows or Linux.

have it on Brunch r80 and Brunch r81
Rammus Recovery
Device: Asus t102h

Tried everything, but only switching the audio device work.
I found the solution online, but it is not a permanent solution.
https://appuals.com/if-playback-doesnt-begin-shortly-try-restarting-your-device/

I assumed it was the HDCP, but I don't know?
https://www.xda-developers.com/chromebooks-intel-play-hd-vudu-netflix-hdmi-hdcp/

sebanc, 2020-05-08:
Hi,

Could you try the new release ? It has several audio fixes which might be related to your issue.

zeroxul, 2020-05-08:
Hey
thanks for your great work
I tested the new update but without success
Got it with and without options=baytrail_chromebook, No changes in either.

I still have a (Trekstor surftab twin 10.1) on which I have the same problem.
Tried Neverware-Cloudready there now and got the same error there too! It's probably not a good sign, but I'll keep testing.
I've already tried different HDMI cables and tried everything, to rule out other sources of error.

sebanc, 2020-05-09:
Thanks for trying it,

I am not 100% sure but i would say this issue is sound related. Could you try adding "module_blacklist=snd_hdmi_lpe_audio" to the kernel command line ? (not sure if audio would work through hdmi after that though but the result will be interesting)

zeroxul, 2020-05-09:
Hey I added it
after that, the sound is only reproduced via the internal speakers.
When I connect the monitor, I don’t get a choice of sound output at the bottom right
I also think it's a sound problem, as if the hardware gets stuck in a process or something.

zeroxul, 2020-05-09:
Maybe worth mentioning that
since the update 20200508, sometimes I see the headphone output, if I don’t have headphones connected.
I don't think that was before the update