sebanc/brunch #116: Update r81 branch kernel source

Zarloc, 2020-05-07:
Hello, I need to build a kernel module for a wifi stick.

I don't want to bother you with search for gathering information, so i test by myself and when it's OK i can report my changes for support it, and you can decide if you want to include it.

But the branch r81 is outdated, the kernel in is 4.19.89 but the latest brunch release `Brunch r81 k4.19 testing 20200503` is in 4.19.113 so i cannot build kernel modules.

sebanc, 2020-05-07:
Hi,

I intend to provide as much support as possible for internal wifi boards but I don't intend to support external adapters because they are too numerous and often poorly supported. What's your onboard wifi ?

I agree that the kernel version on the git is not the latest one, next release will have proper kernel headers to build dkms modules included.

Zarloc, 2020-05-07:
My onboard wifi card is a rtl8723bs which is supported by default in brunch, but it's a crap in SDIO with lot connection trouble and speed problem.

So i use a rtl8822bu (ASUS AC-53 NANO) external stick.

I understand perfectly that you don't want to support external cards, and I don't mind having to compile the module manually at every release.
And if you add support for dkms, it's even simpler.

sebanc, 2020-05-08:
Hi,

The new testing release contains the kernel headers, you should be able to build kernel modules.

However, the dkms install process will not work as it is different from linux:
- Copy firmwares if any to /lib/firmware/...
- Manually copy your module to /lib/modules/ directory
- use "insmod /lib/modules/<module_name>.ko" to load it and see if it works.

When you have it working, i will give you directions to have it load at boot time.

Zarloc, 2020-05-08:
Hi,

Module working fine on new release.
Thank you for kernel headers.

I request your help for load it the right way at boot time.

sebanc, 2020-05-08:
To load the module at boot time, you can create a bootscript (undocumented brunch feature) by running:
```
sudo mkdir -p /var/brunch/bootscripts
echo "insmod /lib/modules/<module_name>.ko" | sudo tee /var/brunch/bootscripts/wifi.sh
sudo chmod 0755 /var/brunch/bootscripts/wifi.sh
```

Zarloc, 2020-05-08:
Work like a charm, thanks.
I close.

closed 2020-05-08 by Zarloc

lfom, 2020-05-08:
@sebanc Latest release (20200508) is much smaller than previous one, is it OK? Thanks

sebanc, 2020-05-08:
@lfom yes, it is normal I removed some useless firmware files

siankatabg, 2020-05-09:
Hello @sebanc 

First, thanks for this project, I recently found about it and it's great 👍 

But as always there's a "but"... 
Anyway, I have a similar issue as the op. I recently updated to r81 (both framework and chromeos) and as the op I found out the kernel source is not uploaded so I can't add few patches to it to make my WiFi/bt working fine (rtl8723bu). Unfortunately the tablet I'm running all this has the well known problematic realtek WiFi/bt combo chip which has just 1 antenna and is using the "rtl8xxxu" module witch needs patches to enable antenna coexistence and so on.

I'm new to all this (chromeos and modules) so no idea how could I do this with dkms (or in some other way), so any help (and of course the kernel source code) would be much appreciated 👍

If you want I'll link you the needed patches so you could check them out and decide if you want to add them in the future as it seems I'm not alone with this buggy chip...

Thanks again for all your efforts and for this project 👍