sebanc/brunch #118: Bootloop

jero237, 2020-05-09:
Hi. I've installed ChromeOS into a 32gb partition on my ssd. But when I try to boot it shows me chrome logo and this appears for 1/3 of a second: "Your system is repairing itself, please wait", and the it reboots forever. What am I doing wrong?

sebanc, 2020-05-09:
Hi,

Could you detail the method you used to install chromeos on your ssd ?

jero237, 2020-05-09:
Booting from an usb created on windows. Using the system from the USB works just fine.

jero237, 2020-05-09:
I've disabled secure boot and it worked. So, I've a question, how can I install grub into the partition I've installed chrome os so I don't need the USB anymore

sebanc, 2020-05-09:
This part is quite complicated which is why there is no official method, however to guide you the steps would be:
1) Identify your windows EFI partition
2) Copy /efi/boot folder from partition 12 of your usb to /efi/brunch on your windows efi partition
3) Add a boot entry in your bios pointing to \efi\brunch\bootx64.efi

jero237, 2020-05-09:
How can I do step 2? I'm kind a noob. I'm struggling to cd /dev/sdb12

sebanc, 2020-05-09:
As you don't seem comfortable with mounting partitions, I would strongly advise not to go further with this as it's both complicated and dangerous for your data.

jero237, 2020-05-09:
Ok, thank you for the advice and for the help.

jero237, 2020-05-09:
Finally, I've managed to do your little guide and It works perfect! The last thing I need to do is to silent grub (I select windows or chrome os from the bios menu with f12, I don't need grub dual boot). How can I silent grub? I saw a few tutoriales but I can't find the grub settings file directory. Thank you in advice.

sebanc, 2020-05-10:
Run "sudo edit-grub-config" and add "console=" after cros_debug.

sebanc, 2020-05-10:
Run "sudo edit-grub-config" and change timeout value from 2 to 0.

closed 2020-05-18 by sebanc