sebanc/brunch #119: Support for Killer AX1650

shad0wfile, 2020-05-09:
Hello,

I've tried to setup chromeos on my 2019 Dell XPS 7590, it's almost there, but wireless isn't working. I know the killer wireless drivers were merged into the 5.x kernel, wondering if they could be ported in? Other then that everything else works, sound, video, mouse, keyboard, OLED doesn't have a brightness scale ability via the keyboard, but that's pretty standard on this machine with base Linux OS.

So, just looking to see if this particular driver could be added? 

Thanks!

sebanc, 2020-05-09:
Hi,

Could you try to add "options=iwlwifi_backport" to the kernel command line (after cros_debug) ?

shad0wfile, 2020-05-09:
Awesome, problem solved. 

Thanks!

shad0wfile, 2020-05-09:
Any thoughts on OLED brightness control? Usually I utilize xrandr or something on Ubuntu/etc, curious if there is anything comparable?

sebanc, 2020-05-09:
I was not aware of this issue and looked a bit into it, unfortunately all programs have x11 dependencies which cannot be resolved in chromeos.

However, some users points that brightness can be adjusted when adding "i915.enable_dpcd_backlight=1" to the kernel command line, would that be a solution for you ?

shad0wfile, 2020-05-09:
No luck, I'll continue to dig around and see if I can find anything. Thanks again for all the help!

sebanc, 2020-05-09:
You can try the croissant telegram group, someone might have encountered this issue and found a way to solve it

shad0wfile, 2020-05-09:
I'll give it a shot, thanks again.

closed 2020-05-09 by shad0wfile

sebanc, 2020-12-31:
If you are still interested, I tried to enable OLED brightness in the brunch-unstable repo release.