sebanc/brunch #121: 4.19.113 kernel source code missing

siankatabg, 2020-05-10:
Hello @sebanc 

First, thanks for this project, I recently found about it and it's great 👍 

But as always there's a "but"... 
Anyway, I recently updated to latest test r81 (both framework and chromeos) and I found out the kernel source code is not uploaded so I can't add few patches to it to make my WiFi/bt working fine (rtl8723bu). Unfortunately the tablet I'm running all this has the well known problematic realtek WiFi/bt combo chip which has just 1 antenna and is using the "rtl8xxxu" module witch needs patches to enable antenna coexistence and so on.

I'm new to all this (chromeos and modules) so no idea how could I do this with dkms (or in some other way), so any help (and of course the kernel source code) would be much appreciated 👍

If you want I'll link you the needed patches so you could check them out and decide if you want to add them in the future as it seems I'm not alone with this buggy chip...

Thanks again for all your efforts and for this project 👍

sebanc, 2020-05-10:
Hi,

I am note sure I understand what is your request, do you want me to apply patches to the included rtl8723bu module ? If so, I don't mind if you point the correct ones and if they are generic (not specific for a certain computer model)

Aside from that the kernel source (not totally up-to-date but recent enough) is on r81 branch in kernel directory and kernel headers to build modules from chromeos are now included since latest version.

siankatabg, 2020-05-10:
My request is to update the kernel source code to the latest 4.19.113 as the one in the r81 branch is still on 4.19.89 and so I can't apply the patches myself and compile (can't load the module after that, even tried to change the vermagic, but that's another topic).

I did tried to compile a different module directly in chromeos, but for some strange reason "uname -r" is still showing the old kernel and I'm new enough to dkms to fail doing this (didn't had much time to read how it works yesterday).

Here are the patches I need to apply to make the rtl8723bu chip to work correctly:
https://patchwork.kernel.org/patch/11031397/
https://patchwork.kernel.org/patch/11170977/
https://patchwork.kernel.org/patch/11175575/

I did tried only these 3, but there are more that I'd like to test out, so I'll try to find a way to compile the module somehow, or I'll just wait for the kernel code to be available.

sebanc, 2020-05-10:
The brunch build system is very complex and I make a lot of modifications so it's very difficult currently to maintain the kernel totally up-to-date, nevertheless what you can do is once you have built the kernel with your patches, replace the whole kernel and modules on partition 7 (instead of just copying the module).

Otherwise, I don't mind implementing those patches if they apply to the chromeos kernel source without too much trouble.

I will try it and keep you updated.

siankatabg, 2020-05-10:
I've found a way to compile the module on my PC using the extracted kernel headers from your rootc image, so I've fixed my issue, but I hope you'll add the patches to your kernel for the next release :+1: 

P.s. - Add this patch to the list too : https://patchwork.kernel.org/patch/11192201/

sebanc, 2020-05-10:
Could you try wifi on the stable I have just posted ? it includes your 4 patches

siankatabg, 2020-05-10:
Sorry for the delay,

I just updated to the r81 stable release and I can confirm that my WiFi and Bluetooth are working fine now. Thanks a lot for the fast response and for the great work :+1: 

I'll now close this issue, have a great day and thanks again!

closed 2020-05-10 by siankatabg