sebanc/brunch #122: Display and/or touchscreen inverted

Freaklimit, 2020-05-10:
Hi,

I'm using a TrekStor SurfTab duo W3, Intel Atom x5 Z8300, 2GB, rammus with the latest Brunch r81 k4.19 testing 20200508 and when I installed ChromeOS the display was upside down but the touch was right. So I went to settings>device>display>orientation to 180 degree to flip the display. The issue is that the display and mouse works fine with the invert but the touch is reverse in X and Y axes. Is there any command or any fix to solve this?

Thanks for all your efforts and for this project.

sebanc, 2020-05-10:
Could you run "dmesg > ~/Downloads/dmesg.txt" in crosh shell and attach the file dmesg.txt in your downloads folder to this issue ?

Freaklimit, 2020-05-11:
Hi,

thanks for your quick answer. Here ist the dmesg.txt file:
[dmesg.txt](https://github.com/sebanc/brunch/files/4611272/dmesg.txt)

sebanc, 2020-05-13:
Hi,

I created a new "brunch-testing" repository and uploaded a test release there, could you install it and try to boot with "fbcon=rotate:2" added to the kernel command line ? (this can only work with the new test release)

Freaklimit, 2020-05-13:
Hi,

I've installed an added the command "fbcon=rotate:2" into the kernel command line, but nothing changed. The issue is still there.

sebanc, 2020-05-14:
Could you post the output of "udevadm info /sys/devices/pci0000:00/808622C1:05/i2c-5/i2c-FTSC1000:00/0018:2808:508F.0003/input/input11" ?

Freaklimit, 2020-05-14:
Hi,

I've put the output in a textfile.
[udevadm info.txt](https://github.com/sebanc/brunch/files/4629818/udevadm.info.txt)

sebanc, 2020-05-16:
Could you try to run "echo 2 | sudo tee /sys/class/graphics/fbcon/rotate" in crosh shell once you are logged in ? Does it change screen orientation ?

Freaklimit, 2020-05-16:
I've tried it, but once again nothing changes. I get only a number "2" as result.
crosh> shell
chronos@localhost / $ echo 2 | sudo tee /sys/class/graphics/fbcon/rotate
2

sebanc, 2020-05-22:
Coud you try the new build there ?
https://github.com/sebanc/brunch-testing/releases

Freaklimit, 2020-05-22:
Nope, once again nothing changes. I'm also starting to despair :(

sebanc, 2020-05-23:
Ok, I have 1 last idea but I would need to be sure of your "product_name" and "sys_vendor". Could you launch a linux live distro and post the output of the below commands ?
```
cat /sys/class/dmi/id/product_name
cat /sys/class/dmi/id/sys_vendor
```

Freaklimit, 2020-05-23:
The outputs of these commands
`cat /sys/class/dmi/id/product_name
SurfTab duo W3
cat /sys/class/dmi/id/sys_vendor
TrekStor`

sebanc, 2020-05-23:
I have just uploaded a new testing release with another attempt to fix it:
https://github.com/sebanc/brunch-testing/releases

If it does not work, could you please attach a dmesg ?

The problem of the transformation matrix is that the one in crostini will not be applied to chromeos and the one you use in linux mint depends on libinput which is not present in chromeos. I have not found a way to add a transformation matrix in chromeos yet.

Freaklimit, 2020-05-23:
Sorry again, but it doesn`t worked. Here is the dmesg.txt file.
[dmesg.txt](https://github.com/sebanc/brunch/files/4672121/dmesg.txt)

sebanc, 2020-05-23:
Indeed the function I modified is apparently not used by your device... I will try something else

sebanc, 2020-06-26:
If you are still interested, could you update to the latest testing build and post a new dmesg ? I added debug statements in the code which might help.

Freaklimit, 2020-07-02:
Sorry for my late answer, here is the new dmesg which you required.
[dmesg.txt](https://github.com/sebanc/brunch/files/4863657/dmesg.txt)

sebanc, 2020-07-02:
Could you run "sudo nano /etc/gesture/80-inverted_touchscreen.conf" in crosh shell, copy the below content in nano, save and reboot ?

```
Section "InputClass"
    Identifier      "Evdev for Inverted TouchScreen"
    MatchProduct    "FTSC1000:00 2808:508F"
    MatchDevicePath "/dev/input/event*"
    Driver          "evdev"
    Option          "InvertY"         "1"
    Option          "InvertX"         "1"
EndSection
```

Freaklimit, 2020-07-02:
I added it in the config, rebooted multiple times, changed the values one of them and than both to "0", but still not working.

sebanc, 2020-07-02:
Could you try removing the line `MatchProduct    "FTSC1000:00 2808:508F` just to check if does something ? (your mouse might be inverted further to that change)

Freaklimit, 2020-07-02:
I´ve done the whole process with the change. The mouse isn´t inverted, it is still normal like before.