sebanc/brunch #123: Surface Pro 7

LFmove, 2020-05-10:
Trying ti install on dualboot (from Windows) everything ok until usbdrive boot.

When chromeos install on the USB It ever stops at [636.710127] input: HDA Intel PCH HDMI/DP,pcm=3 as /devices/pci0000:00:1f.3/sound/card0/input13

I tryed more times, from USB.

One time i tryed ti install on a microSD, but doesn't work.

LFmove, 2020-05-10:
[image]

sebanc, 2020-05-11:
Hi,

Your issue appears to be the exact same as #18. For now, I have just posted instructions on how to use kitakar's kernel 5.4 with brunch and will also try to fix the issue with brunch kernel 4.19.

I am closing this issue but don't hesitate to participate to the discussions in issue #18.

closed 2020-05-11 by sebanc