sebanc/brunch #124: Pipo X7 with UEFI 32 bit only but supporting 64 bit systems - could Chrome OS be possible to boot?

jordanvanbergen, 2020-05-11:
There are OLD computers sold that have modern 64bit (multicore) cpu's, but are fitted with a 32bit UEFI. This means often that one can't boot 64bit operating systems, even when the cpu supports this. 

The problem is that lots of linux distributions come with only an 64bit efi boot option. The 32bit efi files are not included This means that if you have a 32bit efi, and no option to boot in legacy (CSM/bios) mode, you can't boot and therefore arent able to install it.

The problem here is that the pipo x7 has a 32bit UEFI and because the pipo x7 is expecting to find a 32bit uefi file to load but my mint linux USB stick only has the 64bit files. So it ignores the stick. 

Read this about trying to run Ubuntu on Pipo X7/X8: https://wiki.debian.org/InstallingDebianOn/PIPO/PIPO%20X8 

Do you all think I should not bother to try to get Chrome OS running in combination with Brunch on my Pipo X7?

sebanc, 2020-05-11:
Hi,

The 32 bit efi is included by default in brunch. If you have issues installing linux, try to create the brunch usb using the windows method.

As far as i know, it should work.

jordanvanbergen, 2020-05-11:
> Hi,
> 
> The 32 bit efi is included by default in brunch. If you have issues installing linux, try to create the brunch usb using the windows method.
> 
> As far as i know, it should work.

Is this what you suggest: [try to create the brunch usb using the windows method.] in the readme section of this repository?

jordanvanbergen, 2020-05-11:
At the readme from https://github.com/sebanc/brunch I see:
Install ChromeOS from Windows

I read this but came to this point:
7. Make sure you have at least 14gb of free space available

The PIPO X7 only has 32GB and only 24GB for Windows to use and I saw I have 12 GB available only. 

So I think I won't go into all this hussle to get this old Pipo X7 up-and-running although it would have been nice to see Chrome OS running on it. But I think this system will anoy me too much because of the time that's needed to get it probably running.

sebanc, 2020-05-11:
Else, you can use any linux live flashdrive that boots on your pc to create the brunch usb.

Indeed with such a low amount of hdd space, your only choice would be to run chromeos on usb or to install it fully on the hdd.

jordanvanbergen, 2020-05-11:
> Else, you can use any linux live flashdrive that boots on your pc to create the brunch usb.
> 
> Indeed with such a low amount of hdd space, your only choice would be to run chromeos on usb or to install it fully on the hdd.

Maybe I misread the thing about 14 GB minimal. Do you think Ubuntu WSL inside windows will fit in the 12 GB available HD space and the 14 GB minimal is for the USB stick? 

If so I could look what windows 10 version is on the Pipo X7 and if it has the Linux WSL support. Otherwise I should update windows 10 first and that's almost not possible with 12GB free HD space as well. 

And now I read: WSL is available only in 64-bit versions of Windows 10 from version 1607. In Pipo X7 you can only run the 32 bit version of Windows to my knowledge. So the running windows 10 is 32 bit. 

So I think I will leave it as it is and don't bother trying to run Chrome OS on this old machine with 32 UEFI boot only but with 64 bit cpu.

jordanvanbergen, 2020-05-11:
Thanks for your answers. Will save me a lot of work and I think headaches.

sebanc, 2020-05-11:
Windows is indeed probably not the way to go.

I saw somewhere that Fedora 64-bit supports 32-bit UEFI, you could try creating the usb from there.

jordanvanbergen, 2020-05-11:
> Fedora 64-bit supports 32-bit UEFI

I have a working 64 bit windows 10 version running of course with Ubuntu WSL etc. Will I be able to create the Brunch USB from this 64 bit windows that is not the machine itself where it should be installed on? 

You explained: The 32 bit efi is included by default in brunch.  

So if I build the brunch USB from an intel i9 windows 64 bit machine it could/should work as well?

sebanc, 2020-05-11:
Sure it works as well

jordanvanbergen, 2020-05-11:
> Sure it works as well

hahahahhahaha This way I might be trying it after all ;-) I will let you all know if I was able to boot the USB on this machine and what happened after booting ;-)

jordanvanbergen, 2020-05-11:
I have results:

https://imgur.com/6mCXTB9
https://imgur.com/EUMwBRy
https://imgur.com/7K8Bmb0
https://imgur.com/mKP10Og
https://imgur.com/6H9fJ24
https://imgur.com/quOqSNR
https://imgur.com/yonVDdw

So I was able to create the bootable device and after sudo resize-data
I booted a second time. Logged into my Google account, used Chrome browser to browse the internet but then it became unresponsive and after a minute or so no more signal to the HDMI. Maybe because I hit ALT+CTRL+F2 or so. 

No time to look into this now but I have currently a working LIVE chrome OS bootable USB. 

Now I want to install Chrome OS on the machine. What do I need to do as I did not see the option on the screens to install it to the FULL harddisc in the machine? 

So what's the step to really install it to the FULL HD now?

sebanc, 2020-05-11:
The instructions are here:
https://github.com/sebanc/brunch#install-chromeos-on-hdd-from-chromeos

jordanvanbergen, 2020-05-11:
> Nice that it works !
> 
> The instructions for full install are here:
> https://github.com/sebanc/brunch#install-chromeos-on-hdd-from-chromeos

I have multiple SDx and I would like to delete a bunch of them. Do you say this is possible or do you suggest to only use the 24GB Windows partition ?

sebanc, 2020-05-11:
What do you mean ? Like /dev/sda1 /dev/sda2 ... ?

If so the install destination has to be /dev/sda and brunch will erase and recreate all partitions on the disk.

jordanvanbergen, 2020-05-11:
> What do you mean ? Like /dev/sda1 /dev/sda2 ... ?
> 
> If so the install destination has to be /dev/sda and brunch will erase and recreate all partitions on the disk.

it was something like /dev/mmcboot1 or similar. 

Watch this video:

https://youtu.be/3TlZgxykwdk

while writing partition 7 it glitches and falls back to dev login and after 2 seconds screen gone and pipo x7 closes. 

If I boot from USB I can login go to chrome browser internet and after sometime the environment gets unresponsive as well and black screen and no more connection with HDMI. Signal lost. 

So I think this happens before the whole partition scheme is ready. So no more working pipo x7 any longer as partititions are not written 100%. I don't mind. Nice to try to get it working on pipo x7 cheap 99$ small pc and it looked like it would run and it does but after certain time running it comes unresponsive, black screen and connection to HDMI lost no signal and the box will shut it self down.

sebanc, 2020-05-11:
The fact that it switches to tty2 is really weird, first time I see that.

Could you try running "sudo edit-grub-config" and add "intel_idle.max_cstate=1" after "cros_debug" ? This should make the system more stable.

If it does not help, switch to tty2 before login (CTRL+ALT+F2) and try installing directly from there.

jordanvanbergen, 2020-05-11:
> Could you try running "sudo edit-grub-config" and add "intel_idle.max_cstate=1" after "cros_debug" ? This should make the system more stable.

Do you want me to do this on the live USB after boot? I think I have just enough time to arrange this. And after that reboot and try the below again: 

> If it does not help, switch to tty2 before login (CTRL+ALT+F2) and try installing directly from there.

That's what I already did so I did not login and use (CTRL+ALT+F2) directly.

jordanvanbergen, 2020-05-11:
> Do you want me to do this on the live USB after boot? I think I have just enough time to arrange this. And after that reboot and try the below again:
>
>
> That's what I already did so I did not login [...]

Will try tomorrow.

jordanvanbergen, 2020-05-12:
> The fact that it switches to tty2 is really weird, first time I see that.
>
> Could you try running "sudo edit-grub-config" and add "intel_idle.max_cstate=1" after "cros_debug" ? This should make the [...]

This has helped a lot! It has been installed on HD now and working most of the time. 

Videos
------------------------------------------
https://youtu.be/dzKnRqsr0_s
https://youtu.be/0SRSFJ2GVzk
https://youtu.be/ut5cvCQFcaM
https://youtu.be/i2RqUqPS0M0

Pictures
---------------------------------------
https://imgur.com/ycalpC3
https://imgur.com/2XFh5jN
https://imgur.com/L3dEW70
https://imgur.com/tbwW9C7
https://imgur.com/V19FdIc

So still sometimes an occational hang while watching video and not all video is working and sometime black screen when wanting to watch live television or so. 

So the video capabilities aren't 100% but for the rest everything is installing. 

I have left it now to I see automated install off all software.

sebanc, 2020-05-12:
Happy to see that it helped :)

Indeed brunch still has some issues with external displays that i am trying to fix but without much success for now. Also, chromeos itself has a few issues with external display...

jordanvanbergen, 2020-05-12:
> Happy to see that it helped :)
> 
> Indeed brunch still has some issues with external displays that i am trying to fix but without much success for now. Also, chromeos itself has a few issues with external display...

Just to let you know that it installed all play store apps after logging in. That took a while, it did not crash and as everything is installed now it's working "faster". This intel CPU is not the fastest ;-) so it's working and realiable if you don't touch it. While using it it works as well but if you do things with realtime streaming video things become different on an external device (cheap TV with HDMI). 

Sometimes you feel it hangs due to the video things but if I press the power button once on the PIPO X7 you can get it out of the "hanging" mode. When you stop using the video streaming things it's working okay again. 

So with this  intel_idle.max_cstate=1 it became indeed more reliable. 

I used cabled internet so far and I haven't tried wifi but I expect this not to be working. 

But still Chrome OS is working on this PIPO X7 Device while Debian users have bigger issues: https://wiki.debian.org/InstallingDebianOn/PIPO/PIPO%20X8 (no sound for rexample).

sebanc, 2020-05-12:
As far as I could see on debian notes, wifi should work.

You can try "intel_idle.max_cstate=0" instead of "intel_idle.max_cstate=1" to see if it further improves things.

tornadox, 2020-05-23:
I also have a z3735 tv-box. It works under cloudready and debian. Ubuntu however 19 failed several times on this tv-box due graphics issues. Sound also works, but it's so low power so I used it headleass with debian 10 + docker. Might also try running chrome os when the bios flasher arrives (broke bios in settings). @jordanvanbergen which recovery did you use?

jordanvanbergen, 2020-05-23:
> I also have a z3735 tv-box. It works under cloudready and debian. Ubuntu however 19 failed several times on this tv-box due graphics issues. Sound also works, but it's so low power so I used it [...]

I did not use a recovery. I could still go into the bios using HDMI to TV and hitting the DEL or EWS key. I could change things and after USB live Chrome OS was working okay without hangs I installed brunch on the whole HD.

tornadox, 2020-05-23:
I mean which chromes os recovery files you have used to make the installation? rammus?

jordanvanbergen, 2020-05-23:
> I mean which chromes os recovery files you have used to make the installation? rammus?

Yes, Rammus I always have used until now. No need to use a different one until now.

sebanc, 2020-06-02:
I don't think I can help further on this issue so I will close it, don't hesitate to re-open it if needed

closed 2020-06-02 by sebanc

Rajangupta09, 2021-05-20:
I tried the same approach but , USB does not show up on boot manager