sebanc/brunch #125: Grub image boot doesn't work: error: can't find command 'search'

steinano, 2020-05-11:
Finally manage to install to an iso and added the supplied grub entry to the grub menu. Which gives these errors:
error: can't find command search
error: can't find command loopback
error: disk 'loop,gpt7' not found
error: you need to load the kernel first

this is the grub entry that was shown to me after brunch installation

menuentry "ChromeOS (boot from disk image)" {
        img_part=/dev/nvme0n1p4
        img_path=/chromeos.img
        search --no-floppy --set=root --file $img_path
        loopback loop $img_path
        linux (loop,gpt7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
                cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path
        initrd (loop,gpt7)/initramfs.img
}

sebanc, 2020-05-11:
Hi,

It seems the grub efi that's currently installed by your linux distro lacks some features.

What you can do is replace the grubx64.efi from your distro (usually located at /boot/efi/EFI/<your distro>/grubx64.efi) by the brunch one that you can find in /efi/boot/grubx64.efi on partition 12 of your chromeos usb.

steinano, 2020-05-12:
Thanks for your prompt reply. That solved it!
I didn't have that chromeos usb anymore, but I simply copied from the brunch files.

closed 2020-05-12 by steinano