sebanc/brunch #126: PIN logon not available at sign-in screen

Skawtnyc, 2020-05-11:
The only time PIN logon works is when waking from sleep and unlocking the lock screen. It asks for the use password for logon and upon bootup. This is not a huge problem when using the device on my desktop, but for some people with fairly complex passwords it could be an issue. I'm using the latest build of Brunch, with the current stable build of Rammus (81.0.4044.141) on my Surface Go.

sebanc, 2020-05-12:
Hi,

PIN sign-in is apparently only available on a few chromebooks (Pixelbook, Pixel Slate, and possibly the HP X2). Source: https://support.google.com/chromebook/thread/15830355?hl=en

And as far as i can see it's a built in option which cannot be activated through flags or file editing.

If you really want this feature what I would suggest is that you use the pixelslate "nocturne" recovery image (eve recovery image is not compatible with Surface Go).

Skawtnyc, 2020-05-12:
Ah, okay I wasn't aware of that. I'll keep it in mind if it really bothers me, otherwise I will stick with the Rammus build for now.

closed 2020-05-12 by Skawtnyc