sebanc/brunch #127: Partitions are not being created

Ubsefor, 2020-05-12:
Partitions on USB drive for exeption of 7 are not being created

I’m trying to write them on a virtual machine under Parallels Desktop (Ubuntu 20.04), no recovery works. Using brunch 81 stable. Needless to say that the resulting USB stick is not bootable. Maybe I’m missing some important dependencies for creating the bootable USB?
the script output:

`partx: /dev/loop21: failed to read partition table`
`Writing partition 1`
`pv: /dev/loop21p1: No such file or directory`
`Writing partition 2`
`pv: /dev/loop21p4: No such file or directory`
`Writing partition 3`
`pv: /dev/loop21p3: No such file or directory`
`Writing partition 4`
`pv: /dev/loop21p4: No such file or directory`
`Writing partition 5`
`pv: /dev/loop21p3: No such file or directory`
`Writing partition 6`
`pv: /dev/loop21p6: No such file or directory`
`Writing partition 7`
`1.00GiB 0:01:02 [16.3MiB/s] [================================>] 100%            `
`Writing partition 8`
`pv: /dev/loop21p8: No such file or directory`
`Writing partition 9`
`pv: /dev/loop21p9: No such file or directory`
`Writing partition 10`
`pv: /dev/loop21p10: No such file or directory`
`Writing partition 11`
`pv: /dev/loop21p11: No such file or directory`

hussainalhaddad, 2020-05-12:
I'm getting "No space left on device", started with the build 20200510

Writing partition 1
10.0MiB 0:00:00 [1.30GiB/s] [================================>] 100%            
Writing partition 2
32.0MiB 0:00:00 [1.19GiB/s] [================================>] 100%            
Writing partition 3
2.34GiB 0:00:02 [ 916MiB/s] [================================>] 100%            
Writing partition 4
32.0MiB 0:00:00 [ 377MiB/s] [================================>] 100%            
Writing partition 5
1.14GiB 0:00:02 [ 399MiB/s] [===============>                  ] 48% ETA 0:00:02
pv: write failed: No space left on device
Writing partition 6
pv: write failed: No space left on device
Writing partition 7
pv: write failed: No space left on device
Writing partition 8
pv: write failed: No space left on device
Writing partition 9
pv: write failed: No space left on device
Writing partition 10
pv: write failed: No space left on device
Writing partition 11
pv: write failed: No space left on device
Writing partition 12
pv: write failed: No space left on device
ChromeOS installed.

Ubsefor, 2020-05-12:
> I'm getting "No space left on device", started with the build 20200510
>
> Writing partition 1
> 10.0MiB 0:00:00 [1.30GiB/s] [================================>] 100%
> Writing partition 2
> 32.0MiB 0:00:00 [...]

You, quite obviously, need a larger flash drive (larger than yours 4 Gb one) to write this image. It’s about 15 Gb btw. So use at least 16 Gb USB Flash Drive

hussainalhaddad, 2020-05-12:
> You, quite obviously, need a larger flash drive (larger than yours 4 Gb one) to write this image. It’s about 15 Gb btw. So use at least 16 Gb USB Flash Drive

Who said i have 4GB flash?! it's 16GB which i used several times on earlier builds.

Ubsefor, 2020-05-12:
> Who said i have 4GB flash?! it's 16GB which i used several times on earlier builds.

I assumed that you have a 4 Gb drive due to overall written data you given in your log. I can only imagine what’s wrong, but your drive must be the case. Try to fully erase it (or delete a partition table, but I suggest writing some bit mask over the whole drive) and see if it fixes the error

hussainalhaddad, 2020-05-12:
> I assumed that you have a 4 Gb drive due to overall written data you given in your log. I can only imagine what’s wrong, but your drive must be the case. Try to fully erase it (or delete a partition table, but I suggest writing some bit mask over the whole drive) and see if it fixes the error

I tried deleting all partitions several times, i'm able to write different images on it, but not this one?

Ubsefor, 2020-05-12:
> I tried deleting all partitions several times, i'm able to write different images on it, but not this one?

You can’t erase the whole drive by just deleting partitions, you must destroy partition table.
But honestly, I think the script does it automatically using ‘partx’ . I think the partition table gets somehow corrupted while being created, or there are bad sectors on your USB drive (that can be quite possible, if you have used it already as installation destination for ChromeOS, as it recreates some partitions on first boot and is generally used as persistant storage, which is not good for the drive overall health)
There are some tools out there for both windows and linux that can write a bit mask over the drive – if you perform this operation, bad sectors can be marked and replaced with reserved ones by drive’s own hardware. 

Meanwhile, can you try to flash the image on other drive to see if problem persists?

hussainalhaddad, 2020-05-12:
> You can’t erase the whole drive by just deleting partitions, you must destroy partition table.
> But honestly, I think the script does it automatically using ‘partx’ . I think the partition table gets [...]

Somehow, rebooting my machine fixed this issue. Now i'm able to flash the image on the same drive, and it's booting and installing fine!

Thanks for your reply.

Ubsefor, 2020-05-12:
> Somehow, rebooting my machine fixed this issue. Now i'm able to flash the image on the same drive, and it's booting and installing fine!
>
> Thanks for your reply.

Good for you then! What recovery are you using by the way?

hussainalhaddad, 2020-05-12:
> Good for you then! What recovery are you using by the way?

Rammus

Ubsefor, 2020-05-12:
> Rammus

Note taken, thank you!

sebanc, 2020-05-12:
Hi,

Both issues are not related.

@Ubsefor, I am not an expert in VMs so I cannot speak for VMs in general but this VM does not seem to properly support loopback devices.
Nevertheless, you should be able to apply the install from windows instructions (the ones after WSL install) and using dd instead of rufus to write the image.

@hussainalhaddad Thanks for posting that a reboot solves this issue. I have already seen this case in other issues and did not know what to suggest.

Ubsefor, 2020-05-12:
> @Ubsefor, I am not an expert in VMs so I cannot speak for VMs in general but this VM does not seem to properly support loopback devices.

I went on and created an image of Chrome OS in VM and then flashed it on USB stick. Now I’m trying to install Chrome OS as Parallels VM from this stick, just for, you know, testing purposes. Later on I plan to install it on ancient HP ProBook that I have lying around somewhere.

> Nevertheless, you should be able to apply the install from windows instructions (the ones after WSL install) and using dd instead of rufus to write the image.

I’m sitting on a Hackintosh install and I’m not installing Windows again. I can dd there perfectly fine.

sebanc, 2020-05-12:
I don't know Parallels VM but for now as far as i know, nobody was able to launch chromeos in a vm. I leave this issue opened for now in case you have questions when you install it on real hardware but chromeos not installing/booting in a vm is normal.

Ubsefor, 2020-05-12:
> I don't know Parallels VM but for now as far as i know, nobody was able to launch chromeos in a vm. I leave this issue opened for now in case you have questions when you install it on real hardware but chromeos not installing/booting in a vm is normal.

It almost booted itself, but it hangs after initialising some drivers for mouse and keyboard, pressing ctrl + alt + f2/f1 prints a few messages, all saying something like «Chrome OS is probably dead», I guess I will look into this problem later and try to boot it in qemu, and, if possible, debug it and see what the problem is, maybe it will also help you to polish newer brunch releases, if we / I manage to make it work.

Ubsefor, 2020-05-12:
Apart from the problem, stated above, Chrome OS works as live-system pretty good, now I’m trying to install it to HDD

closed 2020-05-18 by sebanc