sebanc/brunch #128: installation stuck

au79912, 2020-05-12:
when booting from the usb, the progress bar went to 100% then it showed this the root filesystem on /dev/sda3 is now 1048576 blocks long
and it got stuck there.
any idea as to how i can fix this?

sebanc, 2020-05-12:
Hi, 

Normaly, it 's still copying files at this stage, did you let it run ?

au79912, 2020-05-12:
> Hi,
> 
> Normaly, it 's still copying files at this stage, did you let it run ?

its still running
its been almost 20 min

sebanc, 2020-05-12:
Is it displaying new things like "patches applied" ... The first boot time depends only on the usb speed so it can be very long on very slow drives

au79912, 2020-05-12:
> Is it displaying new things like "patches applied" ... The first boot time depends only on the usb speed so it can be very long on very slow usb flashdrives

nothing as of yet
ill let it run for 30 more min

sebanc, 2020-05-12:
Do you see error messages somewhere ? If you attach a photo of the screen I can try to tell if something is not correct.

au79912, 2020-05-12:
> Do you see error messages somewhere ? If you attach a photo of the screen I can try to tell if something is not correct.

its working now.
thanks for your help

closed 2020-05-12 by au79912