sebanc/brunch #129: error message when installing on harddrive

au79912, 2020-05-12:
i ran chrome os on a usb stick but when i try to install it, it give me these messages 
any idea as to how i can fix them?
[image]

sebanc, 2020-05-12:
Hi,

Why did you use /dev/sda4 as destination ?

Either you want:
- to install fully on this drive (erasing all data) in which case you should use /dev/sda.
or
- to create an image on the partition /dev/sda4 in which case you have to mount the partition and the destination should be a filename (for example chromeos.img) on the mounted partition.

au79912, 2020-05-12:
> Hi,
> 
> Why did you use /dev/sda4 as destination ?
> 
> Either you want:
> 
> * to install fully on the drive /dev/sda (erasing all data) in which case you should use /dev/sda.
>   or
> * to create an image on the partition /dev/sda4 in which case you have to mount the partition and the destination should be a filename (for example chromeos.img) on the mounted partition.

thanks, I was a bit confused

closed 2020-05-12 by au79912